ARMmbed/mbed-os #11001: FPGA CI shield assumes UART behaviour it shouldn't

stevew817, 2019-07-09:
### Description

The FPGA CI shield UART test currently assumes that the Serial HAL's TX IRQ gets fired whenever a new byte can be sent, whereas our implementation fires the TX IRQ whenever a byte actually got sent.
There doesn't seem to be a definition in the mbed documentation anywhere to specify which of the two is correct, but a test shouldn't just assume one when such an ambiguity exists.

CC @c1728p9 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

c1728p9, 2019-07-10:
@fkjagodzinski can you look into fixing this?

kjbracey-arm, 2019-07-11:
Things like `UARTSerial` assume that a TX IRQ works as is traditional for UART's - it's the "TX data port empty" interrupt indicating that we can now give data to the UART.

There are places it would be nice to have a "TX idle" interrupt indicating that it's finished (so it's now safe to sleep/power off), but that API has not been standardised.

If the documentation isn't clear, that may need work, but as far as I'm aware, the Mbed OS code base relies on the "TX empty" semantics, so that's what needs to be tested.

Looking at it, it seems like UARTSerial also relies on immediate interrupt generation if the interrupt is enabled while the "TX empty" or "RX non-empty" condition holds at the instant "attach" is called. (Not in the common path, but I see a potential race if that isn't the case).

kjbracey-arm, 2019-07-11:
If we have the ability to test that, it would be good to document and standardise that immediate-IRQ behaviour - I can imagine there's quite a lot of manually written interrupt handlers that might be flakey if the HAL doesn't behave like that.

fkjagodzinski, 2019-07-26:
* > If the documentation isn't clear, that may need work, but as far as I'm aware, the Mbed OS code base relies on the "TX empty" semantics, so that's what needs to be tested.

* > it would be good to document and standardise that immediate-IRQ behaviour

These will be fixed with #11032 which adds a defined/undefined behavior section to Serial HAL doxy.

fkjagodzinski, 2020-11-24:
I think we can close this. The doc fix has been merged quite a while ago. ;) https://github.com/ARMmbed/mbed-os/pull/11032/files#diff-4b9311f76dd0e4092ded0f755b879462a78fdff4bbc22e0447cb158a21b3b0e3R127-R130

closed 2020-12-17 by MarceloSalazar