ARMmbed/mbed-os #11016: GCC ARM : stack smashing protection fully implemented

LaurentLouf, 2019-07-10:
### Description

- Target : tested on a Nucleo F103RB
- Toolchain : arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
- Tools : mbed-cli 1.10.0
- mbed-os-5.12.1

To reproduce the error I'm seeing, it's pretty simple. First clone blinky : `mbed import mbed-os-example-blinky` . Then edit a profile, for example `mbed-os/tools/profiles/release.json` and add to GCC_ARM -> common the following `"-fstack-protector-all"`. Your file should look like this : 
```
{
    "GCC_ARM": {
        "common": [
            "-c",
            "-Wall",
            "-Wextra",
            "-Wno-unused-parameter",
            "-Wno-missing-field-initializers",
            "-fmessage-length=0",
            "-fno-exceptions",
            "-fno-builtin",
            "-ffunction-sections",
            "-fdata-sections",
            "-funsigned-char",
            "-MMD",
            "-fno-delete-null-pointer-checks",
            "-fomit-frame-pointer",
            "-Os",
            "-DNDEBUG",
            "-g1",
            "-fstack-protector-all"
        ],
        "asm": [
            "-x",
            "assembler-with-cpp"
        ],
        "c": [
            "-std=gnu99"
        ],
        "cxx": [
            "-std=gnu++98",
            "-fno-rtti",
            "-Wvla"
        ],
        "ld": [
            "-Wl,--gc-sections",
            "-Wl,--wrap,main",
            "-Wl,--wrap,_malloc_r",
            "-Wl,--wrap,_free_r",
            "-Wl,--wrap,_realloc_r",
            "-Wl,--wrap,_memalign_r",
            "-Wl,--wrap,_calloc_r",
            "-Wl,--wrap,exit",
            "-Wl,--wrap,atexit",
            "-Wl,-n"
        ]
    },
etc
```

Then compile using GCC_ARM and the edited profile like this `mbed compile -t GCC_ARM --profile mbed-os/tools/profiles/release.json` . Copy the binary to your target then open the serial communication. What I'm getting is `*** stack smashing detected ***: terminated` . I've modified the blinky example to add a printf as first line of the main, it doesn't print. 

On another target (custom board), I've been doing some gdb and I get the following : 
```
(gdb) break __stack_chk_fail 
Breakpoint 1 at 0x800850c
(gdb) disassemble __stack_chk_init 
Dump of assembler code for function __stack_chk_init:
   0x080084e4 <+0>:	ldr	r3, [pc, #12]	; (0x80084f4 <__stack_chk_init+16>)
   0x080084e6 <+2>:	ldr	r2, [r3, #0]
   0x080084e8 <+4>:	cbnz	r2, 0x80084f0 <__stack_chk_init+12>
   0x080084ea <+6>:	sub.w	r2, r2, #16121856	; 0xf60000
   0x080084ee <+10>:	str	r2, [r3, #0]
   0x080084f0 <+12>:	bx	lr
   0x080084f2 <+14>:	nop
   0x080084f4 <+16>:	asrs	r4, r5, #21
   0x080084f6 <+18>:	movs	r0, #0
End of assembler dump.
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
Reset_Handler () at ./mbed-os/targets/TARGET_STM/TARGET_STM32F1/TARGET_NUCLEO_F103RB/device/TOOLCHAIN_GCC_ARM/startup_stm32f103xb.S:76
76	  ldr   r0, =_estack
(gdb) continue
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, 0x0800850c in __stack_chk_fail ()
(gdb) where
#0  0x0800850c in __stack_chk_fail ()
#1  0x08002782 in mbed_toolchain_init () at ./mbed-os/rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:83
#2  0x080028aa in mbed_start () at ./mbed-os/rtos/TARGET_CORTEX/mbed_boot.c:99
#3  0x08004170 in osRtxThreadFree (thread=0x0) at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_thread.c:1128
#4  0x00000000 in ?? ()
(gdb) 
```

Meaning that the error seems to occur quite early in the process. The RTOS has created the main thread that is being started and we don't go any further than the toolchain init (creation of malloc mutex, env mutex and __libc_init_array (global variable allocation). 

So, could this just be that the feature of stack smash protection is not really fully implemented ? Note that adding or removing the flag does change the binary size quite significantly, around 7.5kB of flash of difference between with and without. 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

kjbracey-arm, 2019-07-15:
It could be that there is a real stack smash here. I haven't heard of anyone using that option on Mbed OS, and I'm not familiar with it. From the docs, I don't see any reason it shouldn't work.

**Can you try it in a debug profile build?** With a release build I don't think we can totally trust the backtrace, and in particular, I would not expect `mbed_toolchain_init` to even be performing a stack check - the docs say:

> This is done by adding a guard variable to functions with vulnerable objects. This includes functions that call "alloca", and functions with buffers larger than 8 bytes.

which doesn't apply to `mbed_toolchain_init`.

It may be there's a real stack smash somewhere in the boot. Or maybe there's something weird that happens that boot that trips it up, but if so, that might need looking at anyway - could trip up the compiler more generally.

LaurentLouf, 2019-07-15:
Thanks for getting back @kjbracey-arm . So I tried with the debug profile build, and the stack smashing is still detected. The backtrace doesn't change much : 
```
10:16 $ arm-none-eabi-gdb --se BUILD/NUCLEO_F103RB/GCC_ARM-DEBUG/mbed-os-example-blinky.elf 
GNU gdb (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.50.20181213-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from BUILD/NUCLEO_F103RB/GCC_ARM-DEBUG/mbed-os-example-blinky.elf...                                 
(gdb) break __stack_chk_fail 
Breakpoint 1 at 0x800b7e8
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
Reset_Handler () at ./mbed-os/targets/TARGET_STM/TARGET_STM32F1/TARGET_NUCLEO_F103RB/device/TOOLCHAIN_GCC_ARM/startup_stm32f103xb.S:76
76	  ldr   r0, =_estack
(gdb) continue
Continuing.
Note: automatically using hardware breakpoints for read-only addresses.

Breakpoint 1, 0x0800b7e8 in __stack_chk_fail ()
(gdb) where
#0  0x0800b7e8 in __stack_chk_fail ()
#1  0x080039ae in mbed_toolchain_init () at ./mbed-os/rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:71
#2  0x08003aae in mbed_start () at ./mbed-os/rtos/TARGET_CORTEX/mbed_boot.c:83
#3  0x08007568 in osThreadGetId () at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_thread.c:1690
#4  0x00000000 in ?? ()
(gdb) 

```

kjbracey-arm, 2019-07-15:
Okay, seeing as you have a debugger attached, you should be able to use it to actually figure out where the corruption is. By stepping out from `__stack_chk_fail`, and looking at the disassembly, you should be able to see the code doing the check, and the register contents should hopefully give you an indication of exactly what stack location it thinks is trashed.

If you set that location as a watchpoint, you should be able to go through the boot and catch all people writing to it. Some will be expected, but then towards the end you will hopefully see the sequence "writing to it to prime check in mbed_toolchain_init", "someone else write to it", "hit __stack_chk_fail".

LaurentLouf, 2019-07-15:
My assembly is a bit rusty, so if yours is not, here's the disassembly of __stack_chk_fail (that's not a lot of instructions, so I'll look it up anyway) : 
```
(gdb) up
#1  0x080039ae in mbed_toolchain_init () at ./mbed-os/rtos/TARGET_CORTEX/TOOLCHAIN_GCC_ARM/mbed_boot_gcc_arm.c:71
71	}
(gdb) disassemble __stack_chk_fail 
Dump of assembler code for function __stack_chk_fail:
   0x0800b7d4 <+0>:	push	{lr}
   0x0800b7d6 <+2>:	sub	sp, #52	; 0x34
   0x0800b7d8 <+4>:	mov	r4, sp
   0x0800b7da <+6>:	ldr	r5, [pc, #44]	; (0x800b808 <__stack_chk_fail+52>)
   0x0800b7dc <+8>:	ldmia	r5!, {r0, r1, r2, r3}
   0x0800b7de <+10>:	stmia	r4!, {r0, r1, r2, r3}
   0x0800b7e0 <+12>:	ldmia	r5!, {r0, r1, r2, r3}
   0x0800b7e2 <+14>:	stmia	r4!, {r0, r1, r2, r3}
   0x0800b7e4 <+16>:	ldmia.w	r5, {r0, r1, r2, r3}
   0x0800b7e8 <+20>:	stmia	r4!, {r0, r1, r2}
   0x0800b7ea <+22>:	mov	r0, sp
   0x0800b7ec <+24>:	strb	r3, [r4, #0]
   0x0800b7ee <+26>:	bl	0x800b898 <strlen>
   0x0800b7f2 <+30>:	mov	r1, sp
   0x0800b7f4 <+32>:	mov	r2, r0
   0x0800b7f6 <+34>:	movs	r0, #2
   0x0800b7f8 <+36>:	bl	0x8002c58 <write(int, void const*, size_t)>
   0x0800b7fc <+40>:	movs	r0, #6
   0x0800b7fe <+42>:	bl	0x8010ac8 <raise>
   0x0800b802 <+46>:	movs	r0, #127	; 0x7f
   0x0800b804 <+48>:	bl	0x8003170 <_exit(int)>
   0x0800b808 <+52>:	cmp	r2, #136	; 0x88
   0x0800b80a <+54>:	lsrs	r1, r0, #32
End of assembler dump.
```

kjbracey-arm, 2019-07-15:
It's the disassembly from the next frame up we need - the code that's calling stack_chk_fail. And the registers from that context.

LaurentLouf, 2019-07-15:
Well then : 
```
(gdb) disassemble 0x080039ae
Dump of assembler code for function mbed_toolchain_init:
   0x08003948 <+0>:	push	{lr}
   0x0800394a <+2>:	sub	sp, #12
   0x0800394c <+4>:	ldr	r3, [pc, #100]	; (0x80039b4 <mbed_toolchain_init+108>)
   0x0800394e <+6>:	ldr	r3, [r3, #0]
   0x08003950 <+8>:	str	r3, [sp, #4]
   0x08003952 <+10>:	ldr	r3, [pc, #100]	; (0x80039b8 <mbed_toolchain_init+112>)
   0x08003954 <+12>:	ldr	r2, [pc, #100]	; (0x80039bc <mbed_toolchain_init+116>)
   0x08003956 <+14>:	str	r2, [r3, #0]
   0x08003958 <+16>:	ldr	r3, [pc, #92]	; (0x80039b8 <mbed_toolchain_init+112>)
   0x0800395a <+18>:	movs	r2, #11
   0x0800395c <+20>:	str	r2, [r3, #4]
   0x0800395e <+22>:	ldr	r3, [pc, #88]	; (0x80039b8 <mbed_toolchain_init+112>)
   0x08003960 <+24>:	movs	r2, #28
   0x08003962 <+26>:	str	r2, [r3, #12]
   0x08003964 <+28>:	ldr	r3, [pc, #80]	; (0x80039b8 <mbed_toolchain_init+112>)
   0x08003966 <+30>:	ldr	r2, [pc, #88]	; (0x80039c0 <mbed_toolchain_init+120>)
   0x08003968 <+32>:	str	r2, [r3, #8]
   0x0800396a <+34>:	ldr	r0, [pc, #76]	; (0x80039b8 <mbed_toolchain_init+112>)
   0x0800396c <+36>:	bl	0x8005dd4 <osMutexNew>
   0x08003970 <+40>:	mov	r2, r0
   0x08003972 <+42>:	ldr	r3, [pc, #80]	; (0x80039c4 <mbed_toolchain_init+124>)
   0x08003974 <+44>:	str	r2, [r3, #0]
   0x08003976 <+46>:	ldr	r3, [pc, #80]	; (0x80039c8 <mbed_toolchain_init+128>)
   0x08003978 <+48>:	ldr	r2, [pc, #80]	; (0x80039cc <mbed_toolchain_init+132>)
   0x0800397a <+50>:	str	r2, [r3, #0]
   0x0800397c <+52>:	ldr	r3, [pc, #72]	; (0x80039c8 <mbed_toolchain_init+128>)
--Type <RET> for more, q to quit, c to continue without paging--c
   0x0800397e <+54>:	movs	r2, #11
   0x08003980 <+56>:	str	r2, [r3, #4]
   0x08003982 <+58>:	ldr	r3, [pc, #68]	; (0x80039c8 <mbed_toolchain_init+128>)
   0x08003984 <+60>:	movs	r2, #28
   0x08003986 <+62>:	str	r2, [r3, #12]
   0x08003988 <+64>:	ldr	r3, [pc, #60]	; (0x80039c8 <mbed_toolchain_init+128>)
   0x0800398a <+66>:	ldr	r2, [pc, #68]	; (0x80039d0 <mbed_toolchain_init+136>)
   0x0800398c <+68>:	str	r2, [r3, #8]
   0x0800398e <+70>:	ldr	r0, [pc, #56]	; (0x80039c8 <mbed_toolchain_init+128>)
   0x08003990 <+72>:	bl	0x8005dd4 <osMutexNew>
   0x08003994 <+76>:	mov	r2, r0
   0x08003996 <+78>:	ldr	r3, [pc, #60]	; (0x80039d4 <mbed_toolchain_init+140>)
   0x08003998 <+80>:	str	r2, [r3, #0]
   0x0800399a <+82>:	bl	0x800add0 <__libc_init_array>
   0x0800399e <+86>:	nop
   0x080039a0 <+88>:	ldr	r3, [pc, #16]	; (0x80039b4 <mbed_toolchain_init+108>)
   0x080039a2 <+90>:	ldr	r2, [sp, #4]
   0x080039a4 <+92>:	ldr	r3, [r3, #0]
   0x080039a6 <+94>:	cmp	r2, r3
   0x080039a8 <+96>:	beq.n	0x80039ae <mbed_toolchain_init+102>
   0x080039aa <+98>:	bl	0x800b7d4 <__stack_chk_fail>
=> 0x080039ae <+102>:	add	sp, #12
   0x080039b0 <+104>:	ldr.w	pc, [sp], #4
   0x080039b4 <+108>:	asrs	r4, r5, #22
   0x080039b6 <+110>:	movs	r0, #0
   0x080039b8 <+112>:	lsrs	r0, r0, #26
   0x080039ba <+114>:	movs	r0, #0
   0x080039bc <+116>:	movs	r2, #96	; 0x60
   0x080039be <+118>:	lsrs	r1, r0, #32
   0x080039c0 <+120>:	lsrs	r4, r4, #25
   0x080039c2 <+122>:	movs	r0, #0
   0x080039c4 <+124>:	lsrs	r0, r4, #25
   0x080039c6 <+126>:	movs	r0, #0
   0x080039c8 <+128>:	lsrs	r0, r6, #26
   0x080039ca <+130>:	movs	r0, #0
   0x080039cc <+132>:	movs	r2, #112	; 0x70
   0x080039ce <+134>:	lsrs	r1, r0, #32
   0x080039d0 <+136>:	lsrs	r4, r2, #26
   0x080039d2 <+138>:	movs	r0, #0
   0x080039d4 <+140>:	lsrs	r0, r2, #26
   0x080039d6 <+142>:	movs	r0, #0
End of assembler dump.
```
```
(gdb) info registers 
r0             0x72657420          1919251488
r1             0x616e696d          1634625901
r2             0xa646574           174351732
r3             0x0                 0
r4             0x20002628          536880680
r5             0x8012aa8           134294184
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x80059e9           134240745
sp             0x20002640          0x20002640 <_main_stack+4064>
lr             0x80039af           134232495
pc             0x80039ae           0x80039ae <mbed_toolchain_init+102>
xPSR           0x1000000           16777216
msp            0x20004fa8          0x20004fa8
psp            0x20002608          0x20002608 <_main_stack+4008>
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x2                 2
```

kjbracey-arm, 2019-07-15:
Okay, if that dump can be trusted, it's complaining because stack location 0x20002644 has been corrupted. It thinks it should contain 0, but it contains 0xa646574. 

Which, I note is the text `ted\n`. Would be consistent with someone overrunning a buffer, if there were actually any buffers here. Can't see any obvious candidate sources for that string.

Try sticking a data watchpoint in at 0x20002644 on this image.

kjbracey-arm, 2019-07-15:
If that location's correct, you should see it being written to at PC = 0x08003950, which is priming the check. Maybe put a breakpoint there to confirm the location if you don't see it happen. Could be that the "SP" reading is wrong on the backtrace.

Then the next write should be the corruption.

0xc0170, 2019-08-13:
@LaurentLouf Have you made any progress? As stated earlier, we haven't tested this flag so can't provide much help here. From what I read, it should work (flag enable, setting the guard value and implementing fail handler). Nothing else should be needed. I might start with less (without -all suffix, to see if it is working properly in the test program). As all applies this protection to all functions.

LaurentLouf, 2019-08-13:
No, priorities shifted and we found the cause of the hardfault by other means, so no progress made since last message.

0xc0170, 2019-08-23:
If that is the case, we can close this one ?

0xc0170, 2019-08-27:
@LaurentLouf if you do not plan to look at this further, we might consider closing this as "won't fix" -the flag is not officially supported (should work however). This can be reopened with an update anytime.

I would like to find out why it fails so early in the process.

LaurentLouf, 2019-08-27:
Well the problem is still there, since you just have to activate tack smashing protection on blinky to have the error. So the issue is not really resolved

LaurentLouf, 2019-08-27:
Ok, closing then, I don't have time to investigate any further

closed 2019-08-27 by LaurentLouf