ARMmbed/mbed-os #11025: Cellular TCPSocket: how to get info in sigio

pilotak, 2019-07-11:
### Description
I would like to use `sigio` for indication that socket has been closed by remote or that i have a data pending to receive, but i could not find any example. Could you please provide such a code how to retreive `rx_avail` and `closed` in sigio callback? so I can distinguish why it was called.
The functionality has been added in https://github.com/ARMmbed/mbed-os/pull/10840

Thank you
```cpp
// TCPSocket _socket;

void Test::socketCb() {
    debug("socket change\n");
   // some magic here
}

void Test::connect(){
   ...
   _socket.sigio(callback(this, &Test::socketCb));
}
```

Modem: Ublox SARA-G350
Board: STM32F412ZG
### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

kjbracey-arm, 2019-07-15:
SIGIO deliberately provides no information as a design decision, like all POSIX-style signals. It's a prompt for you to make the same calls you would make in a blocking non-event driven design.

You need to make a socket call to find out what happened. Eg if you're expecting to read data, make a read call. If the connection has closed, that read call will tell you. (By returning 0 for a clean close, or an error for an aborted connection).

Google around a bit and you should find tips on how to do asynchronous socket I/O on Linux or whatever - it's the same API design.

Some previous Mbed discussion here: https://github.com/ARMmbed/mbed-os/issues/3620

pilotak, 2019-07-16:
Thanks a lot for explanation, that solved it!

closed 2019-07-16 by pilotak

kjbracey-arm, 2019-07-16:
Cool. One final important tip for program correctness - when using SIGIO+non-blocking, always make as many socket calls as you need to do, up until you get the "would block" error. Only be doing that can you be certain that there will be another SIGIO to wake you up again. Don't assume one SIGIO per reception.

Similarly on start-up - always attempt a read (or write) immediately after connecting. It's possible there may be data already in the buffer by the time you get to it. Don't wait for a SIGIO.

pilotak, 2019-07-16:
Thank you for the tip, good to know.
Once connected i always send handshake otherwise server closes connection after a few seconds.

my `sigio` (called by eventQueue to bring to blocking code):

```cpp
void serverCb() {
    nsapi_size_or_error_t response = server.recv(data, sizeof(data));

    if (response > NSAPI_ERROR_OK) {
        debug("Message arrived\n");

        while (response > NSAPI_ERROR_OK) {
            ...
            response = server.recv(data, sizeof(data));
        }

    } else {
        if (response == NSAPI_ERROR_NO_SOCKET) {
            debug("Server closed connection\n");
        }

        while (server.recv(data, sizeof(data)) != NSAPI_ERROR_WOULD_BLOCK) {}  // ensure nothing left
    }
}
```

kjbracey-arm, 2019-07-16:
I'd write the whole thing as a single loop, with only one `recv` in it. Don't forget that a closure or error can come in any `recv` call there, and you don't want the error/closure paths repeated at each `recv` call point.

For example you might get "SIGIO, response = 100 bytes, response = ERROR_NOSOCKET", in the case where you got some data and then the connection was dropped.

kjbracey-arm, 2019-07-16:
Also, for TCP, it's best to treat any error other than "would block" as fatal. Don't look for specific errors. TCP is by definition a reliable transport, so if it gives an error, that means it's broken the connection.

(A response of "0" is the clean non-error close.)

pilotak, 2019-07-16:
did you mean like that?
```cpp
void serverCb() {
    nsapi_size_or_error_t response = server.recv(data, sizeof(data));

    while (server.recv(data, sizeof(data)) != NSAPI_ERROR_WOULD_BLOCK) {
        if (response > NSAPI_ERROR_OK) {
            debug("Message arrived\n");
            ...

        } else {
            debug("Closed connection\n");
        }
    }
}
```

kjbracey-arm, 2019-07-16:
Not sure what your protocol looks like; exactly how to structure it will depend on how you know when you're finished. Maybe:

    while (true) {
        response = recv(...);
        if (response == would block) {
            // come back on next sigio
            break;
        } else if (response < 0) {
            error handling;
            return;
        } else if (response == 0) {
            clean closure handling (process buffer?)
            return;
        }
        // do whatever you do with partial data (nothing? error if too long?))
    }

Don't forget that with TCP the `recv` data boundaries may bear no relationship to your protocol. You may get whatever you're expecting in bits, or you may be able to get multiple "units" in 1 read. The latter you can deal with by limiting your `recv` length, but you always need to be able to accumulate bits into your buffer.

pilotak, 2019-07-16:
alright, thanks a lot