ARMmbed/mbed-os #11031: STM serial_device IRQ handler loops infinitely when buffer size >256 bytes

caoyuan96421, 2019-07-12:
### Description

In the STM platform _serial_device.c_ driver (I think all series of STM targets are the same in this issue), handler `serial_irq_handler_asynch`, the local variable `i` is defined as `uint8_t`. When the size of the rx_buff passed in to the original call to `read` is larger than 256 bytes, the following piece of code would loop forever due to overflowing of `i`.

```C++
if (obj_s->events & SERIAL_EVENT_RX_CHARACTER_MATCH) {
        if (buf != NULL) {
            for (i = 0; i < obj->rx_buff.pos; i++) {         //   <---------------- PROBLEM
                if (buf[i] == obj->char_match) {
                    obj->rx_buff.pos = i;
                    return_event |= (SERIAL_EVENT_RX_CHARACTER_MATCH & obj_s->events);
                    serial_rx_abort_asynch(obj);
                    break;
                }
            }
        }
    }
```

Result is hanging up of the MCU since this is in an IRQ. This is tested on NUCLEO_L476RG board.
### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

jeromecoutant, 2019-07-12:
Hi

I agree. As rx_buff.pos is size_t type, we could have some issue :-(

Please, raise a pull request, I will approve it !

Thx

caoyuan96421, 2019-07-13:
@jeromecoutant 
Please approve PR #11042.

Thanks!

jeromecoutant, 2019-07-18:
Hi
Should be closed as #11042 is merged
Thx

closed 2019-07-25 by caoyuan96421