ARMmbed/mbed-os #11040: New Mbed Exporters for Eclipse and VSCode: design proposal

vmedcy, 2019-07-12:
### Description

This is a request for comments on new feature that enables the IDE projects exported by Mbed CLI to use `mbed compile` command to build the project instead of the `make` command, and adjust debug launch configurations to use the ELF files generated by Mbed CLI.

Such feature provides seamless integration of vendor post-binary hooks that implement advanced post-build ELF processing with the IDE debug experience. In particular, this is required to support PSoC 64 Secure MCUs that require ELF signed with Mbed CLI python hook, see https://github.com/ARMmbed/mbed-os/pull/11018.

### Proposal

Existing Mbed CLI exporters are based on makefile exporter:

- eclipse_gcc_arm
- eclipse_arm
- eclipse_iar
- vscode_gcc_arm
- vscode_arm
- vscode_iar

The exporters are implemented as python classes in mbed-os/tools/export, with the following inheritance:
```
Exporter
|- Makefile
   |- Eclipse
   |  |- EclipseGcc
   |  |- EclipseArmc5
   |  |- EclipseIAR
   |- VSCode
      |- VSCodeGcc
      |- VSCodeArmc5
      |- VSCodeIAR
```

This is a proposal to implement new exporters that invoke "mbed compile" as the build action, instead of relying on make build flow using generated Makefile:

- eclipse_mbed_gcc_arm
- eclipse_mbed_arm
- eclipse_mbed_iar
- vscode_mbed_gcc_arm
- vscode_mbed_arm
- vscode_mbed_iar

The debug configurations generated by such exporters (Eclipse *.launch files or VSCode launch.json) should be able to locate the generated ELF at the appropriate locations as defined by Mbed CLI (BUILD/TARGET/TOOLCHAIN/appname.elf instead of BUILD/appname.elf).

The new exporter classes will inherit directly from base Exporter class, implementing required abstract method generate similarly to existing Eclipse/VSCode project generators, but without calling any Makefile generator code.
```
Exporter
|- Makefile
|   |- Eclipse
|   |  |- EclipseGcc
|   |  |- EclipseArmc5
|   |  |- EclipseIAR
|   |- VSCode
|      |- VSCodeGcc
|      |- VSCodeArmc5
|      |- VSCodeIAR
|- EclipseMbed
|      |- EclipseMbedGcc
|      |- EclipseMbedArmc5
|      |- EclipseMbedIAR
|- VSCodeMbed
|      |- VSCodeMbedGcc
|      |- VSCodeMbedArmc5
|      |- VSCodeMbedIAR
```

The exporters should be able to expand the IDE build commands from the project templates (.cproject.tmpl or tasks.tmpl) to use the Mbed CLI build command instead of "make":

    mbed compile --target $target --toolchain $toolchain --profile $profile

The existing exporters can be extended to just insert the default "make" invocation instead of the custom build command. This allows to reuse the project templates for both flows, avoiding code duplication and minimizing maintenance effort for new exporters.
The launch configurations can be similarly expanded to lookup the ELF either in BUILD or BUILD/$target/$toolchain.
The existing VSCode exporter documentation can be updated to just the new exporters instead of manually patching tasks.json and launch.json:
https://os.mbed.com/docs/mbed-os/v5.13/tutorials/visual-studio-code.html

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

See also: https://github.com/ARMmbed/mbed-os/issues/8415

cc @theotherjimmy @bridadan @maclobdell @AndriiLishchynskyi @amak-cy @morser499 @romanjoe

theotherjimmy, 2019-07-12:
@vmedcy In the future, please use the @ARMmbed/mbed-os-tools Alias for the tools team instead of mentioning people directly. This avoids issues when the tools team's membership changes.

loverdeg-ep, 2019-07-12:
@theotherjimmy We all know you have the answers though.

theotherjimmy, 2019-07-12:
@loverdeg-ep Thanks, that's flattering. I'm going to let @ARMmbed/mbed-os-toos handle this one.

loverdeg-ep, 2019-07-17:
@janjongboom

madchutney, 2019-08-02:
I have raised this internally, to look at prioritisation and ongoing maintenance options.

There is a new team looking after the build tools (including exporters) which does not currently have a the experience to provide feedback in this area without significant effort.

maclobdell, 2019-08-15:
@madchutney - any update?  This one will become critical soon.  I know of some people that prefer similar environments in which mbed cli commands are called under the IDE hood.  This is how Mbed Studio is working right?

loverdeg-ep, 2019-08-16:
Also see @janjongboom's https://github.com/janjongboom/mbed-vscode-generator

madchutney, 2019-09-03:
@maclobdell The long term strategy is still being discussed but short term priorities of the tools team priorities do not include exporters, so we won't be looking at this for the time being.

JojoS62, 2019-12-20:
> Also see @janjongboom's https://github.com/janjongboom/mbed-vscode-generator
But this exporter or tool are using the make exporter, not direct mbed-cli. It requires to install a make tool (not standard on windows and not included in gcc-arm toolchain). And the tool must be started manually again after adding or changing subdirs, json configs or updating mbed-os.
mbed-cli handles all this automatically and can be seen as a reference build system already.
I've some experience in writting an exporter, but maintaining them is no fun. So, with using mbed-cli as the base you get rid off a lot of problems.
My favorite is also VSCode, the configuration is very easy and can be simply copied by a few files. Only the Intellisense needs the include path from the exporter.

loverdeg-ep, 2019-12-20:
Excellent point. I didn't realize it was using make.
No longer a fan. But it does work.

janjongboom, 2019-12-24:
@loverdeg-ep @JojoS62 It's using the Make exporter, but it doesn't require make to be installed on the system. It sets up everything to build and debug using Mbed CLI, and uses Mbed CLI to set up the define's for your type hinting / intellisense.

JojoS62, 2019-12-25:
thanks for your correction, sorry that I got it wrong. I will try this soon.
And Merry Christmas.

adbridge, 2020-10-15:
As Mbed 5 tools are no longer supported and the tools have been rewritten for Mbed 6, this issue in it's current form is no longer applicable. I will pass this onto the new tools team to capture any requirements from it as they continue to evolve the new tools. I'll thus close this - any further required discussion around this issue should happen on our forums ...

closed 2020-10-15 by adbridge