ARMmbed/mbed-os #11044: USBSerial not working in for FRDM-K22f, MBED 5.13

rkreifeldt, 2019-07-14:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

linlingao, 2019-07-16:
@armmbed/nxp

0xc0170, 2019-08-13:
@maciejbocianski should this work for K22F?

cc @ARMmbed/team-nxp

maciejbocianski, 2019-08-13:
I wouldn't expect that it works as this target doesn't have `USBDEVICE` in target configuration [(targets.json).](https://github.com/ARMmbed/mbed-os/blob/b82cfcc38be1b93fdb7a7192100569f060011569/targets/targets.json#L1200-L1246)
Freescale has generic implementation for few kinetics devices [(USBPhy_Kinetis.cpp)](https://github.com/ARMmbed/mbed-os/blob/b82cfcc38be1b93fdb7a7192100569f060011569/usb/device/targets/TARGET_Freescale/USBPhy_Kinetis.cpp#L17-L21) but actually only `K64F` was tested that is why only `K64F` has USB enabled https://github.com/ARMmbed/mbed-os/blob/b82cfcc38be1b93fdb7a7192100569f060011569/targets/targets.json#L1521

@fkjagodzinski @c1728p9

rkreifeldt, 2019-08-15:
@maciejbocianski 
If you look at my original report I noted that the USBDevice was not included in the targets.json

I assumed this was a temp problem that would be easily fixed. Is no one at NXP working on this?

So @armMbed/team-nxp are you still supporting MBED as a platform? Do you plan to get things up and working for MBED 5?

rkreifeldt, 2019-08-15:
Should I be worried that ARMmbed/team-nxp gets a 404 error when you click the link? In NXP still supporting MBED?

mmahadevan108, 2019-08-15:
@rkreifeldt I will take a look at this and get back to you.

0xc0170, 2019-08-21:
> Should I be worried that ARMmbed/team-nxp gets a 404 error when you click the link? In NXP still supporting MBED?

Teams are not publicly visible. they are.

mmahadevan108, 2019-08-21:
@rkreifeldt I am working with the ARM team to enable USBDEVICE inside K22F. Current mbed USB test RAM requirements cannot be met by K22F.

mmahadevan108, 2019-08-21:
Below PR to add USB support on K22F
https://github.com/ARMmbed/mbed-os/pull/11284

0xc0170, 2019-08-22:
PR integrated. @rkreifeldt please retest on latest master and report back if any issues

rkreifeldt, 2019-08-22:
Thanks I will, I assume I'll have to grab the latest master from Github instead of getting it from the online export tools.

0xc0170, 2019-08-28:
Yes, I'll close this as resolved. IF there are issues, pls update

closed 2019-08-28 by 0xc0170