ARMmbed/mbed-os #11045: Building for DISCO_F746NG on ARM toolchain results in a link warning

RonEld, 2019-07-14:
### Description

Build an application for `DISCO_F746NG` target will result in a link warning:
~~~
[DEBUG] Return: 0
[Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
[Warning] JfazrC@44,9: 'MBED_RAM_START' macro redefined [-Wmacro-redefined]
[Warning] JfazrC@45,9: 'MBED_RAM_SIZE' macro redefined [-Wmacro-redefined]
[DEBUG] Errors: Warning: L3912W: Option 'legacyalign' is deprecated.
[DEBUG] Errors: /tmp/JfazrC:44:9: warning: 'MBED_RAM_START' macro redefined [-Wmacro-redefined]
[DEBUG] Errors: #define MBED_RAM_START              0x20000000
[DEBUG] Errors:         ^
[DEBUG] Errors: <command line>:5:9: note: previous definition is here
[DEBUG] Errors: #define MBED_RAM_START 0x20010000
[DEBUG] Errors:         ^
[DEBUG] Errors: /tmp/JfazrC:45:9: warning: 'MBED_RAM_SIZE' macro redefined [-Wmacro-redefined]
[DEBUG] Errors: #define MBED_RAM_SIZE               0x50000
[DEBUG] Errors:         ^
[DEBUG] Errors: <command line>:6:9: note: previous definition is here
[DEBUG] Errors: #define MBED_RAM_SIZE 0x40000
[DEBUG] Errors:         ^
[DEBUG] Errors: 2 warnings generated.
[DEBUG] Errors: Finished: 0 information, 1 warning and 0 error messages.
~~~
As mentioned [this post](https://forums.mbed.com/t/disco-f746ng-compilation-warning-incompatible-redefinition-of-macro-mbed-ram-size-in-tmp-wkrril/4898) it probably affects functionality of the application.

Reason is that in the [scatter file](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F746xG/device/TOOLCHAIN_ARM_STD/stm32f746xg.sct#L45):
~~~
#define MBED_RAM_START              0x20000000
#define MBED_RAM_SIZE               0x50000
~~~
But in the generated `profile-ld` there is the following information:
~~~ 
        "--predefine=\"-DMBED_RAM_SIZE=0x40000\"", 
        "--predefine=\"-DMBED_RAM_START=0x20010000\"", 
~~~

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

jeromecoutant, 2019-07-15:
Hi @RonEld 
I confirm the issue.
I tried some time ago #10323 but I didn't get enough time to make it clean.

RonEld, 2019-07-15:
@jeromecoutant Thank you for your confirmation!
Does #10323 fix this issue?

0xc0170, 2019-08-13:
@ARMmbed/team-st-mcd Please review

jeromecoutant, 2019-10-18:
Please check #11711
Thx

jeromecoutant, 2019-11-07:
To be closed as #11711 is merged

closed 2019-11-07 by 0xc0170