ARMmbed/mbed-os #11047: Cypress target was removed

jenia81, 2019-07-15:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jenia81, 2019-07-15:
cc @moshe-shahar

maclobdell, 2019-07-15:
@ARMmbed/team-cypress  Please share an update on when the PSA targets will be added back.  I was notified of Cypress' plan to remove the PSA targets temporarily a few weeks ago, however this presents a problem because patch releases are not supposed to have breaking changes.   Please let us know what your plan is.

csd-cypress, 2019-07-18:
> @ARMmbed/team-cypress Please share an update on when the PSA targets will be added back. I was notified of Cypress' plan to remove the PSA targets temporarily a few weeks ago, however this presents [...]

@maclobdell the 062 PSA target was submitted in error. We do not plan to support PSA targets for 062 boards. This support is planned for 064 boards only but this is not planned until later in the year.

40Grit, 2019-07-18:
@csd-cypress I was going to design a 62 into a product and use MbedOS's PSA support...

csd-cypress, 2019-07-18:
@40Grit if you want send me an email at christian.daudt (at) cypress.com and I can put you in touch with our security product team to discuss options.

0xc0170, 2019-08-13:
@csd-cypress Thanks for the answer. I consider this resolved.

closed 2019-08-13 by 0xc0170