ARMmbed/mbed-os #11048: IM880B startup files have wrong line-endings. Breaks checkout in Mac or Linux.

SeppoTakalo, 2019-07-15:
### Description

Following files have Windows line endings and therefore break checkout & rebase when done in Linux or Mac environment:
```
targets/TARGET_STM/TARGET_STM32L1/TARGET_IM880B/device/TOOLCHAIN_ARM_MICRO/startup_stm32l151xba.S
targets/TARGET_STM/TARGET_STM32L1/TARGET_IM880B/device/TOOLCHAIN_ARM_STD/startup_stm32l151xba.S
targets/TARGET_STM/TARGET_STM32L1/TARGET_IM880B/device/TOOLCHAIN_GCC_ARM/startup_stm32l151xba.S
targets/TARGET_STM/TARGET_STM32L1/TARGET_IM880B/device/TOOLCHAIN_IAR/startup_stm32l151xba.S
```

How to reproduce:
```
~/src/mbed-os$ git reset --hard origin/master
~/src/mbed-os$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   targets/TARGET_STM/TARGET_STM32L1/TARGET_IM880B/device/TOOLCHAIN_ARM_MICRO/startup_stm32l151xba.S
	modified:   targets/TARGET_STM/TARGET_STM32L1/TARGET_IM880B/device/TOOLCHAIN_ARM_STD/startup_stm32l151xba.S
	modified:   targets/TARGET_STM/TARGET_STM32L1/TARGET_IM880B/device/TOOLCHAIN_GCC_ARM/startup_stm32l151xba.S
	modified:   targets/TARGET_STM/TARGET_STM32L1/TARGET_IM880B/device/TOOLCHAIN_IAR/startup_stm32l151xba.S

```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

jeromecoutant, 2019-07-15:
I confirm the issue :-(

linlingao, 2019-07-15:
@SeppoTakalo Thanks for the fix. I've been agonizing over this!

PR #11049 (2019-07-17): Normalize line endings for IM880B startup files.
### Description

Fix #11048 by normalizing line endings in `.S` files. 

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@kjbracey-arm 

### Release Notes
closed 2019-07-17 by evedon