ARMmbed/mbed-os #11055: ARMmbed--- Invalid copy assignmen Error

deventank7, 2019-07-16:
I have imported simple MQTT code and I am getting this error. 
How can I solve this issue?

Invalid copy assignment of a NonCopyable resource: NonCopyable<T> &mbed::NonCopyable<mbed::Timer>::operator=(const NonCopyable<T> &) [T = mbed::Timer]

kjbracey-arm, 2019-07-16:
You're trying to copy a non-copyable object - a `Timer`. Most "driver" objects are not copyable - they need to be constructed somewhere and stay operating there; they're not simple objects.

The error message should have told you where this is being attempted.

Maybe you're accidentally passing some high-level API object by value instead of by reference.

linlingao, 2019-07-17:
@deventank7 This does not appear to be Mbed OS issue. Please close it if you concur.

linlingao, 2019-07-29:
Will close this due to lack of response.

closed 2019-07-29 by ciarmcom