ARMmbed/mbed-os #11057: FUTURE_SEQUANA: missing serial_free

fkjagodzinski, 2019-07-16:
### Description

`FUTURE_SEQUANA` (also `FUTURE_SEQUANA_M0`) does not have the [`serial_free`](https://github.com/ARMmbed/mbed-os/blob/9cdfe3778392dbd1286c2dc4bde9eea72f411b8e/hal/serial_api.h#L128) implemented.

This was exposed by https://github.com/ARMmbed/mbed-os/pull/10965#issuecomment-509369593 and may be reproduced with any code calling `serial_free()` such as:

```cpp
#if !DEVICE_SERIAL
#error [NOT_SUPPORTED] SERIAL not supported for this target
#endif

#include "hal/serial_api.h"

int main()
{
    serial_t serial;
    serial_init(&serial, USBTX, USBRX);
    serial_baud(&serial, 9600);
    serial_format(&serial, 8, ParityNone, 1);

    const int msg[] = {'t', 'e', 's', 't', '\r', '\n'};
    for (size_t i = 0; i < sizeof msg; i++) {
        serial_putc(&serial, msg[i]);
    }

    serial_free(&serial);
}
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

fkjagodzinski, 2019-07-16:
@lrusinowicz do you think you could take care of that?

lrusinowicz, 2019-07-29:
> 
> 
> @lrusinowicz do you think you could take care of that?

I shall submit a fix within 2 or 3 days. Sorry for slow response - vacation time.

PR #11131 (2019-08-05): FUTURE_SEQUANA: Add missing serial_free() implementation
### Description

This PR adds missing serial_free() implementation to Sequana HAL implementation.
Should fix #11057; test logs attached.

[tests_sequana_gcc_20190731.log](https://github.com/ARMmbed/mbed-os/files/3451507/tests_sequana_gcc_20190731.log)

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
closed 2019-08-05 by SeppoTakalo