ARMmbed/mbed-os #11061: Cypress QSPI and QSPIF removed

maclobdell, 2019-07-16:
### Description

https://github.com/ARMmbed/mbed-os/pull/10692 removed QSPI feature and QSPIF component from PSoC6 based Cypress targets.  

A brief look at the review comments indicated that there were compile problems with the QSPI.

Some or all of these problems may be fixed by https://github.com/ARMmbed/mbed-os/pull/11006

@Team-Cypress - please let us know when you plan to add back QSPI/QSPIF support and if the fixes in https://github.com/ARMmbed/mbed-os/pull/11006 are acceptable.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2019-08-09:
>@Team-Cypress - please let us know when you plan to add back QSPI/QSPIF support and if the fixes in #11006 are acceptable.

This issue and the referenced PR could be resolved. Any update @ARMmbed/team-cypress ?

maclobdell, 2019-09-09:
This is resolved now.  Tested with mbed-os version 5.14.0 (release candidate) using mbed-os-example-filesystem.  It works with no modifications to the app or configuration.  

```
--- Mbed OS filesystem example ---
Mounting the filesystem... K
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    1
    2
    3
    4
    5
    6
    7
    8
    9
    10
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
```

closed 2019-09-09 by maclobdell