ARMmbed/mbed-os #11065: Mbed CLI displays incorrect flash memory size

linlingao, 2019-07-17:
### Description

I'm seeing this on CY8CKIT_062_WIFI_BT. It could happen on other targets.

To reproduce this:
mbed import mbed-os-example-wifi
cd mbed-os-example-wifi
mbed compile -m cy8ckit_062_wifi_bt -t gcc_arm

Mbed CLI displays the following:
| Module             |           .text |       .data |            .bss |
|--------------------|-----------------|-------------|-----------------|
| [fill]             |       288(+288) |     14(+14) |         65(+65) |
| [lib]/c.a          |   29760(+29760) | 2472(+2472) |         89(+89) |
| [lib]/gcc.a        |     3112(+3112) |       0(+0) |           0(+0) |
| [lib]/misc         |       180(+180) |       4(+4) |         28(+28) |
| [lib]/nosys.a      |         32(+32) |       0(+0) |           0(+0) |
| [lib]/stdc++.a     |           0(+0) |       0(+0) |           0(+0) |
| [misc]             |           0(+0) |       0(+0) |           0(+0) |
| main.o             |       596(+596) |       0(+0) |           4(+4) |
| mbed-os/components |       216(+216) |       0(+0) |           0(+0) |
| mbed-os/drivers    |       154(+154) |       0(+0) |           0(+0) |
| mbed-os/features   |   54474(+54474) |   107(+107) | 154081(+154081) |
| mbed-os/hal        |     1288(+1288) |       4(+4) |         67(+67) |
| mbed-os/platform   |     3380(+3380) |   260(+260) |       348(+348) |
| mbed-os/rtos       |     9876(+9876) |   168(+168) |     5981(+5981) |
| mbed-os/targets    |   52208(+52208) |   355(+355) |     3829(+3829) |
| Subtotals          | 155564(+155564) | 3384(+3384) | 164492(+164492) |
Total Static RAM memory (data + bss): 167876(+167876) bytes
Total Flash memory (text + data): 158948(+158948) bytes

Now examine the map file:
ROM starts at 0x1000.0000 with the first 0x2000 bytes reserved for M0+ image.
.cy_m0p_image   0x0000000010000000     0x19bc
                0x0000000010000000                . = ALIGN (0x4)
                0x0000000010000000                __cy_m0p_code_start = .
 *(.cy_m0p_image)
 .cy_m0p_image  0x0000000010000000     0x19bc BUILD/CY8CKIT_062_WIFI_BT/GCC_ARM/mbed-os/targets/TARGET_Cypress/TARGET_PSOC6/psoc6cm0p/TARGET_CM0P_SLEEP/psoc6_01_cm0p_sleep.o
                0x0000000010000000                cy_m0p_image
                0x00000000100019bc                __cy_m0p_code_end = .

Since this is not labeled as .text, I suspect Mbed CLI probably doesn't account for this section.

The M4 text section starts at 0x1000.2000:
.text           0x0000000010002000    0x93f38
                0x0000000010002000                . = ALIGN (0x4)
                0x0000000010002000                __Vectors = .
.rodata starts at 0x1002.7fd0

The text/rodata ends at 0x1009.5f60
                0x0000000010095f60                __etext = .

So Total Flash memory (text + data) should be 0x95f68 or 599K. Mbed CLI calculates just 155K, which might be just the text.

!--
    Required
    Add detailed description of what you are reporting.
    Good example: https://os.mbed.com/docs/mbed-os/latest/contributing/workflow.html
    Things to consider sharing:
    - What target does this relate to?
    - What toolchain (name + version) are you using?
    - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
    - What is the SHA of Mbed OS (git log -n1 --oneline)?
    - Steps to reproduce. (Did you publish code or a test case that exhibits the problem?)
-->

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

linlingao, 2019-07-17:
@maclobdell @mark-edgeworth @mbed-os-tools

DrynnBavis, 2019-07-29:
I saw something similar a while ago but for a different MCU, might be worth reading here:
https://github.com/ARMmbed/mbed-os/issues/10604

0xc0170, 2019-08-13:
cc @ARMmbed/mbed-os-tools

mark-edgeworth, 2019-08-15:
Could be a duplicate of https://github.com/ARMmbed/mbed-os/issues/10922

0xc0170, 2019-08-21:
Looks like duplicate, lets move to older issue referenced above

closed 2019-08-21 by 0xc0170