ARMmbed/mbed-os #11068: Uvision optimisation level

chrissnow, 2019-07-18:
### Description

The default optimisation level is O0 which wastes a lot of resources.

I think O1 is a more appropriate level as dead code\variables are removed without causing debugging issues.

From
[http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0472k/chr1359124221739.html](url)
> Although the debug view produced by -O0 corresponds most closely to the source code, users might prefer the debug view produced by -O1 because this improves the quality of the code without changing the fundamental structure.

Happy to sort a PR if this seems reasonable.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2019-08-13:
This might land soon: https://github.com/ARMmbed/mbed-os/pull/11126

Should resolve this one.

chrissnow, 2019-08-13:
@0xc0170 that wont help, the debug level is hard coded when exported.

0xc0170, 2019-08-13:
I see, in the template. I dont recall now wbut our profile sets it to Og (once Pr is in) then template value should be ignored , shouldn't it?

chrissnow, 2019-08-13:
I don't think the profiles affect this, It would be better if it did set it though.

I have had a look and can't see anything that tries to override  the<Optim> field.
I also suspect that if -Ox was passed as an LD flag it would cause issues since Keil will also pass it the IDE value

[image]

0xc0170, 2019-08-14:
The field not, that one is in the template. We have -O from profiles set via cmd option - I recall it's in misc field in the uvision - check that - you should see it in the cmd when building. From what I recall misc overwrites what is in the template (some options are available via misc and uvision checkboxes, but not all).

chrissnow, 2019-08-14:
My Misc controls has
`-fdata-sections -Wno-deprecated-register -g -c --target=arm-arm-none-eabi -fshort-enums -fshort-wchar -Wno-armcc-pragma-push-pop -Wno-reserved-user-defined-literal -mcpu=cortex-m3 -Wno-armcc-pragma-anon-unions -fno-exceptions -mthumb -fno-rtti -include ./mbed_config.h`

My experience is that that drop down does affect the optimisation level, and the compiler control string seems to update to match when it is set.
```
-xc -std=gnu11 --target=arm-arm-none-eabi -mcpu=cortex-m3 -c
-fno-rtti -funsigned-char -fshort-enums -fshort-wchar
-gdwarf-3 -O0 -ffunction-sections -I . -I
```

0xc0170, 2019-08-14:
Thanks for testing.  The PR should be OK then 👍

PR #11212 (2019-08-20): Change default optimisation level to O1 when exporting to uVision
### Description

  When exporting to uVision the default level is O0 which is wasteful as dead code and variables are included. O1 removes this without causing debugging issues.

From
[http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0472k/chr1359124221739.html](url)
> Although the debug view produced by -O0 corresponds most closely to the source code, users might prefer the debug view produced by -O1 because this improves the quality of the code without changing the fundamental structure.

Fixes #11068

### Pull r [... cut]
closed 2019-08-20 by 0xc0170