ARMmbed/mbed-os #11075: TCP Socket Fails in Multiple Thread Environment

ThePhilomathsApprentice, 2019-07-19:
### Description
Hello Guys,
I'm trying to take some data over Ethernet Modbus Protocol on the Ethernet Network and Send the Data over to the cloud using MQTT protocol via Ethernet.
Both these tasks are done in separate threads.

There is a another parallel thread which is hosting the webserver.
This thread replies with an HTML page when sent a POST request from a web browser on the URL(Ip Address+port(192.xxx.x.xx:8080)).
The sample code for the same is [here](https://os.mbed.com/teams/sandbox/code/http-webserver-example/).

I'm facing trouble regarding the connection of TCPSockets in the Data Sending Thread.
TCPSockets don't initialize(-3009/-3005/-3003) after the Webserver has responded to any request received on the network.

Initially when the Program starts, all the three threads(1(Modbus TCP)+1(Data Send)+1(Webserver)) are started.
the thread_Modbus_TCP and thread_Data_Send work fine as Long as there is no GET or POST request from the network on the specific Ip Address+port(192.xxx.x.xx:8080).

As soon as I open the URL(Ip Address+port(192.xxx.x.xx:8080)), the next time thread_Data_Send is scheduled it is unable to initialize TCPSockets and goes in an infinite loop of trying to open the socket.

The thread_Modbus_TCP & thread_Data_Send threads have local TCPSocket intializations and are correspondingly closed in the thread itself before going to sleep.

Some points to note:
1) The thread_Modbus_TCP doesn't have any problems trying to initialise a TCPSocket Before or even AFTER the webpage has been requested on the network.
2) it is only when the thread_Data_Send is scheduled does it have problems opening the socket and gives errors(-3009/-3005/-3003, -3009 being more consistent than others.) 
3) While the thread_Data_Send is horrendously/seemingly infinitely trying to open the socket, the firmware responds to the GET requests on the network perfectly fine.
The webpage, when refreshed, is getting response from my K64F perfectly and it gets reloaded again.

I'm unable to find what exactly is interfering in each others connectivity.

For Ethernet's Initialization, I'm Using eth.connect, where eth is an object of EthernetInterface( EthernetInterface eth; )
For Sockets Initialization, I'm Using TCPSocket socket/sock, where socket and sock  are an object of TCPSocket in thread_Modbus_TCP and thread_Data_Send respectively.
AND, The library has used TCPSocket* httpSocket = new TCPSocket();, where httpSocket sock is an object of TCPSocket in Webserver_thread.

Can Someone Please tell me if I'm doing something wrong or any workaround to solve the problem ?
Problem is persistent when compiled in either mbed-cli or MCUXpresso IDE(with SDK for K64F & Processor: MK64FN1MOXXX12(Package: QFP100)).

### Issue request type
- What target does this relate to? --> K64F
- What toolchain (name + version) are you using? --> GCC_ARM(v8.2.1)
- What tools (name + version - is it mbed-cli, online compiler or IDE) are you using? --> mbed-cli and MCUXpresso IDE 

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-08-13:
cc @ARMmbed/mbed-os-ipcore

michalpasztamobica, 2019-09-30:
Hi @ThePhilomathsApprentice . Sorry we have kept you waiting for so long.

We have a [Greentea test](https://github.com/ARMmbed/mbed-os/blob/f30f7633e0f3def104e100ed8c7dff0df77cd1c7/TESTS/netsocket/tcp/tcpsocket_echotest.cpp#L131) showcasing multithreaded operation with a `TCPSocket`. I suggest to use this as a starting point. `TCPSocket` generally is thread-safe and we've seen it work well with K64F.
The [example](https://os.mbed.com/teams/sandbox/code/http-webserver-example) you have pasted above seems rather old - for example it relies on a long-deprecated [easy_connect library](https://github.com/ARMmbed/easy-connect).
Regarding the modbus implementation - I guess we'd have to see the source code to tell if anything is wrong, but I suggest you start by looking into the newer multithreading examples.

Please let us know if there is anything else we can help with 😃

SeppoTakalo, 2019-09-30:
The example is also using `TCPServer` class [which is deprecated](https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/TCPServer.h#L38..L40) in Mbed OS 5.10.

See new API from: https://os.mbed.com/docs/mbed-os/v5.14/apis/tcpsocket.html#server-socket

Regarding the [error codes](https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/nsapi_types.h#L37..L59):
```
    NSAPI_ERROR_PARAMETER           = -3003,     /*!< invalid configuration */
    NSAPI_ERROR_NO_CONNECTION       = -3004,     /*!< not connected to a network */
    NSAPI_ERROR_NO_SOCKET           = -3005,     /*!< socket not available for use */
    NSAPI_ERROR_NO_ADDRESS          = -3006,     /*!< IP address is not known */
    NSAPI_ERROR_NO_MEMORY           = -3007,     /*!< memory resource not available */
    NSAPI_ERROR_NO_SSID             = -3008,     /*!< ssid not found */
    NSAPI_ERROR_DNS_FAILURE         = -3009,     /*!< DNS failed to complete successfully */
```

Sounds like you might be using string based APIs if you constantly fail with DNS resolving.. Various reasons might be because of that, one being that DNS resolving requires one socket. By default we allow only 4 sockets at the same time. [This can be changed](https://os.mbed.com/docs/mbed-os/v5.14/reference/configuration-connectivity.html). See `lwip.socket-max`.

ThePhilomathsApprentice, 2019-10-01:
Hello Guys, 
Sorry for a late update.

The Sockets weren't properly handled in the Library I used for Webserver, which was fixed a few weeks ago.

By the time I fixed it, I completely forgot about this thread that I started here, and forgot to update it here.

But anyways, thanks for your support.

Closing this issue for now. 

Thank again. ;)

closed 2019-10-01 by ThePhilomathsApprentice