ARMmbed/mbed-os #11081: None of the USB API examples work on MAX32630FTHR using mbed 5.13

idea--list, 2019-07-20:
### Description

Target: MAX32630FTHR board
Mbed-OS: V5.13.1
Tools: mbed-os online compiler, alternative: Mbed Studio v0.53

I need to implement USBMSD to access the SD card inserted in a MAX32630FTHR board.
First problem: the board has an SD-card slot out of the box, but in targets.json only "BLE" is listed as feature "STORAGE" is not!
So i created this mbed_app.json file in the root of my project:
`{
"target_overrides": {
    "MAX32630FTHR": {
         "target.features_add": ["STORAGE"],
         "target.components_add": ["SD"],
         "sd.SPI_MOSI": "P0_5",
         "sd.SPI_MISO": "P0_6",
         "sd.SPI_CLK": "P0_4",
         "sd.SPI_CS": "P0_7"
    }
}
}`

Only after the above addition can one use the SD functionality of the board (IMHO it should work without any extra code addition as many simply will not find out how to do it).

Then i realized i also need USBMSD to access the content of the sd card while the board is connected to a PC. So i found the documentation about USB APIs here:
[https://os.mbed.com/docs/mbed-os/v5.13/apis/index.html](https://os.mbed.com/docs/mbed-os/v5.13/apis/index.html)

Copy&pasted the example code into the online compiler, compiled successfully without any errors. Deployed the .bin file on the board. After resetting the board i get this error on the serial monitor:
`++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x3865
Error Value: 0x0
Current Thread: main  Id: 0x20004AF0 Entry: 0xAF8F StackSize: 0x1000 StackMem: 0x20003598 SP: 0x2000444C 
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt="MAX32630"
-- MbedOS Error Info --
This board does not have a hardware USB driver`

Well... googled for it, turns out others have similar issues on another board recently:
[https://os.mbed.com/questions/85899/Get-USB-to-Work-on-DISCO-F429ZI/](https://os.mbed.com/questions/85899/Get-USB-to-Work-on-DISCO-F429ZI/)
It suggests adding "USBDEVICE" to the "device_has" list in targets.json would help...

So i install Mbed Studio (because in the online compiler one can not alter targets.json file).
After adding "USBDEVICE" to the definition of MAX32630FTHR board, i start clean building my project. Just before it would complete i get an error:
`Error: L6218E: Undefined symbol get_usb_phy()  (referred from BUILD/MAX32630FTHR/ARMC6/mbed-os/usb/device/USBAudio/USBAudio.o).`
Meaning with Mbed Studio i can not even upload it to the board, but at least now i can find USBDEVICE related files in the build directory...

Then i try all the other USB related APIs hoping at least one will work and let me go further, but unfortunately all of them throw the same error.

By now i already feel quite frustrated, look further and find the tutorial section in the documentation of mbed here:
[https://os.mbed.com/docs/mbed-os/v5.13/tutorials/index.html](https://os.mbed.com/docs/mbed-os/v5.13/tutorials/index.html)

 Great, now i will find out what do i wrong. Excitement increases as i find the section called "Using USB APIs". There is even a tutorial link for the USBAudio component but guess what... clicking on the link just reloads the same page without linking to the tutorial.

At this point i got stuck and need help please what do i need to get USBMSD work on this board as the documentation is really poor in this regard.

### Issue request type

    [x] Question
    [ ] Enhancement
    [x] Bug

AGlass0fMilk, 2019-07-21:
Hi @idea--list,

Unfortunately, that target does not currently have an Mbed USBDevice implementation. You can see the targets that have been ported to Mbed's USBDevice API here:

https://github.com/ARMmbed/mbed-os/tree/master/usb/device/targets

idea--list, 2019-07-26:
Hi @AGlass0fMilk ,
I am a bit confused in that regard as i found an USBMSD [example ](https://os.mbed.com/teams/MaximIntegrated/code/FTHR_USBMSD_Demo/) code on os.mbed.com, latest update was in February 2018, so before mbed os v5.10 has been released.
If i import that example to the online compiler, there are some folders like "sd-driver", "USBMSD_BD" and also one called "USBDEVICE". The code is working, however since mbed v5.10 USBDEVICE API is already part of the core. So maybe a more skilled person could add USBDevice support into mbed core for this board based on that older example in just some minutes.

0xc0170, 2019-08-13:
@idea--list The example referenced is for old USB (it has been deprecated,  thus the last update was long time ago). You can see it here : https://os.mbed.com/users/mbed_official/code/USBDevice/ (this is the parent library). You can still use it (mbed 2).  As the target not yet supported in the new usb, feel free to implement usb support and send a pull request.

cc @ARMmbed/team-maximintegrated

idea--list, 2019-08-26:
@0xc0170 I began learning c/c++ just a few months ago, but maybe i will try to (re-?)implement USBDEVICE to this board. However it makes me wonder why the old USB **had** to be removed at all? I mean it was not broken! I remember i could compile and run any examples by Maxim that relied on the "old" USBDEVICE even in mbed os 5.10.

If it is removed and still not reimplemented for this board, why does this website claim this board being compatible with mbed os v5.13 if in reality it is not? (I mean i can blink a led without any issues, but as soon as i try to use USBDEVICE or BLE then all the examples simply do not work at all).

How should i begin reimplementing USBDEVICE for this board? I mean i found some files [here ](https://github.com/ARMmbed/mbed-os/tree/master/hal/usb) (that seem to be moving around to make it easy to find them...) Are these of any use during the reimplementation?

Lastly [here ](https://github.com/ARMmbed/mbed-os/tree/master/features/unsupported/USBDevice) i found something called USBDEVICE. Is it the old implementation that has been removed? If so can these be useful during reimplementation?

Sorry for asking so much, but without answers i can not even try to begin the process.

0xc0170, 2019-09-04:
> @0xc0170 I began learning c/c++ just a few months ago, but maybe i will try to (re-?)implement USBDEVICE to this board. However it makes me wonder why the old USB had to be removed at all? I mean it was not broken! I remember i could compile and run any examples by Maxim that relied on the "old" USBDEVICE even in mbed os 5.9.

@maciejbocianski Please review

You either use older mbed library with usb (older unsupported version) or need to port new one to maxim.  
cc @ARMmbed/team-maximintegrated 

> If it is removed and still not reimplemented for this board, why does this website claim this board being compatible with mbed os v5.13 if in reality it is not? (I mean i can blink a led without any [...]

It should be updated, cc @MarceloSalazar 

> Lastly here i found something called USBDEVICE. Is it the old implementation that has been removed? If so can these be useful during reimplementation?

yes, ignore unsupported folder. Use usb implementation that is in the tree in hal folder.

idea--list, 2020-09-12:
@0xc0170 
Read the docs of mbed V6.2: [here](https://os.mbed.com/docs/mbed-os/v6.2/apis/drivers-architecture.html) and [here](https://os.mbed.com/docs/mbed-os/v6.2/apis/usb-overview.html) while also examined the directory structure of current mbed-os.

My understanding is summed up in this picture:
[image]

Everything i marked with blue background is part of the core of mbed-os's USB-stack (also including USBPhy.h) and i should not touch any files of that part.

Only the USBPhyHw part i marked with red background is the component that should be implemented by the board vendor (or any contributor). These files (USBPhyHw.h and USBPhy_template.cpp) come as empty template files with mbed-os, i just should copy these under targets/TARGET_Maxim/ and implement them there. Would code these 2 files right the new USB-stack of mbed should go live on this target. Are my assumptions correct?

maciejbocianski, 2020-09-13:
@idea--list Your understanding is correct. See MbedOS porting guide for more deatils https://os.mbed.com/docs/mbed-os/v6.2/porting/usb-device.html

zamboglu-maxim, 2020-09-24:
> EXTERNAL EMAIL
>
> @0xc0170<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2F0xc0170&data=02%7C01%7Czafer.zamboglu%40maximintegrated.com%7C3460c6fc02194fa6a34608d86089d4ef%7Cfbd909dfea694788a554f24b7854ad03%7C0%7C0%7C637365492392773952&sdata=CJRXPTMmMygVwz32H7Z0Cgi7kSh9T3hn0O%2BqTrM7460%3D&reserved=0> @maciejbocianski<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmaciejbocianski&data=02%7C01%7Czafer.zamboglu%40maximintegrated.com%7C3460c6fc02194fa6a34608d86089d4ef%7Cfbd909dfea694788a554f24b7854ad03%7C0%7C0%7C637365492392783955&sdata=DsOo3jIGG1JByyMnyGUARjh5Br%2B4Lj4rwIdTBR62fsI%3D&reserved=0>
> Found this documentation<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.cypress.com%2Ffile%2F134171%2Fdownload&data=02%7C01%7Czafer.zamboglu%40maximintegrated.com%7C3460c6fc02194fa6a34608d86089d4ef%7Cfbd909dfea694788a554f24b7854ad03%7C0%7C0%7C637365492392793947&sdata=NBf%2BZh8xHQEOP4cXarRf7YOl8pJgoGX9f1%2BoJl%2BUopE%3D&reserved=0> about USB which summarizes the most important things about USB (other docs you linked or i found elsewhere are simply failing making anyone to understand how USB works).
> This time i got far further trying to implement mbed OS's new USB stack on this board than back in last year.
> However the code i came up with does not run on my MAX32630FTHR (but at least it does not cause any hard faults) and as i do not have an oscilloscope i can not debug my code and find the solution.
>
> I tried my best, but can not succeed with this feature for the MAX32630. Is there anyone in the mbed team who has this board and could test&debug my code?

@Sadik Ozer<mailto:Sadik.Ozer@maximintegrated.com>

Sadik, 2020-10-14:
Please don't CC me in this thread. Might be a confusion. Send CC to Sadik Ozer instead.

Sadik, 2020-12-05:
I unsubscribed from this list because I think someone wanted to mention Sadik.Ozer and mentioned me instead. I don't know what this thread or repository is about.-------- Ursprüngliche Nachricht --------Von: ARM mbed CI <notifications@github.com>Datum: Sa., 5. Dez. 2020, 09:21An: ARMmbed/mbed-os <mbed-os@noreply.github.com>Cc: Sadik Özoguz <sadikoezoguz@googlemail.com>, Mention <mention@noreply.github.com>Betreff: Re: [ARMmbed/mbed-os] None of the USB API examples work on MAX32630FTHR using mbed 5.13 (#11081)
@idea--list it has been 5 days since the last reminder. Could you please update the issue header as previously requested?

PR #14075 (2021-01-19): Maxim targets; Add USB stack implementation

### Summary of changes 
USB HW implementation added for Maxim boards.
Fixes #11081 

#### Impact of changes 

This change will conflict with legacy `USBDevice` library. User must not use that library along with this changes.

#### Migration actions required
closed 2021-01-19 by 0xc0170