ARMmbed/mbed-os #11083: Cellular Information

JRubis, 2019-07-22:
### Description

Hi, I have noticed some inconsistencies in the data returned when using Cellular Information. The correct data is returned, however, in most cases, the AT command that was used to query the module is prepended and in another, it isn't. In addition, get_serial_number() also prepends "RDY" Likewise get_manufacture() prepends a \r. Is there a reason for this? I would have expected these methods to return the value of the parameter without the AT command prepended. get_iccid() is the only method that appears to do this.

get_serial_number	"RDY\r\nAT+CGSN\r\r\n866429030121264\r\n\r\n"
get_manufacturer	"\rAT+CGMI\r\r\nQuectel\r\n\r\n"
get_model		"AT+CGMM\r\r\nBG96\r\n\r\n"	
get_revision		"AT+CGMR\r\r\nBG96MAR02A07M1G\r\n\r\n"	
get_imsi		        "AT+CIMI\r\r\n31041"	
get_iccid		        "89015104272289721398"	

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

AriParkkila, 2019-07-26:
@JRubis You need to call `QUECTEL_BG96::init` before any other Cellular APIs like `CellularInformation`. Also `CellularStateMachine` calls `init` at  `NetworkInterface::connect()`.

0xc0170, 2019-08-13:
> @JRubis You need to call QUECTEL_BG96::init before any other Cellular APIs like CellularInformation. Also CellularStateMachine calls init at NetworkInterface::connect().

@JRubis Does that answer your question?

JRubis, 2019-08-13:
Yes it seems to work Ok now. Thank you.

closed 2019-08-13 by JRubis