ARMmbed/mbed-os #11085: LittleFS fopen() is really slow on QSPI flash

janjongboom, 2019-07-22:
### Description

Not sure what the reason is but I find `fopen()` to be (unacceptably) slow when using it with LittleFS mounted file system on a QSPI flash chip. This is on the DISCO-L475VG-IOT01A1 board:

```cpp
        Timer t;
        t.start();
        FILE *write_file = fopen("/fs/weights/y_pred_some_file.idx", "w+");
        t.stop();
        printf("fopen took %d ms.\n", t.read_ms());
```

This takes between 129 and 151 ms. on my board. Is this normal?

@geky 

### Issue request type

    [X] Question
    [ ] Enhancement
    [X] Bug

geky, 2019-07-23:
This is more likely due to directory structure layout than QSPI specifically. I'm guessing it's related to https://github.com/ARMmbed/littlefs/issues/75. Unfortunately all I can say right now is that I am planning on looking at it but it will take time.

janjongboom, 2019-07-24:
@geky Thanks for pointing me to that issue.

0xc0170, 2019-08-13:
Shall this be closed ? As it's being tracked in littlefs repo issue 75

0xc0170, 2019-10-24:
If anyone follows this one, please follow ARMmbed/littlefs#75. 

I'll close this as there is an issue related in the upstream repository.

closed 2019-10-24 by 0xc0170