ARMmbed/mbed-os #11087: Mbed Cellular API for Generic GSM.

yasir-khilji-64, 2019-07-22:
### Description
So I've been trying to make the Mbed OS Cellular API with GSM Sim800/Sim808. But the mbed asserts network interface error.  
[mbed_app.json.txt](https://github.com/ARMmbed/mbed-os/files/3418115/mbed_app.json.txt)
Serial Terminal Output:
mbed-os-example-cellular

Built: Jul 22 2019, 19:10:07
Establishing connection

++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: iface
Location: 0x8002315
File: .\main.cpp+259
Error Value: 0x0
Current Thread: main  Id: 0x200012C4 Entry: 0x800300B StackSize: 0x1000 StackMem: 0x20001A68 SP: 0x200029F4
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=NUCLEO_F103RB
-- MbedOS Error Info --

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

yasir-khilji-64, 2019-07-23:
Will anybody help us rectify the issue. We are on a really tight schedule here.

AriParkkila, 2019-07-26:
@yasir-khilji-64 You are probably missing something like `GENERIC_AT3GPP.provide-default` in the `mbed_app.json` file. You need to change GENERIC_AT3GPP to actual Sim800's modem type.

yasir-khilji-64, 2019-07-26:
I tried your suggestion. But same issue prevails.

AriParkkila, 2019-07-29:
@yasir-khilji-64 `Error Message: Assertion failed: iface` indicates that no network interface was found, which typically means that you use `NetworkInterface::get_default_instance()` without a default modem defined. Please check that your `BUILD/../mbed_config.h` file defines, e.g. `MBED_CONF_GENERIC_AT3GPP_PROVIDE_DEFAULT 1`.

yasir-khilji-64, 2019-07-29:
There is no file named mbed_config.h under BUILD

yasir-khilji-64, 2019-07-29:
No it doesn't defines it. 
`#define MBED_CONF_GENERIC_AT3GPP_PROVIDE_DEFAULT                              0`

yasir-khilji-64, 2019-07-29:
so now
MBED_CONF_CELLULAR_DEBUG_AT 1
GENERIC_AT3GPP_BAUDRATE 19200
GENERIC_AT3GPP_PROVIDE_DEFAULT 1
but i'm not seeing anything on the console now. Not MBED OS error message now.

AriParkkila, 2019-08-01:
@yasir-khilji-64 You probably need to change `GENERIC_AT3GPP_BAUDRATE` to the same for the modem. Those cellular configs should not change console settings at all, maybe you could check that you haven't modified those such as `platform.stdio-baud-rate`.

AriParkkila, 2019-09-13:
I think this issue can be closed due to the remaining problems, if any, are related to stdio-tracing and not to the original question about Cellular API.

loverdeg-ep, 2019-09-14:
@trowbridgec

0xc0170, 2019-09-16:
> so now
MBED_CONF_CELLULAR_DEBUG_AT 1
GENERIC_AT3GPP_BAUDRATE 19200
GENERIC_AT3GPP_PROVIDE_DEFAULT 1
but i'm not seeing anything on the console now. Not MBED OS error message now.

Should  be resolved, if any other, please use a forum for further questions

closed 2019-09-16 by 0xc0170

trowbridgec, 2019-09-16:
Yeah, looks to me like @yasir-khilji-64 needs to configure all of the settings from [here](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/targets/GENERIC/GENERIC_AT3GPP/mbed_lib.json) in the mbed_app.json file.

yasir-khilji-64, 2019-10-28:
Hello community. 
So we got it working in September. But now it isn't establishing connection again. 
My mbed_app.json is here. 

```
{
    "config": {
        "sock-type": "TCP",
        "echo-server-hostname": {
            "help": "Echo server host name.",
            "value": "\"echo.mbedcloudtesting.com\""
        },
        "echo-server-port": {
            "help": "Echo server port number.",
            "value": 7
        },
        "trace-level": {
            "help": "Options are TRACE_LEVEL_ERROR,TRACE_LEVEL_WARN,TRACE_LEVEL_INFO,TRACE_LEVEL_DEBUG",
            "macro_name": "MBED_TRACE_MAX_LEVEL",
            "value": "TRACE_LEVEL_INFO"
        }
  },
    "target_overrides": {
        "*": {
            "target.network-default-interface-type": "CELLULAR",
            "mbed-trace.enable": true,
            "lwip.ipv4-enabled": true,
            "lwip.ethernet-enabled": false,
            "lwip.ppp-enabled": true,
            "lwip.tcp-enabled": true,
            "platform.stdio-convert-newlines": true,
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "platform.stdio-buffered-serial": true,
            "cellular.debug-at": true,
            "nsapi.default-cellular-plmn": "\"41004\"",
            "nsapi.default-cellular-sim-pin": 0,
            "nsapi.default-cellular-apn": "\"zonginternet\"",
            "nsapi.default-cellular-username": 0,
            "nsapi.default-cellular-password": 0,
			"GENERIC_AT3GPP.provide-default": true,
			"GENERIC_AT3GPP.tx": "PC_10",
			"GENERIC_AT3GPP.rx": "PC_11",
			"GENERIC_AT3GPP.baudrate":115200
        },
        "DISCO_L496AG": {
            "target.macros_add": [
                "CELLULAR_DEVICE=STModCellular"
            ],
            "target.components_add": ["STMOD_CELLULAR"],
            "stmod_cellular.provide-default": "true"
        }
    }
}
```

yasir-khilji-64, 2019-10-28:
Also in the mbed_config.h 
```
#define MBED_CONF_GENERIC_AT3GPP_BAUDRATE                                     115200
#define MBED_CONF_GENERIC_AT3GPP_PROVIDE_DEFAULT                              1
#define MBED_CONF_GENERIC_AT3GPP_RX                                           PC_11
#define MBED_CONF_GENERIC_AT3GPP_TX                                           PC_10
```

0xc0170, 2019-10-28:
@yasir-khilji-64 What has changed , have you updated your app ?

yasir-khilji-64, 2019-10-28:
@0xc0170, So, I just imported the mbed-os-cellular-example from github, changed the mbed_app.json file according to my setup. and it is not working. 
I also cloned my previous repo for the working cellular example and still it isn't working. 
Any thoughts ?

yasir-khilji-64, 2019-10-28:
Hey @0xc0170, so the issue is here
```
[00256437ms][INFO][CELL]: AT TX (11): AT+CGDCONT=
[00256442ms][INFO][CELL]: AT TX ( 1): 4
[00256447ms][INFO][CELL]: AT TX ( 1): ,
[00256452ms][INFO][CELL]: AT TX ( 1): "
[00256457ms][INFO][CELL]: AT TX ( 6): IPV4V6
[00256462ms][INFO][CELL]: AT TX ( 1): "
[00256467ms][INFO][CELL]: AT TX ( 1): ,
[00256472ms][INFO][CELL]: AT TX ( 1): "
[00256476ms][INFO][CELL]: AT TX (12): zonginternet
[00256482ms][INFO][CELL]: AT TX ( 1): "
[00256487ms][INFO][CELL]: AT TX ( 1): <cr>
[00256492ms][INFO][CELL]: AT RX (17): <cr><ln>+CME ERROR: 3<cr><ln>
```
Sim800 Series AT Commands specify `<PDP_type>   (Packet Data Protocol type)   IPInternet Protocol (IETF STD 5)`, while in the app it is setting it to `IPV4V6`

yasir-khilji-64, 2019-10-28:
Hey community, so I've found some issue in the `AT_CellularContext.cpp`. 
We are working with the Sim800 series Cellular Modem which has PDP context of 1-3 with only IP PDP Type. AT_CellularContext tries to set a new Context with ID 4 and IPv4v6 PDP Type. 
Please rectify this. We are really on tight schedule.

AnttiKauppila, 2019-10-28:
So if your APN (zonginternet) only supports IPv4 you should disable IPv6 and IPv4v6 from:
https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/targets/GENERIC/GENERIC_AT3GPP/GENERIC_AT3GPP.cpp#L24 and recompile+test
If that does not work, you can try to delete those existing contexts by issuing +CGDCONT=<cid> for all of those contexts separately.

yasir-khilji-64, 2019-10-28:
@AnttiKauppila, the thing is, my APN supports both. Issue is with the GSM Board. 
It only supports 1-3 PDP Context Identifier with on IP PDP Context type.

yasir-khilji-64, 2019-10-28:
@AnttiKauppila Can you tell me how to delete these specific contexts?

AnttiKauppila, 2019-10-28:
@yasir-khilji-64 This should do (run before you try to connect()):
`ATHandler *at = CellularDevice::get_default_instance()->get_at_handler();
    for (int i=0; i<3; i++) {
        print_function("\n\n DELETE CONTEXT\n\n");
        at->cmd_start("AT+CGDCONT=");
        at->write_int(i);
        at->cmd_stop();
        at->resp_start();
        at->resp_stop();
        at->clear_error();
    }`

AnttiKauppila, 2019-10-28:
or even
`for (int i=0; i<3; i++) {
    at->at_cmd_discard("+CGDCONT", "=", "%d", i);
    at->clear_error();
}`

trowbridgec, 2019-10-28:
You can also write your own version of the `get_context()` function. We ran into a similar issue here: https://github.com/ARMmbed/mbed-os/pull/10442.

The logic as it is today ignores an existing context (even if the APN is blank) if it doesn't exactly match the provided APN. This PR lets you override `get_context()` to suit your needs.

yasir-khilji-64, 2019-10-28:
@trowbridgec actually we are really tight on the schedule, so only want to use tested things

mirelachirica, 2019-10-29:
> Can you tell me how to delete these specific contexts?

Contexts can also be deleted by setting "clear-on-connect" flag to true in cellular's mbed_lib.json.

yasir-khilji-64, 2019-10-31:
Thanks for the help community. Cellular Interface is working perfectly.

yasir-khilji-64, 2019-12-16:
Hey community, does anyone know how to get GSM Location using PPP interface?

40Grit, 2019-12-16:
Best to ask at forums.mbed.com these days.

@trowbridgec can provide some insight here though.

trowbridgec, 2019-12-17:
In our experience, you cannot maintain a PPP connection with a cellular module while also sending AT commands to the module as well (e.g. asking for location information). You have to either:

1. Tear down the PPP connection, send the necessary AT commands, and then bring back up the PPP connection

2. Use 2 communications channels from the application MCU to the cellular module (e.g. 2 UARTs). From what I've seen, a lot of the cellular modules have multiple UART connections for this purpose.

I haven't found a better way to do this yet, but I could be missing something...