ARMmbed/mbed-os #11088: Quectel BG96 Wake from PSM

JRubis, 2019-07-22:
### Description

Hi, I am working with a Quectel BG96, ST module connected via the STMOD+ connector on an STM32L4-discovery board. In general, the module seems to be working fine. However, today I went to evaluate PSM functionality and hit a snag. 

It appears that it's up to the application to manage PSM. This is based on what I have read here: https://github.com/ARMmbed/mbed-os/issues/10737

To be honest, I wasn't really expecting that. I thought that the cellular state machine would manage PSM, waking up module when needed. I am not sure why an application needs to be burdened by this.  But OK, for now, let's assume the application needs to do that.

That means the application would need a method to wake the module up. Unless I am missing something (Entirely Possible), it appears there isn't a provided method to toggle the Power Key pin on the BG96 in order to wake it up from PSM. There is a soft_power_on(), but that method resets the module eliminating the benefit of PSM.

What am I missing?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

JRubis, 2019-07-24:
@AriParkkila @mirelachirica

AriParkkila, 2019-07-26:
@JRubis Please see BG96 Hardware Design manual:
```
Either of the following methods will wake up the module from PSM:
 Drive PWRKEY pin to low level will wake up the module.
 When the T3412 timer expires, the module will be automatically woken up.
```

JRubis, 2019-07-26:
Hi @AriParkkila, yes I am pretty familiar with the BG96 and have read the manual many times :-). I have been experimenting with this device and its various modes of operation (PSM, eDRX) for almost a year now using another RTOS and Cellular Framework.

I think what you are suggesting is that a developer would simply use a DO command at the application level to wake up the module?  While that would work, my thought is there should be a standard API the application could call to do this. The implementation may differ from one module to the next so shouldn't this be abstracted away from application-level code? BTW, I really believe the management of PSM would be best handled at a lower layer.

Also, using the T3412 timer might work for synchronous connections, but in the case where the application needs to make asynchronous connections, it would not be the best way.

AriParkkila, 2019-07-26:
@JRubis What kind of API do you have in mind? That sounds like a great idea, but cellular modules seem to be quite different in how to wake up from PSM.

JRubis, 2019-07-26:
@AriParkkila call it something like wake_up_from_psm() and place it in the CellularDevice class similar to other API(s) in that class such as soft_power_on() which are implementation-specific.

trowbridgec, 2019-07-26:
I agree with @JRubis. We have experimented with PSM on a Telit ME910 (#10737), and I believe it makes sense to abstract some of that away into the lower level APIs.

trowbridgec, 2019-07-26:
For context, there's some more PSM discussion in the later comments of #8666 (starting [here](https://github.com/ARMmbed/mbed-os/issues/8666#issuecomment-485898645)).

40Grit, 2019-07-26:
Yes please. @JRubis the Embedded Planet has been struggling with the same issue.

JRubis, 2019-08-07:
Since the cellular state machine is not managing PSM, and the application needs to, how does the application know when the modem has entered PSM? I do not see a specific event listed for that.

closed 2019-10-25 by ciarmcom