ARMmbed/mbed-os #11089: nRF52840 - Attach to UART RX Interrupt not working

MikeDK, 2019-07-23:
### Description

In our product we use a BMD-340 Module from Rigado, which is essentially a Nordic nRF52840.
We use a UART connected to the VCOM pins of a Segger JLink for debugging purposes - TX is used to put out log messages, which is done via a RawSerial object and its async write() function ... and for command input we use the objects attach(cbFunc, SerialBase::RxIrq) function and handle available characters in the interrupt context.

Until mbed-os-5.11.4 this was working fine on the nRF52.

Since mbed-os-5.12 we are not getting any rx interrupt handler calls anymore.

When I debug the attach call, I can see that the execution path goes via serial_irq_set() and then nordic_nrf5_serial_configure() where it does not walk into the if or one of the else if cases, so nothing is done there. This explains why i get not interrupt handler call.
What I don't understand is why.

When looking at the struct serial_s *uart_object structure within nordic_nrf5_serial_configure(), the members rx_asynch and update are both false.
I guess that's the reason why no case is executed.

Should it be possible to do tx async write() calls while having an int handler on the rx side?

I realize there has been some major change in the serial code in the last versions ... has the concept changed too?

Regards,
MikeDK

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

trowbridgec, 2019-07-23:
Just a note, I saw this issue as well on an nRF52840, and it was resolved with mbed-os 5.13.1.

MikeDK, 2019-07-23:
@trowbridgec Unfortunately, I cannot confirm this ... I am using 5.13.1 at the moment, and I still have the problem :/

trowbridgec, 2019-07-23:
Ah, actually now that I think about it, the issue I had was not being able to transmit _to_ the device (not receive data _from_ the device)...

MikeDK, 2019-07-24:
O.K., I have an update on this.

It happens that we are using one of the two NFC pins for UART RX (because we are about to use every single pin of the BMD-340 module) ... I soldered the RX connection to another pin for testing purposes, and now it is working ;)

Background: As described [here](https://rigado.zendesk.com/hc/en-us/articles/225362747-BMD-300-BMD-301-BMD-330-BMD-340-BMD-350-pin-assignments), the two NFC pins are connected together with a shunt resistor (which shall prevent shortcuts when connecting an antenna there).
Because of this many things won't work on that pins ... we already discovered that we can't use one of those pins as SIO2 for a QSPI eeprom ... so it's not surprising UART won't work there either :/

So ... The issue will be closed, problem detected and solved ;)

closed 2019-07-24 by MikeDK