ARMmbed/mbed-os #11091: DISCO_F769NI MIPI-DSI LCD Support

tdjastrzebski, 2019-07-23:
Trying to run mbed-os 5.1.x on flagship STM32F769I-DISCO board I was surprised to realize that MIPI DSI LCD probably was not supported. I found nothing related to DSI in pin map files under https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F769xI/TARGET_DISCO_F769NI
Am I missing something obvious or STM32F769I-DISCO LCD is just not yet supported?
If so, are there any known reasons/issues?
I think STM32F769I-DISCO LCD was supported in mbed-os 2.x

linlingao, 2019-07-29:
@ARMmbed/team-st-mcd

0xc0170, 2019-08-13:
Did you look for examples for the board? I found at least one LCD example: https://os.mbed.com/teams/ST/code/DISCO-F769NI_LCD_demo/

0xc0170, 2019-08-13:
This is the library for the board: https://os.mbed.com/teams/ST/code/LCD_DISCO_F769NI/#fd7c63b20d54

I'll close this issue as resolved

closed 2019-08-13 by 0xc0170

tdjastrzebski, 2019-08-13:
I have never said there were no LCD examples for STM32F769I-DISCO board.
However, I found no support for mbed-os 5.x, no support in pin map files, no 2D Chrom-ART Accelerator support.

0xc0170, 2019-08-14:
We do not provide LCD in the Mbed OS at the moment, thus external example and libraries as in the example.

tdjastrzebski, 2019-08-14:
And this is exactly my point. Flagship STM32F769I-DISCO board is not fully supported in Mbed OS!