ARMmbed/mbed-os #11098: Misleading APPLICATION_ADDR undeclared error if bootloader missing or unable to fit

maclobdell, 2019-07-23:
# Description

Mbed OS version: 5.13.1 (most likely earlier also)
Compiler: GCC, ARM, IAR
Relevant tools versions tested:  mbed-cli (1.10.0), mbed-os-tools (0.0.8)

When building an application that uses managed bootloader mode, such as a Pelion client example (e.g. mbed-os-pelion-example), the error below with undeclared APPLICATION_ADDR can result.

I have found the reason that APPLICATION_ADDR is not defined can be caused by these conditions (could be more):  

1) the area reserved for the bootloader is too small.  (e.g. if "target.header_offset" is set too low and thus overlaps with the bootloader).
2) the bootloader file is not found.  (e.g. if there is a typo in "target.bootloader_img"                     : "bootloader/mbed-bootloader_filename.hex"), then it results in the same error.  	

Ideally, these conditions should result in a descriptive error, but instead result in the below messages, which do not indicate the source of the problem.  

```
[Warning] arm_uc_pal_delta_paal_original_reader.c@46,0: "TRACE_GROUP" redefined
[Error] arm_uc_pal_delta_paal_original_reader.c@29,49: 'APPLICATION_ADDR' undeclared (first use in this function)
[ERROR] .\mbed-cloud-client\update-client-hub\modules\delta-paal\source\arm_uc_pal_delta_paal_original_reader.c:46:0: warning: "TRACE_GROUP" redefined
 #define TRACE_GROUP  "UCPI"

In file included from ./mbed-cloud-client/update-client-hub/modules/common/update-client-common/arm_uc_common.h:27:0,
                 from .\mbed-cloud-client\update-client-hub\modules\delta-paal\source\arm_uc_pal_delta_paal_original_reader.c:24:
./mbed-cloud-client/update-client-hub/modules/common/update-client-common/arm_uc_trace.h:51:0: note: this is the location of the previous definition
 #define TRACE_GROUP  "UC"

.\mbed-cloud-client\update-client-hub\modules\delta-paal\source\arm_uc_pal_delta_paal_original_reader.c: In function 'arm_uc_deltapaal_original_reader':
.\mbed-cloud-client\update-client-hub\modules\delta-paal\source\arm_uc_pal_delta_paal_original_reader.c:29:49: error: 'APPLICATION_ADDR' undeclared (first use in this function)
 #define MBED_CONF_APP_APPLICATION_START_ADDRESS APPLICATION_ADDR
                                                 ^
.\mbed-cloud-client\update-client-hub\modules\delta-paal\source\arm_uc_pal_delta_paal_original_reader.c:67:25: note: in expansion of macro 'MBED_CONF_APP_APPLICATION_START_ADDRESS'
     uint32_t appStart = MBED_CONF_APP_APPLICATION_START_ADDRESS;
                         ^
.\mbed-cloud-client\update-client-hub\modules\delta-paal\source\arm_uc_pal_delta_paal_original_reader.c:29:49: note: each undeclared identifier is reported only once for each function it appears in
 #define MBED_CONF_APP_APPLICATION_START_ADDRESS APPLICATION_ADDR
                                                 ^
.\mbed-cloud-client\update-client-hub\modules\delta-paal\source\arm_uc_pal_delta_paal_original_reader.c:67:25: note: in expansion of macro 'MBED_CONF_APP_APPLICATION_START_ADDRESS'
     uint32_t appStart = MBED_CONF_APP_APPLICATION_START_ADDRESS;
                         ^

[mbed] ERROR: "d:\python\python.exe" returned error.
       Code: 1
       Path: "D:\maclobdell\mbed-os-pelion-example"
       Command: "d:\python\python.exe -u D:\maclobdell\mbed-os-pelion-example\mbed-os\tools\make.py -t GCC_ARM -m CY8CPROTO_062_4343W --source . --build .\BUILD\CY8CPROTO_062_4343W\GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

It would be very useful if the build tools could catch these common mistakes and provide clear error messages to aid the developer.  Thanks!

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

trowbridgec, 2019-07-23:
I have run into this as well when the compiler can't find the bootloader file.

0xc0170, 2019-08-09:
cc @ARMmbed/mbed-os-tools

loverdeg-ep, 2019-08-09:
Be nice if we had some industry or ARM standard partitions and descriptions.

loverdeg, 2020-11-22:
Is this still valid? Is 5.13 supported?