ARMmbed/mbed-os #11100: UBLOX-NINA-B1 (nRF52832) with new mbedOS-5.13.1 running out of RAM

edablan, 2019-07-23:
### Description
Target: MTB_UBLOX_NINA_B1
Toolchain: IAR v8.32.2
Tools: exported using mbed-cli v1.10.0 to IAR
mbedOS:  5941d17183

We have been using the MTB_UBLOX_NINA_B1 running previous versions of mbedOS (< v5.13.0) using the soft device with no memory usage issues (running out of RAM). However, when we switched to mbedOS v5.13.0 we begin to run out of RAM fairly quickly. We cut all of our buffers in half (on our firmware application), and were still running out of memory (whereas before this was not an issue). We have buffers with a total of 10KB of RAM. Then I further reduced the buffers by a quarter of the original size. Finally at 2.75KB RAM usage of the buffers it built.  

I know the soft device has been replaced, but is the new Cordio LL on mbedOS v5.13 really that much larger? I haven't changed the stack sizes of my threads. Still relying on the defaults, are the default stack sizes significantly more in mbedOS 5.13.1 vs everything pre mbedOS v5.13.0? Is there something obvious I'm missing here?

### Issue request type

    [ X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-08-09:
cc @ARMmbed/mbed-os-pan Please review

woodsking2, 2019-08-12:
at nRF52832 with gcc, Cordio use about 21713byte static ram and 12500 byte dynamic.
You'd better to use soft device.

paul-szczepanek-arm, 2019-08-12:
The default configuration of the cordio stack needs changing to suit your needs. By default all the features are enabled.
Have a look in:
/features/FEATURE_BLE/mbed_lib.json
/features/FEATURE_BLE/targets/TARGET_CORDIO/mbed_lib.json
features/FEATURE_BLE/targets/TARGET_CORDIO_LL/mbed_lib.json
and disable/change what you don't need.
For memory usage you should have a look at:
```
ble::vendor::cordio::buf_pool_desc_t NRFCordioHCIDriver::get_buffer_pool_description()
```
Which allocated static ram memory pool

jrobeson, 2019-10-08:
shouldn't the default config have the equivalent configuration as the softdevice version and still work?

closed 2019-10-25 by ciarmcom