ARMmbed/mbed-os #11102: M2351: Compile failed on GCC

ccli8, 2019-07-24:
### Description

This is re-launch of #10650, but on latest master branch. To re-produce this error, just follows #10650.

**mbed-os**:
0b8ae1b098 Merge pull request #11084 from mtomczykmobica/IOTSTOR-832

```
Compile [  1.2%]: NanostackRfPhys2lp.cpp
Compile: arm-none-eabi-g++ -std=gnu++14 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m23 -mthumb -DAPPLICATION_ADDR=0x10040000 -DAPPLICATION_SIZE=0x40000 -DAPPLICATION_RAM_ADDR=0x30008000 -DAPPLICATION_RAM_SIZE=0x10000 -DMBED_ROM_START=0x10040000 -DMBED_ROM_SIZE=0x40000 -DMBED_ROM1_START=0x10040000 -DMBED_ROM1_SIZE=0x40000 -DMBED_RAM_START=0x30008000 -DMBED_RAM_SIZE=0x10000 -DMBED_RAM1_START=0x30008000 -DMBED_RAM1_SIZE=0x10000 -DARM_MATH_ARMV8MBL -DTARGET_FLASH_CMSIS_ALGO -DLPTICKER_DELAY_TICKS=3 -DDEVICE_PORTOUT=1 -DTARGET_LIKE_CORTEX_M23 -DDEVICE_PORTINOUT=1 -DDEVICE_SPI=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DTARGET_LIKE_MBED -DTARGET_M2351KIAAEES -DTOOLCHAIN_GCC_ARM -D__MBED__=1 -DDEVICE_TRNG=1 -DTARGET_RELEASE -DDEVICE_RTC=1 -DTARGET_CORTEX_M -DTARGET_NUMAKER_PFM_M2351 -D__MBED_CMSIS_RTOS_CM -DMBED_BUILD_TIMESTAMP=1563951789.448218 -D__CMSIS_RTOS -DDEVICE_STDIO_MESSAGES=1 -DTARGET_CORTEX -DDEVICE_INTERRUPTIN=1 -DTOOLCHAIN_GCC -DTARGET_M23 -DDEVICE_PORTIN=1 -DDEVICE_SPISLAVE=1 -DTARGET_NAME=NUMAKER_PFM_M2351 -DDEVICE_ANALOGOUT=1 -DMBED_FAULT_HANDLER_DISABLED -DDEVICE_I2C_ASYNCH=1 -DCOMPONENT_NSPE=1 -DDEVICE_I2C=1 -DDOMAIN_NS=1 -DDEVICE_SPI_ASYNCH=1 -DMBED_TZ_DEFAULT_ACCESS=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_PWMOUT=1 -DDEVICE_FLASH=1 -DCOMPONENT_FLASHIAP=1 -DDEVICE_SERIAL=1 -DDEVICE_I2CSLAVE=1 -DTARGET_M23_NS -DDEVICE_USTICKER=1 -DTARGET_NUVOTON -DDEVICE_MPU=1 -D__CORTEX_M23 -DDEVICE_ANALOGIN=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_M2351 -DDEVICE_LPTICKER=1 -DDEVICE_SLEEP=1 -DDEVICE_SERIAL_ASYNCH=1 @.\BUILD\NUMAKER_PFM_M2351\GCC_ARM\.includes_b74845699efa43f72db4d6a1063dd0c3.txt -include .\BUILD\NUMAKER_PFM_M2351\GCC_ARM\mbed_config.h -MD -MF BUILD\NUMAKER_PFM_M2351\GCC_ARM\mbed-os\components\802.15.4_RF\stm-s2lp-rf-driver\source\NanostackRfPhys2lp.d -o BUILD\NUMAKER_PFM_M2351\GCC_ARM\mbed-os\components\802.15.4_RF\stm-s2lp-rf-driver\source\NanostackRfPhys2lp.o .\mbed-os\components\802.15.4_RF\stm-s2lp-rf-driver\source\NanostackRfPhys2lp.cpp
[DEBUG] Return: 0
Compile [  1.4%]: FlashIAPBlockDevice.cpp
Compile: arm-none-eabi-g++ -std=gnu++14 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m23 -mthumb -DAPPLICATION_ADDR=0x10040000 -DAPPLICATION_SIZE=0x40000 -DAPPLICATION_RAM_ADDR=0x30008000 -DAPPLICATION_RAM_SIZE=0x10000 -DMBED_ROM_START=0x10040000 -DMBED_ROM_SIZE=0x40000 -DMBED_ROM1_START=0x10040000 -DMBED_ROM1_SIZE=0x40000 -DMBED_RAM_START=0x30008000 -DMBED_RAM_SIZE=0x10000 -DMBED_RAM1_START=0x30008000 -DMBED_RAM1_SIZE=0x10000 -DARM_MATH_ARMV8MBL -DTARGET_FLASH_CMSIS_ALGO -DLPTICKER_DELAY_TICKS=3 -DDEVICE_PORTOUT=1 -DTARGET_LIKE_CORTEX_M23 -DDEVICE_PORTINOUT=1 -DDEVICE_SPI=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DTARGET_LIKE_MBED -DTARGET_M2351KIAAEES -DTOOLCHAIN_GCC_ARM -D__MBED__=1 -DDEVICE_TRNG=1 -DTARGET_RELEASE -DDEVICE_RTC=1 -DTARGET_CORTEX_M -DTARGET_NUMAKER_PFM_M2351 -D__MBED_CMSIS_RTOS_CM -DMBED_BUILD_TIMESTAMP=1563951789.448218 -D__CMSIS_RTOS -DDEVICE_STDIO_MESSAGES=1 -DTARGET_CORTEX -DDEVICE_INTERRUPTIN=1 -DTOOLCHAIN_GCC -DTARGET_M23 -DDEVICE_PORTIN=1 -DDEVICE_SPISLAVE=1 -DTARGET_NAME=NUMAKER_PFM_M2351 -DDEVICE_ANALOGOUT=1 -DMBED_FAULT_HANDLER_DISABLED -DDEVICE_I2C_ASYNCH=1 -DCOMPONENT_NSPE=1 -DDEVICE_I2C=1 -DDOMAIN_NS=1 -DDEVICE_SPI_ASYNCH=1 -DMBED_TZ_DEFAULT_ACCESS=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_PWMOUT=1 -DDEVICE_FLASH=1 -DCOMPONENT_FLASHIAP=1 -DDEVICE_SERIAL=1 -DDEVICE_I2CSLAVE=1 -DTARGET_M23_NS -DDEVICE_USTICKER=1 -DTARGET_NUVOTON -DDEVICE_MPU=1 -D__CORTEX_M23 -DDEVICE_ANALOGIN=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_M2351 -DDEVICE_LPTICKER=1 -DDEVICE_SLEEP=1 -DDEVICE_SERIAL_ASYNCH=1 @.\BUILD\NUMAKER_PFM_M2351\GCC_ARM\.includes_b74845699efa43f72db4d6a1063dd0c3.txt -include .\BUILD\NUMAKER_PFM_M2351\GCC_ARM\mbed_config.h -MD -MF BUILD\NUMAKER_PFM_M2351\GCC_ARM\mbed-os\components\storage\blockdevice\COMPONENT_FLASHIAP\FlashIAPBlockDevice.d -o BUILD\NUMAKER_PFM_M2351\GCC_ARM\mbed-os\components\storage\blockdevice\COMPONENT_FLASHIAP\FlashIAPBlockDevice.o .\mbed-os\components\storage\blockdevice\COMPONENT_FLASHIAP\FlashIAPBlockDevice.cpp
[DEBUG] Return: 1
[DEBUG] Output: C:\Users\ccli8\AppData\Local\Temp\ccdUIrPo.s: Assembler messages:
[DEBUG] Output: C:\Users\ccli8\AppData\Local\Temp\ccdUIrPo.s:992: Error: instruction not supported in Thumb16 mode -- `subs r0,r1'
[DEBUG] Output: C:\Users\ccli8\AppData\Local\Temp\ccdUIrPo.s:1058: Error: instruction not supported in Thumb16 mode -- `adds r2,#1'
Traceback (most recent call last):
  File "C:\ccli8\iot\mbed\mbed-examples\mbed-os-example-blinky\mbed-os\tools\toolchains\mbed_toolchain.py", line 522, in compile_queue
    res['command']
  File "C:\ccli8\iot\mbed\mbed-examples\mbed-os-example-blinky\mbed-os\tools\toolchains\mbed_toolchain.py", line 649, in compile_output
    raise ToolException(stderr)
tools.utils.ToolException: C:\Users\ccli8\AppData\Local\Temp\ccdUIrPo.s: Assembler messages:
C:\Users\ccli8\AppData\Local\Temp\ccdUIrPo.s:992: Error: instruction not supported in Thumb16 mode -- `subs r0,r1'
C:\Users\ccli8\AppData\Local\Temp\ccdUIrPo.s:1058: Error: instruction not supported in Thumb16 mode -- `adds r2,#1'
```

@kjbracey-arm 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-08-12:
@kjbracey-arm Is this our issue we should address ?

kjbracey-arm, 2019-08-12:
Is this really related to #10650? Those instructions don't come from atomic_impl.h, I think. A text search for "subs" isn't showing any obvious candidates for the origin.

ccli8, 2019-08-13:
The error comes from below. After fixing it manually, mbed-os-example-blinky can compile successfully on M2351/GCC_ARM.

**mbed-os**
f8dc035ae4 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #11139 from sethitow/stm32f413-crash-capture

**platform/internal/mbed_atomic_impl.h**

<pre>

#elif defined __clang__ || defined __GNUC__
#define DO_MBED_LOCKFREE_NEWVAL_2OP_ASM(OP, Constants, M)       \
    __asm volatile (                                            \
        <b>".syntax unified\n\t"                                   \</b>
        "LDREX"#M "\t%[newValue], %[value]\n\t"                 \
        #OP       "\t%[newValue], %[arg]\n\t"                   \

</pre>

kjbracey-arm, 2019-08-13:
Oh, got it. It's just the missing `.syntax unified` again, on the new assembler fragment.

(Just back from holidays, so a bit rusty, and why this hasn't been dealt with for 3 weeks).

My fault - I failed to incorporate the #10705 fix into the new assembler that was pending in #10274.

I'll submit the fix.

PR #11208 (2019-08-15): Atomics: GCC fix for M23 (again)
### Description

Add another missing unified syntax directive. Was previously fixed in 03f1ac3ffd, but the same problem was not addressed in the pending PR that added the `NEWVAL_2OP` assembler.

Fixes #11102.

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

Targets 5.14, as fixing PR #10274.

### Reviewers

@ccli8
closed 2019-08-15 by adbridge