ARMmbed/mbed-os #11107: MQTT Network Socket Timeout Not Set

JRubis, 2019-07-25:
### Description

I believe that both of these methods in the MQTTNetwork class should be setting the socket timeout based on the timeout passed. They are not.

Current:

```
 int read(unsigned char* buffer, int len, int timeout) {
        return socket->recv(buffer, len);
    }

 int write(unsigned char* buffer, int len, int timeout) {
        return socket->send(buffer, len);
    }
```

Proposed Fix:

```
 int read(unsigned char* buffer, int len, int timeout) {
	socket->set_timeout(timeout);
        return socket->recv(buffer, len);
    }

 int write(unsigned char* buffer, int len, int timeout) {
	socket->set_timeout(timeout);
        return socket->send(buffer, len);
    }
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

AriParkkila, 2019-07-26:
@JRubis I can't find `MQTTNetwork` class in Mbed OS?

JRubis, 2019-07-26:
Hi @AriParkkila, sorry it's in the HelloMQTT example. Maybe this wasn't the right place for me to file a bug against an example versus the cores OS? 

https://os.mbed.com/teams/mqtt/code/HelloMQTT/file/e38aaf532823/MQTTNetwork.h/

0xc0170, 2019-08-09:
@JRubis Correct, please provide a pull request in the example.

I'll close this issue then.

closed 2019-08-09 by 0xc0170