ARMmbed/mbed-os #11109: [BLE][Cordio] How can I turn off the Central Role

EmbedEngineer, 2019-07-25:
### Description

Hi everyone,

I am currently trying to reduce the size of the Feature_BLE part in our project. Therefore I tried to find a way to turn off the Central and Observer Role since we are only a peripheral with a GATT Server. 
So far I tried to override the following settings in the mbed_app.json:
```
            "ble.ble-role-observer": false,
            "ble.ble-role-central": false,
            "ble.ble-feature-gatt-client": false,
```
Unfortunately I then get several errors in the Gap.tpp file when compiling:
```
[Error] Gap.tpp@222,1: no 'ble_error_t ble::interface::Gap<Impl>::createSync(ble::peer_address_type_t, const ble::address_t&, uint8_t, ble::slave_latency_t, ble::sync_timeout_t)' member function declared in class 'ble::interface::Gap<Impl>'
[Error] Gap.tpp@237,1: no 'ble_error_t ble::interface::Gap<Impl>::createSync(ble::slave_latency_t, ble::sync_timeout_t)' member function declared in class 'ble::interface::Gap<Impl>'
[Error] Gap.tpp@243,41: no 'ble_error_t ble::interface::Gap<Impl>::cancelCreateSync()' member function declared in class 'ble::interface::Gap<Impl>'
[Error] Gap.tpp@249,67: no 'ble_error_t ble::interface::Gap<Impl>::terminateSync(ble::periodic_sync_handle_t)' member function declared in class 'ble::interface::Gap<Impl>'
[Error] Gap.tpp@259,1: no 'ble_error_t ble::interface::Gap<Impl>::addDeviceToPeriodicAdvertiserList(ble::peer_address_type_t, const ble::address_t&, ble::advertising_sid_t)' member function declared in class 'ble::interface::Gap<Impl>'
[Error] Gap.tpp@273,1: no 'ble_error_t ble::interface::Gap<Impl>::removeDeviceFromPeriodicAdvertiserList(ble::peer_address_type_t, const ble::address_t&, ble::advertising_sid_t)' member function declared in class 'ble::interface::Gap<Impl>'
[Error] Gap.tpp@283,52: no 'ble_error_t ble::interface::Gap<Impl>::clearPeriodicAdvertiserList()' member function declared in class 'ble::interface::Gap<Impl>'
[Error] Gap.tpp@289,53: no 'uint8_t ble::interface::Gap<Impl>::getMaxPeriodicAdvertiserListSize()' member function declared in class 'ble::interface::Gap<Impl>'
```
So is there any way I can achieve to do that. Am I missing something?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

EmbedEngineer, 2019-07-25:
Ok I figured out that i have to add to add
```
            "ble.ble-feature-extended-advertising": false,
            "ble.ble-feature-periodic-advertising": false,
```
to make it work. In Addition I turned everything else off except from BLE_FEATURE_SECURITY.
This saves about 14kB. Is there any way to reduce the size further? At the moment the whole feature still needs about 170kB. Since we are trying to make a firmware update over BLE every kB that can be saved helps.

0xc0170, 2019-08-12:
cc @ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2019-08-12:
have a look in 
```
features/FEATURE_BLE/mbed_lib.json
```
it has all the options and documents what depends on what

closed 2019-10-25 by ciarmcom