ARMmbed/mbed-os #11111: Cellular & wait

pilotak, 2019-07-26:
### Description
When i run following code i'm **global up** in about 18s-19s
```cpp
#include "mbed.h"
#include "UBLOX_AT.h"
#include "CellularContext.h"

EventQueue eQueue(32 * EVENTS_EVENT_SIZE);
CellularContext *mdm;
CellularDevice *mdm_device;
DigitalOut led(PE_5);
Timer t;

int mdm_connect_id = 0;
nsapi_connection_status_t connection_status = NSAPI_STATUS_DISCONNECTED;
uint8_t registration_status = CellularNetwork::StatusNotAvailable;

bool mdmSetup();
class myUblox : public UBLOX_AT {
  public:
    explicit myUblox(FileHandle *fh);

    virtual nsapi_error_t hard_power_on();
    virtual nsapi_error_t hard_power_off();
    virtual nsapi_error_t soft_power_on();
    virtual nsapi_error_t soft_power_off();

  private:
    DigitalOut pwrOn;
    DigitalOut rst;
};

myUblox::myUblox(FileHandle *fh) :
    UBLOX_AT(fh),
    pwrOn(MDM_PWRON_pin, 0),
    rst(MDM_RST_pin, 1) {
}

nsapi_error_t myUblox::hard_power_on() {
    rst.write(0);
    ThisThread::sleep_for(1000);
    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::hard_power_off() {
    rst.write(1);

    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::soft_power_on() {
    pwrOn.write(1);
    ThisThread::sleep_for(150);
    pwrOn.write(0);
    ThisThread::sleep_for(100);

    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::soft_power_off() {
    pwrOn.write(1);
    ThisThread::sleep_for(1000);
    pwrOn.write(0);

    return NSAPI_ERROR_OK;
}

CellularDevice *CellularDevice::get_target_default_instance() {
    static UARTSerial serial(MDM_TX_pin, MDM_RX_pin, 115200);

#if defined(MDM_HW_FLOW)
    serial.set_flow_control(SerialBase::RTSCTS, MDM_RTS_pin, MDM_CTS_pin);
#else
    static DigitalOut rts(MDM_RTS_pin, 0);
#endif

    static myUblox device(&serial);
    return &device;
}
void mdmDisconnect() {
    debug("Disconnecting from network\n");
    mdm->disconnect();
}

void mdmConnect() {
    debug("MDM connect\n");
    nsapi_error_t ret = mdm->connect();

    debug("Connection: %i\n", ret);

    if (ret == NSAPI_ERROR_OK || ret == NSAPI_ERROR_IN_PROGRESS) {
        debug("Network OK\n");
        mdm_connect_id = 0;
        return;
    }

    if (ret == NSAPI_ERROR_NO_MEMORY) {
        debug("No memory, reseting MCU\n");
        NVIC_SystemReset();
    }

    debug("Connecting failed\n");
    mdm_connect_id = eQueue.call_in(5000, mdmConnect);

    if (!mdm_connect_id) {
        debug("Calling mdm connect failed, no memory\n");
    }
}

void mdmReconnect() {
    if (!mdm_connect_id) {
        debug("Reconnecting network\n");
        mdmConnect();

    } else {
        debug("mdm connect in progress\n");
    }
}

void mdmReset() {
    debug("Reseting MDM\n");
    mdm->attach(NULL);
    mdmDisconnect();
    mdm_device->shutdown();
    mdm_device->soft_power_off();
    mdm_device->hard_power_off();
    // mdm_device->stop();

    int qid = eQueue.call_in(5000, mdmSetup);

    if (!qid) {
        debug("Could not setup mdm\n");
    }
}

void mdmCb(nsapi_event_t type, intptr_t ptr) {
    if (type >= NSAPI_EVENT_CELLULAR_STATUS_BASE && type <= NSAPI_EVENT_CELLULAR_STATUS_END) {
        cell_callback_data_t *ptr_data = reinterpret_cast<cell_callback_data_t *>(ptr);
        cellular_connection_status_t event = (cellular_connection_status_t)type;

        if (ptr_data->final_try && ptr_data->error == NSAPI_ERROR_OK) {
            debug("Final_try\n");
            eQueue.call(mdmReset);
        }

        if (event == CellularDeviceReady && ptr_data->error == NSAPI_ERROR_OK) {
            debug("DeviceReady\n");

        } else if (event == CellularSIMStatusChanged && ptr_data->error == NSAPI_ERROR_OK) {
            debug("SIM: %i\n", static_cast<int>(ptr_data->status_data));

        } else if (event == CellularRegistrationStatusChanged && ptr_data->error == NSAPI_ERROR_OK) {
            if (registration_status != ptr_data->status_data && ptr_data->status_data != CellularNetwork::StatusNotAvailable) {
                debug("Registration: %i\n", ptr_data->status_data);
                registration_status = ptr_data->status_data;

                if (registration_status == CellularNetwork::RegisteredHomeNetwork ||
                        registration_status == CellularNetwork::RegisteredRoaming ||
                        registration_status == CellularNetwork::AlreadyRegistered) {
                    if (connection_status == NSAPI_STATUS_DISCONNECTED) {
                        int qid = eQueue.call(mdmReconnect);

                        if (!qid) {
                            debug("Calling mdm connect failed, no memory\n");
                        }
                    }

                } else {
                    if (connection_status != NSAPI_STATUS_CONNECTING) {
                        mdmCb(NSAPI_EVENT_CONNECTION_STATUS_CHANGE, NSAPI_STATUS_DISCONNECTED);
                    }
                }
            }

        } else if (event == CellularAttachNetwork && ptr_data->error == NSAPI_ERROR_OK) {
            if (ptr_data->status_data == CellularNetwork::Attached) {
                debug("Attached\n");

            } else {
                debug("Dettached\n");
            }

        } else if (event == CellularStateRetryEvent && ptr_data->error == NSAPI_ERROR_OK) {
            int retry = *(const int *)ptr_data->data;
            debug("Retry count: %i\n", retry);

        } else if (event == CellularCellIDChanged && ptr_data->error == NSAPI_ERROR_OK) {
            debug("Cell ID changed: %i\n", static_cast<int>(ptr_data->status_data));

        } else if (event == CellularRegistrationTypeChanged && ptr_data->error == NSAPI_ERROR_OK) {
            debug("RegistrationType changed: %i\n", static_cast<int>(ptr_data->status_data));
        }

    } else if (type == NSAPI_EVENT_CONNECTION_STATUS_CHANGE) {
        debug("Connection status: %i\n", ptr);

        if (connection_status != ptr) {
            if (ptr == NSAPI_STATUS_GLOBAL_UP) {
                debug("Global up: %lu\n", t.read_ms());
                t.stop();
            }

            connection_status = (nsapi_connection_status_t)ptr;
        }
    }
}

bool mdmSetup() {
    debug("Device setup\n");
    mdm = CellularContext::get_default_instance();

    if (mdm != NULL) {
        mdm_device = mdm->get_device();

        if (mdm_device != NULL) {
            mdm_device->hard_power_on();

            uint16_t timeout[8] = {1, 2, 4, 8, 16, 32, 64, 128};
            mdm_device->set_retry_timeout_array(timeout, 8);

            mdm->set_credentials(MBED_CONF_APP_APN);

#if defined(MBED_CONF_APP_SIM_PIN)
            mdm->set_sim_pin(MBED_CONF_APP_SIM_PIN);
#endif

            mdm->attach(&mdmCb);
            mdm->set_blocking(false);

            mdmReconnect();

            return true;

        } else {
            debug("No interface\n");
        }

    } else {
        debug("No device\n");
    }

    return false;
}

int main() {
    Thread eQueueThread;

    if (eQueueThread.start(callback(&eQueue, &EventQueue::dispatch_forever)) != osOK) {
        debug("eQueueThread error\n");
    }

    t.start();
    mdmSetup();

    while (1) {
    }
}
```
but when i add to empty `while` (under `main()`) i'm global up randomly in 82s, 110s, 26s, 77s it sometimes disconnects (`NSAPI_STATUS_DISCONNECTED`) and connects up during that time
```cpp
led = !led;
ThisThread::sleep_for(500);
```
There must be something going on with the wait, either the processor is not listening serial port while the other thread is sleeping or some bug in lib

Modem: uBlox SARA-G350
Processor: STM32F412ZG

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

AriParkkila, 2019-07-26:
@pilotak Can you share AT traces? You could also try to decrease the main thread priority or increase serial RX buffer e.g. `drivers.uart-serial-rxbuf-size: 4096´?

pilotak, 2019-07-26:
with `drivers.uart-serial-rxbuf-size: 4096` results: 18s, 20s, 44s, 84s, 16s, 154s
with `int main() { osThreadSetPriority(osThreadGetId(), osPriorityBelowNormal);` and `buf size: 4096` results: 18s, 154s, 18s, 82s

It's interesting with traces turned on, it's connected within 20s regardless above config all the time. Also what i noticed is when i get `CellularNetwork::Attached` and i don't get `NSAPI_STATUS_GLOBAL_UP` within a second i know it's going to take a long time, cause if it is immediately i know it will be done in 20s

```

Device setup
[INFO][CELL]: New CellularContext  (0x20005c68)
[INFO][CELL]: CellularContext set blocking 0
Reconnecting network
MDM connec
[INFO][CELL]: CellularContext connect
Connection status: 3
[DBG ][CEL: callback: 4106, err: 0, data: 4096onnection: 0
Network OK

[INFO][CELL]: art connecting (timeout 1000 ms)
[INFO][CELL]: AT TX ( 2): AT
NFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): AT
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><lOK<cr><ln>
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1)<cr>
[INFO][CELL]: AT RX ( 2): AT
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>K<cr><ln>
[DBG ][CELL]: Init => Device ready
90m[DBG ][CELL]: callback: 4106, err: 0, data: 4096
[DBG ][CELL]: AT flush
[INFO][CELL]: AT TX ( 4): ATE0
[INFO][CELL AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): AT
[INFO][CELL]: AT RX ( 3): E0<cr>
[INFO][CELL]: AT RX ( 5): <cr><ln>OK<cr>
[INFO][CELL]: AT RX 1): <ln>
[INFO][CELL]: AT TX ( 9): AT+CMEE=1
[INFO][CL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: AT TX ( 9): ACFUN=1
[INFO][CELL]: AT TX ( 1): <cr>
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: Device ready => In
[DBG ][CELL]: callback: 4106, err: 0, data: 4096
[INFO][CELL Start connecting (timeout 1000 ms)
[INFO][CELL]: AT TX ( 2): AT
39m[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: AT T( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <>OK<cr><ln>
[DBG ][CELL]: Init => Device ready
[DBG ][CELL callback: 4106, err: 0, data: 4096
[DBG ][CELL]: AT flush
[INF][CELL]: AT TX ( 4): ATE0
[INFO][CELL]: AT TX ( 1): <cr>0m
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <ln>OK<cr>n>
[INFO][CELL]: AT TX ( 9): AT+CMEE=1
[INFO][CELL]: AT TX ( 1): <c
[INFO][CELL]: AT RX ( 4): <cr><ln>OK
[INFO][CELL]: AT RX ( 2): <cr>n>
[INFO][CELL]: AT TX ( 9): AT+CFUN=1
[INFO][CELL]: AT TX ( 1): <>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: Mod ready
[DBG ][CELL]: callback: 4096, err: 0, data: 4096
[DBG ][CELL]: Device ready => SIM pin
[... 180 lines cut ...]
[INFO][CELL]: AT TX ( 1): "
[INFO][CELL]: AT TX 8): internet
[INFO][CELL]: AT TX ( 1): "
[INFO][CELL]:T TX ( 1): <cr>
[INFO][CELL]: AT RX ( 4): <cr><ln>OK
[INFO][CELL]: AT RX ( : <cr><ln>
[INFO][CELL]: AT TX (21): AT+UPSD=0,7,"0.0.0.0"
[INFO][CELL]:  TX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: AT TX12): AT+UPSD=0,6,
[INFO][CELL]: AT TX ( 1): 0
[INFO][CELL]AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: AT TX (12): AT+UDA=0,3
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): <cr><ln>OK<cr>
[INFO][CELL]: ARX ( 1): <ln>
[INFO][CELL]: AT TX (12): AT+UPSND=0,8
[INFO][CELL]: ATX ( 1): <cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>+UPS
[INFO][CELL]: ATX (11): ND: 0,8,1<cr><ln>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
Connection status: 1
Global up: 20025
```

AriParkkila, 2019-07-29:
@pilotak Could you try with `"cellular.debug-at": false` and/or `"trace-level": "TRACE_LEVEL_ERROR"` to get only the most relevant traces.

pilotak, 2019-07-29:
with both your suggestions set:
first try, perfect
```
Device setup
Reconnecting network
MDM connect
Connection status: 3
Connection: 0
Network OK
[ERR ][CELL]: AT error code 4
DeviceReady
SIM: 0
Retry count: 0
Retry count: 1
Retry count: 2
Retry count: 3
Registration: 1
Cell ID changed: 17069
Attached
Connection status: 1
Global up: 19294
```

second try - this is where the problem is
```
DeviceReady
SIM: 0
[ERR ][CELL]: Failed to set CEREG/CGREG URC's for registration
Retry count:
SIM: 0
Retry count: 0
Retry count: 1
Retry count: 2
Retry count: 3
Cell ID changed: 17069
Attached
Connection status: 2
[ERR ][CELL]: AT error code 4
Connection status: 2
Connection status: 2
[ERR ][CELL]: AT error code 100
Connection status: 2
Connection status: 1
Global up: 157290
```

pilotak, 2019-07-29:
third try, 115s
```
DeviceReady
SIM: 0
Retry count: 0
Registration: 2
Retry count: 1
Retry count: 2
Retry count: 3
Registration: 1
Cell ID changed: 17069
[ERR ][CELL]: AT error code 100
Attached
Connection status: 2
[ERR ][CELL]: AT error code 4
Connection status: 2
[ERR ][CELL]: AT error code 100
Connection status: 2
Connection status: 1
Global up: 115375
```

fourth
```
DeviceReady
SIM: 0
Retry count: 0
Registration: 2
Retry count: 1
Retry count: 2
Retry count: 3
Registration: 1
Cell ID changed: 17069
Attached
[ERR ][CELL]: AT error code 4
Connection status: 2
Connection status: 2
[ERR ][CELL]: AT error code 100
Connection status: 2
Connection status: 1
Global up: 115333
```

pilotak, 2019-07-29:
When there is **no code 4** it runs fine
```
DeviceReady
SIM: 0
Registration: 2
[ERR ][CELL]: AT error code 100
Retry count: 0
Retry count: 1
Retry count: 2
Retry count: 3
Registration: 1
Cell ID changed: 15358
Attached
Connection status: 1
Global up: 19255
```

AriParkkila, 2019-07-29:
@pilotak It looks like modem is not always ready after `soft_power_on`, could try adding some more sleep time there?

pilotak, 2019-07-29:
with more sleep time, AT errors disappeared, first boot (after applying power) is always connected in 20s. When it's global up i perform reset:
```cpp
mdm->attach(NULL);
mdm->disconnect();
mdm_device->shutdown();
mdm_device->soft_power_off();
mdm_device->hard_power_off();
mdm_device->stop();

eQueue.call_in(5000, mdmConnect);
```
than global up is in random interval again, always big delay after receiving `CellularNetwork::Attached`

pilotak, 2019-07-29:
## Update
i had no traces enables, errors are still there here is log
```
DeviceReady
SIM: 0
Retry count: 0
Retry count: 1
Registration: 1
Cell ID changed: 15358
[WARN][CELL]: Get network registration failed (type 1)!
Retry count: 2
[ERR ][CELL]: AT error code 3
Registration: -1
Attached
```
and then big delay

AriParkkila, 2019-07-30:
@pilotak `UBLOX_AT_CellularContext` (e.g. `activate_profile`) seems to have quite long timeouts while context is being activated.
@ARMmbed/team-ublox could have some idea how to debug this more?

fahim-ublox, 2019-07-30:
Internal ticket : PGM_NO_MBD-257

pilotak, 2019-07-30:
i did a bit of research, here is an updated code, modem should be turned off by AT command not by PWR_ON pin (i took the previous code from C027), thats for U2 series. It's working realiably with
- `TRACE_LEVEL_DEBUG`
- `mbed-trace.enable: true`
- `cellular.debug-at: true`

```cpp
#include "mbed.h"
#include "UBLOX_AT.h"
#include "CellularContext.h"
#if MBED_CONF_MBED_TRACE_ENABLE
    #include "CellularLog.h"
    #include "mbed-trace/mbed_trace.h"
    #define TRACE_GROUP  " APP"
#endif

EventQueue eQueue(32 * EVENTS_EVENT_SIZE);
CellularContext *mdm;
CellularDevice *mdm_device;
DigitalOut led(PE_5);
Timer t;

int mdm_connect_id = 0;
nsapi_connection_status_t connection_status = NSAPI_STATUS_DISCONNECTED;
uint8_t registration_status = CellularNetwork::StatusNotAvailable;

bool mdmSetup();
class myUblox : public UBLOX_AT {
  public:
    explicit myUblox(FileHandle *fh);

    virtual nsapi_error_t hard_power_on();
    virtual nsapi_error_t hard_power_off();
    virtual nsapi_error_t soft_power_on();
    virtual nsapi_error_t soft_power_off();

  private:
    DigitalOut pwrOn;
    DigitalOut rst;
};

myUblox::myUblox(FileHandle *fh) :
    UBLOX_AT(fh),
    pwrOn(MDM_PWRON_pin, 0),
    rst(MDM_RST_pin, 0) {
}

nsapi_error_t myUblox::hard_power_on() {
    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::hard_power_off() {
    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::soft_power_on() {
    pwrOn.write(1);
    ThisThread::sleep_for(150);
    pwrOn.write(0);
    ThisThread::sleep_for(1000);

    return NSAPI_ERROR_OK;
}

nsapi_error_t myUblox::soft_power_off() {
    ATHandler *at_cmd = mdm_device->get_at_handler();
    at_cmd->lock();
    at_cmd->cmd_start("AT+CPWROFF");
    at_cmd->cmd_stop();
    at_cmd->unlock();
    mdm_device->release_at_handler(at_cmd);
[... 201 lines cut ...]
    mbed_cellular_trace::mutex_wait_function_set(trace_wait);
    mbed_cellular_trace::mutex_release_function_set(trace_release);
}
#endif

int main() {
#if MBED_CONF_MBED_TRACE_ENABLE
    trace_init();
#endif{
    Thread eQueueThread;

    if (eQueueThread.start(callback(&eQueue, &EventQueue::dispatch_forever)) != osOK) {
        tr_err("eQueueThread error\n");
    }

    mdmSetup();

    while (1) {
        led = !led;
        ThisThread::sleep_for(500);
    }
}
```

and if i uncomment `mbed_trace_exclude_filters_set("CELL");` it stops being reliable and there is always big delay after `CellularNetwork::Attached`

pilotak, 2019-08-05:
@AriParkkila since ublox has not touched it for a week i did a research in libs and i managed to find why it's working with debug and why not without. It's all about delays - `printf` takes time to print all the stuff so the lib is not that fast (not sending next request to modem that fast) thus it's working propelly.

I found that `ATHandler` has an option to specify delay between AT command and actually this is a problem solver, but ublox doesn't use this by default
https://github.com/ARMmbed/mbed-os/blob/987533859add1a593784aefcd1e84e764818bd9e/features/cellular/framework/AT/ATHandler.h#L74

So i went ahead and updated the my code class to use timeout
```cpp
uint16_t myUblox::get_send_delay() const {
    return 10;
}
```

but it didn't work because it turns out that fist call to `ATHandler::get_instance` asks it's own `AT_CellularDevice::get_send_delay()` which gives `0` by default.

Then in next call to `get_instance()` it correctly asks `myUblox::get_send_delay()` but because `AT_CellularDevice::get_send_delay()` already gave timeout (`0`) it doesn't get updated with mine
https://github.com/ARMmbed/mbed-os/blob/fb7bbd25f1f4ef668f1c37b8b45861eafba6c027/features/cellular/framework/AT/AT_CellularDevice.cpp#L164-L165

AriParkkila, 2019-08-06:
@pilotak `get_send_delay` is a virtual function called at constructor, to be fixed #11121.

pilotak, 2019-08-26:
#11201 fixes this issue

closed 2019-08-26 by pilotak