ARMmbed/mbed-os #11121: Cellular: Virtual function call in constructor

ghseb, 2019-07-30:
### Description
`AT_CellularDevice`'s constructor calls `get_at_handler()` which calls `get_send_delay()`. `get_send_delay()` is virtual and therefore should not be called in a constructor. I observed that the method returned the default send delay of zero even if the method is overwritten (by `GEMALTO_CINTERION`) in my case.

Compiler is IAR 8.40 (which is not officially supported).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

pilotak, 2019-08-15:
@0xc0170  Why is this resolved? the issue is still there

0xc0170, 2019-08-23:
This one should: https://github.com/ARMmbed/mbed-os/commit/b849bc1e3de20d522ef50f90a0d83f20045685f2 ?
cc @mirelachirica

ghseb, 2019-08-26:
As far as i can see b849bc1 fixes the issue.

closed 2019-08-26 by 0xc0170