ARMmbed/mbed-os #11124: Auto-installing python failes on icetea dependencie trollius

Achilleshiel, 2019-07-30:
### Description
Today I deployed mbed on a system with no previous MBED or python use.
Running `mbed deploy` gave the following error:
```
[mbed] WARNING: You can install all missing modules by running "pip install -r requirements.txt" in "(...)/code/mbed-os"
```
However, running `pip install -r requirements.txt` fails and gives
```
ERROR: Could not find a version that satisfies the requirement trollius==1.0.4 (from pyshark-legacy->icetea<1.3,>=1.2.1->-r mbed-os/requirements.txt (line 21)) (from versions: 2.1.post2, 2.2, 2.2.post1)
ERROR: No matching distribution found for trollius==1.0.4 (from pyshark-legacy->icetea<1.3,>=1.2.1->-r mbed-os/requirements.txt (line 21))
```
I resolved this by just removing `icetea` from `requirements.txt` and just ran `pip install icetea`  
I'm currently running on branch `mbed-os-5.13` with SHA `5941d17183 (HEAD, tag: mbed-os-5.13.1, tag: latest) Merge pull request #11023 from ARMmbed/release-candidate`  

My gitlab CI-build always ran a succesful new build with `mbed deploy` but is now broken as well. So I experience the problem on Windows and on Linux.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

Reda-RM, 2019-07-30:
The same issue happens in many PRs here.

SeppoTakalo, 2019-07-30:
We are now investigating how to mitigate the problem. Original trollius package disappeared from PyPi repository.

madchutney, 2019-07-31:
trollius 1.0.4 has now been restored to PyPI

```
pip install --no-cache-dir trollius==1.0.4
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting trollius==1.0.4
  Downloading https://files.pythonhosted.org/packages/ca/21/74eedbeba3649fe9f1c807958c36a4cf54e775d263ff4cb3944e605d2328/trollius-1.0.4-py2-none-any.whl (105kB)
     |████████████████████████████████| 112kB 4.7MB/s 
Requirement already satisfied: futures in ./.local/share/virtualenvs/graham01-beul3flx/lib/python2.7/site-packages (from trollius==1.0.4) (3.3.0)
Installing collected packages: trollius
Successfully installed trollius-1.0.4
 bash-3.2$ pip list
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package    Version
---------- -------
futures    3.3.0  
mbed-cli   1.10.0 
pip        19.2.1 
setuptools 41.0.1 
trollius   1.0.4  
wheel      0.33.4 
bash-3.2$ 
```

SeppoTakalo, 2019-07-31:
As trollius has been restored, this is no longer an issue for us.

Thanks for reporting. We have also started some internal work to remove hard dependencies into trollius package.

closed 2019-07-31 by SeppoTakalo

naralasaimanoj, 2019-07-31:
I am still getting the same error.

Collecting trollius==1.0.4 (from -r /tmp/requirements.txt (line 54))
  Could not find a version that satisfies the requirement trollius==1.0.4 (from -r /tmp/requirements.txt (line 54)) (from versions: 2.1.post2)
No matching distribution found for trollius==1.0.4 (from -r /tmp/requirements.txt (line 54))