ARMmbed/mbed-os #11125: Test_LoRaWANTimer.start failed

supertoast, 2019-07-30:
### Description

mbed-os 5.13.1
gcc 9.1.0
mbed-cli 1.10.0

Running unittest for lorawan fail in test Test_LoRaWANTimer.start.
```
mbed-cli test --unittests -r lorawantimer
...
Unit tests built successfully.
Running tests...
UpdateCTestConfiguration  from :/home/stone/workspace/mbed-os/BUILD/unittests/DartConfiguration.tcl
Parse Config file:/home/stone/workspace/mbed-os/BUILD/unittests/DartConfiguration.tcl
   Site: UltraWeasel
   Build name: Linux-g++
UpdateCTestConfiguration  from :/home/stone/workspace/mbed-os/BUILD/unittests/DartConfiguration.tcl
Parse Config file:/home/stone/workspace/mbed-os/BUILD/unittests/DartConfiguration.tcl
Test project /home/stone/workspace/mbed-os/BUILD/unittests
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 29
    Start 29: features-lorawan-lorawantimer

29: Test command: /home/stone/workspace/mbed-os/BUILD/unittests/features-lorawan-lorawantimer
29: Test timeout computed to be: 1500
29: Running main() from gmock_main.cc
29: [==========] Running 6 tests from 1 test case.
29: [----------] Global test environment set-up.
29: [----------] 6 tests from Test_LoRaWANTimer
29: [ RUN      ] Test_LoRaWANTimer.constructor
29: [       OK ] Test_LoRaWANTimer.constructor (0 ms)
29: [ RUN      ] Test_LoRaWANTimer.get_current_time
29: [       OK ] Test_LoRaWANTimer.get_current_time (0 ms)
29: [ RUN      ] Test_LoRaWANTimer.get_elapsed_time
29: [       OK ] Test_LoRaWANTimer.get_elapsed_time (0 ms)
29: [ RUN      ] Test_LoRaWANTimer.init
29: [       OK ] Test_LoRaWANTimer.init (0 ms)
29: [ RUN      ] Test_LoRaWANTimer.start
29: mbed assertation failed: obj.timer_id != 0, file: /home/stone/workspace/mbed-os/features/lorawan/system/LoRaWANTimer.cpp, line 57 
1/1 Test #29: features-lorawan-lorawantimer ....***Exception: SegFault  0.19 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.19 sec

The following tests FAILED:
	 29 - features-lorawan-lorawantimer (SEGFAULT)
Errors while running CTest
make: *** [Makefile:107: test] Fehler 8

Unit test run failed.
[mbed] ERROR: "/home/stone/workspace/mbed-os-build/venv/bin/python2" returned error.
       Code: 2
       Path: "/home/stone/workspace/mbed-os"
       Command: "/home/stone/workspace/mbed-os-build/venv/bin/python2 /home/stone/workspace/mbed-os/UNITTESTS/mbed_unittest.py --regex lorawantimer --build ./BUILD/unittests"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

The following patch worked for me:
```
diff --git a/UNITTESTS/features/lorawan/lorawantimer/Test_LoRaWANTimer.cpp b/UNITTESTS/features/lorawan/lorawantimer/Test_LoRaWANTimer.cpp
index 3389de828b..1c38ca1770 100644
--- a/UNITTESTS/features/lorawan/lorawantimer/Test_LoRaWANTimer.cpp
+++ b/UNITTESTS/features/lorawan/lorawantimer/Test_LoRaWANTimer.cpp
@@ -17,6 +17,7 @@
 
 #include "gtest/gtest.h"
 #include "LoRaWANTimer.h"
+#include "equeue_stub.h"
 
 using namespace events;
 
@@ -71,6 +72,10 @@ TEST_F(Test_LoRaWANTimer, start)
 {
     timer_event_t ev;
     memset(&ev, 0, sizeof(ev));
+    struct equeue_event ptr;
+    equeue_stub.void_ptr = &ptr;
+    equeue_stub.call_cb_immediately = true;
+    object->init(ev, my_callback);
     object->start(ev, 10);
 }
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

kivaisan, 2019-08-01:
The problem is with MBED_ASSERT. Fix PR: https://github.com/ARMmbed/mbed-os/pull/11132

SeppoTakalo, 2019-08-01:
Fixed by #11132

closed 2019-08-01 by SeppoTakalo