ARMmbed/mbed-os #11128: (D)TLSSocket thread safety

vidavidorra, 2019-07-30:
### Description

How thread safe is the `DTLSSocket`? I'm interested in running the `recv` on a separate thread than where I'd do the `send`s from. I've tested it and it seems to work okay, but want to know whether this is officially supported and I can expect it to be stable in the long run.
I couldn't find any detailed information on this in the Mbed OS Reference for `Socket` and DTLS/TLS Socket but maybe this information is somewhere else?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-08-09:
cc @ARMmbed/mbed-os-ipcore

0xc0170, 2019-08-09:
It should be by default.

Drivers and other parts have this stated (thread safe/interrupt safe).

michalpasztamobica, 2019-08-09:
Hi @vidavidorra . Yes, (D)TLSSocket is thread-safe.
We even have a [greentea test](https://github.com/ARMmbed/mbed-os/blob/f8dc035ae4221626423cf99a4a216868cd599393/TESTS/netsocket/tls/tlssocket_echotest.cpp#L136) that does exactly what you said - it sends in one thread and receives in another thread.

kjbracey-arm, 2019-08-12:
There is a limit though - I don't believe the socket system can handle multiple readers or multiple writers - only 1 of each.

Having two blocking send calls or two blocking read calls from different threads on the same socket will confuse the blocking code, and I think it might actually cause system issues. (Although maybe only in edge cases - eg closing a socket only waking one blocking writer).

Doing such a thing would be generally a bit weird, but there are some conceivable uses - particularly in multi-CPU systems. You could have multiple threads reading and writing datagrams from one socket to do parallel processing.

Not really likely in Mbed OS-type systems though, so not currently supported.

(But the filehandle equivalent blocking `UARTSerial` and `select` does cope with this - it wouldn't be that unusual for two threads to both be blocking waiting for serial port TX buffer space.  The output may end up interleaved, but we don't want to actually deadlock or anything - all characters will come out).

vidavidorra, 2019-08-12:
Thanks for the responses guys. @kjbracey-arm I'm only planning to have one reader and one writer (each on a separate thread) so it should be allright. Good to know about the multiple blocking calls though!

closed 2019-08-12 by vidavidorra