ARMmbed/mbed-os #11129: target.restrict_size does not define APPLICATION_ADDR

vidavidorra, 2019-07-31:
### Description

I've used an unmanaged bootloader with a custom target in v5.11.6 and everything was working fine for me. I override the `target.restrict_size` parameter in the _mbed_app.json_ and use `POST_APPLICATION_ADDR` in the bootloader to start the application.
I've now updated to v5.13.0 and when I build the same, I get an error that `POST_APPLICATION_ADDR` is not defined.
The docs for 5.13.0 still mention that those should be created so I don't understand why they fail for me now.
- https://os.mbed.com/docs/mbed-os/v5.13/reference/bootloader-configuration.html
- https://os.mbed.com/docs/mbed-os/v5.13/tutorials/bootloader.html

I also tried to use `target.mbed_app_start` as described in the unmanaged bootloader section in the docs but with the same result. Neither of the following gets defined.

- `POST_APPLICATION_ADDR`
- `MBED_APP_START`
- `APPLICATION_ADDR`

Am I overlooking something here?

BTW I'm building this all using the CMake export and local GNU embedded toolchain.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

DrynnBavis, 2019-08-01:
When you build the project, do you see the size and offset messages in the log?
For example, mine's set to restrict up to 0x00010000 and so my offset message looks like this:
```
Using ROM regions application, post_application in this build.
  Region application: size 0x10000, offset 0x8000000
  Region post_application: size 0x170000, offset 0x8010000
```

vidavidorra, 2019-08-01:
I'm using the export (Mbed CLI 1.10.0) and the exporter doesn't show anything like that for the ROM regions.

See below the full _mbed_app.json_ I'm using. `target.mbed_app_start` was initially not there but I've added that trying things to get this working.

```json
{
  "target_overrides": {
    "*": {
      "target.restrict_size": "0x00020000",
      "target.mbed_app_start": "0x08020000",
      "target.components_add": ["DATAFLASH"],
      "dataflash.binary-size": 1,
      "platform.default-serial-baud-rate": 115200,
      "platform.stdio-baud-rate": 115200
    }
  }
}
```

### Export output

```shell
$ mbed export -i cmake_gcc_arm -m STM32F405RG --app-config ./submodules/mbed-system/bootloader/mbed_app.json --profile ./submodules/mbed-system/mbed_profiles/release.json --source ./submodules/mbed-system/bootloader/source/ --source ./submodules/mbed-system/custom_mbed_os_targets --source ./config --source ./submodules/mbed-system/source/config --source ./submodules/mbed-system/submodules --source ./submodules/mbed-system/source/progress_bar --build .
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/Users/jdebruijn/withthegrid/projects/cp-pole" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: pywin32
       You can install all missing modules by running "pip install -r requirements.txt" in "submodules/mbed-system/submodules/mbed-os"
       On Mac you might have to install packages as your user by adding the "--user" flag
---
Scan: source
Scan: custom_mbed_os_targets
Scan: config
Scan: config
Scan: submodules
Scan: progress_bar
```

vidavidorra, 2019-08-01:
Also, I have a different application in which I use the following two lines and that does show the ROM region usage as shown below. It, however doesn't show anything with a `post_application` region, which is what I need if I'm correct.

```json
      "target.mbed_app_start": "0x08000000",
      "target.mbed_app_size": "0x00020000",
```

```
Using ROM region application in this build.
  Region application: size 0x20000, offset 0x8000000
```

DrynnBavis, 2019-08-01:
>       "target.restrict_size": "0x00020000",
      "target.mbed_app_start": "0x08020000",

These two conflict, using the `restrict_size` parameter uses the "managed bootloader" method. This will handle the start address of your bootloader to be the start of ROM and then your main app will be programmed to start of ROM + the `restrict_size` value. This value is accessible in your cpp code with the macro `POST_APPLICATION_ADDR`. You could if you want specify a different start point for your bootloader, but then you'd be using the "unmanaged bootlaoder" method. If that's the case, you need to restrict the size of your bootloader using `mbed_app_size` as you've discovered.  Read more about bootloader configuration [here](https://os.mbed.com/docs/mbed-os/v5.9/tools/configuring-tools.html)

Using the **unmanaged** bootloader method will define two macros you can access in cpp:
- `APPLICATION_ADDR`
- `APPLICATION_SIZE`
which have the values you'd expect in them.

Using the **managed** bootloader will not define these, but instead it will define
- `POST_APPLICATION_ADDR`

If you'd like to continue using the unmanaged bootloader, but you'd like the value of `POST_APPLICATION_ADDR` for readability, I'd recommend doing something like this:
```
const size_t POST_APPLICATION_ADDR = APPLICATION_ADDR + APPLICATION_SIZE;
```

DrynnBavis, 2019-08-01:
If you aren't getting `POST_APPLICATION_ADDR` to be defined and you're using a managed bootloader, this is likely because the target processor you're compiling for doesn't have it's sectors defined. 

What processor are you using? Is this a custom target or one supported by mbed?

vidavidorra, 2019-08-01:
Yeah, I'm using the unmanaged bootloader. Previously (v5.11.6) I did get the `POST_APPLICATION_ADDR` but I guess something has changed later on. When I **only** set the `mbed_app_start` and `mbed_app_size` I do get those (`APPLICATION_ADDR` and `APPLICATION_SIZE`) defined so I think I'll just use this then. Will be fine for my case since the bootloader and the application are consecutive in flash in my case at least.

```c++
#define POST_APPLICATION_ADDR ((APPLICATION_ADDR) + (APPLICATION_SIZE))
```

@DrynnBavis thanks for your help!

closed 2019-08-01 by vidavidorra