ARMmbed/mbed-os #11140: Region Merging is not padding correctly

DrynnBavis, 2019-08-01:
### Description
I've got a managed bootloader application of which I've set the `target.restrict_size` to a value of `0x40000`. When I use this binary as a bootloader for my main application, I get the following in my build logs:
```
Using ROM regions bootloader, application in this build.
  Region bootloader: size 0x10000, offset 0x8000000
  Region application: size 0x170000, offset 0x8010000
Compile [100.0%]: mbed_application.c
```
This is problematic because the bootloader calls `mbed_app_start(POST_APPLICATION_ADDR)` which has a value of `0x8040000`. Essentially, the linker is placing my main program at `0x8010000` while the bootloader is actually booting into `0x8040000`. The behaviour of this is my application just "hangs" after the bootloader completes. 

Note that I've also printed the address, sp, and pc, before booting the assembly code within `start_new_application()` is called, and I got the following results:
```
calling start_new_application
address -> 0x08010000
sp ------> 0x20050000
pc ------> 0x080132DD
```

Is this a bug on mbed's side? Or do I need to do something in my main program's config to pad out until `0x8040000`?

### Issue request type
Target: custom MCU with STM32F413RH processor
Toolchain: GCC_ARM 8.2.1
Tool: mbed-cli
Vers: cfa7938 (HEAD, tag: mbed-os-5.12.2)

    [x] Question
    [ ] Enhancement
    [ ] Bug

AGlass0fMilk, 2019-08-01:
I can't comment on the padding issue at the moment -- can you post both of your 'mbed_app.json' files for the bootloader and the application? Also post the 'custom_target.json' if you have one.

As for `sp` and `pc`, the first one is the initial **stack pointer** and the second one is the initial **program counter** (address of next instruction to be executed). In this case, it's pointing to the Reset Vector of your application. This is the real entry point of your application and does all the hardware/system/RTOS initialization before jumping to your `main` function.

See this diagram below showing the reset vector format on Cortex M4 devices:

[image]
(Page 37 of [this file](http://infocenter.arm.com/help/topic/com.arm.doc.dui0553b/DUI0553.pdf))

So in your second case, the bootloader/application merging seems to be working -- the `sp` and `pc` values are reasonable, indicating it really is the vector table of your application binary. **However,** there is a problem!

The reason your program is hanging is because the initial program counter pointer is **unaligned** -- that is the value `0x080132DD` is not word-aligned (since it's a 32-bit processor, must be divisible by 4). `0x080132DD` is an odd number. Attempting to execute an unaligned instruction is illegal in CM4 processors and will cause a HardFault/MemoryManagement Fault.

Unfortunately this kind of bug pops up from time to time, so it's good to know the root cause of this issue.

This is certainly a problem with the online compiler tools...

Until someone from Mbed addresses this issue I would recommend trying to get set up with Mbed-CLI and building offline, that way you have a lot more control over the build and various tool/software versions.

DrynnBavis, 2019-08-01:
> can you post both of your 'mbed_app.json' files for the bootloader and the application? Also post the 'custom_target.json' if you have one.

@AGlass0fMilk I've edited my OP to include the 3 files you asked for. Just as a reminder, the `device_name` listed in my target definition isn't the processor I'm actually using (since my processor - `STM32F413RH` - doesn't seem to have support from mbed), so instead I'm using `STM32F413ZHTx` since it has sectors defined in `arm_pack_manager/index.json` that _seem_ to be similar to my processor's ROM mapping. Please see #11120 for more info on that.

I had a feeling that `0x080132DD` wasn't a valid address. I don't see what I'm doing on my end wrong that's causing this though. Recall from `mbed_start_application()` from `mbed_application.c` that the definition of `pc` is
```
pc = *((void **)address + 1);
```
To print this, I was c-style casting it to a char pointer like so
```
printf("pc ------> 0x%08X\r\n", (char*)pc);
```
Maybe I'm just printing this wrong? Or does the logic for `pc` make sense? My interpretation is, we're casting `address` to a double void pointer, incrementing the address of this pointer by 1 (whatever that means, please enlighten me here!) and then finally de referencing it back to a single star void pointer.

loverdeg-ep, 2019-08-01:
I recall ST parts having memory regions all over the place versus 1 contiguous flash region.

[image]

Gut feel is you should confirm those sectors that seem to be similar.

DrynnBavis, 2019-08-01:
> Gut feel is you should confirm those sectors that seem to be similar.
@loverdeg-ep tagged you on the other issue with an update to my understanding of the sectors for this chip.

AGlass0fMilk, 2019-08-01:
To verify the actual value programmed into the device you can use a hex editor to view the binary.

Try opening the binary up with something like https://hexed.it/?hl=en and take a screenshot.

DrynnBavis, 2019-08-01:
> To verify the actual value programmed into the device for the initial program counter/reset vector you can use a hex editor to view the binary.

What address is this exactly? Sorry. I've got the file open in HexEd.it just not sure what address you want to see.

AGlass0fMilk, 2019-08-01:
The beginning of the application. So whatever that address is that the bootloader jumps to.

DrynnBavis, 2019-08-01:
> The beginning of the application. So whatever that address is that the bootloader jumps to.

[image]
This is the hex at the offset of `0x00020000`, the expected offset of the main program from the `main.bin` compiled file of the master application with the merged bootloader in it.

DrynnBavis, 2019-08-01:
@AGlass0fMilk Something else interesting, I've changed the target on my bootloader and main application to an older MCU I had on hand that I previously had success with, and this was the `sp` and `pc` print:
```
calling start_new_application
address -> 0x08020000
sp ------> 0x20030000
pc ------> 0x08024DC5
```

Again, here the program counter is pointing to an odd number, and thus seemingly unaligned. **However** the program does seem to be working fine. My blinky application successfully booted from the bootloader. Nothing changed but the target.

DrynnBavis, 2019-08-05:
Just a bump for the new week
@AGlass0fMilk what do you think of this from @40Grit:

> In the past @AGlass0fMilk was getting a hardfault after booting mcuboot. i beleive it was due to an invalid vector table relocation?
vtor is the register to confirm.

How would I go about confirming that vtor is indeed pointing to the correct address of the vector table? I guess I'm asking how would I check the value in vtor after the bootloader boots, and then where would I find what the expected address of the vector table should be?

loverdeg-ep, 2019-08-06:
@DrynnBavis 
@AGlass0fMilk is travelling currently.

The vector table is at address 0 by default. Where it gets relocated to must be aligned correctly.
The first thing the processor does after power on reset is load the value contained at the 0th entry of the table into the stack pointer and then executes the reset vector.

As I recall mbed generally ends up relocating the vector table to somewhere in SRAM

See the following from http://infocenter.arm.com/help/topic/com.arm.doc.dui0553b/DUI0553.pdf

[image]

loverdeg-ep, 2019-08-06:
@kjbracey-arm may or may not be able to cut to the heart of the topic for you, upon returning from leave.

DrynnBavis, 2019-08-07:
@loverdeg-ep thanks for the response. I was looking at this datasheet yesterday, and I'm pretty convinced it is indeed being relocated to SRAM. I'm just frustrated in trying to figure out how to debug this properly. 

What I've learned is that the bootloader has it's own v table and after finishing (i.e. I call `mbed_app_start(POST_APPLICATION_ADDR);`, something happens and I think the `sp` is supposed to point at the vector table of my new application where it's now been offset to some place in SRAM. Ideally at this point a reset will occur and initialisation work the processor needs to do will be performed to configure RTOS and such before the application starts.

What exactly is expected to happen between my calling of `mbed_app_start(POST_APPLICATION_ADDR);` from the bootloader and the start of my main application? Knowing this would help a lot, if anyone knows.

40Grit, 2019-08-07:
@drynnbavis are you able to step debug?

If so, have you loaded the .elf files for both your bootloader and application?

DrynnBavis, 2019-08-07:
@40Grit I haven't tried yet. For some reason when I build with the debug profile I get this crash from `__disable_irq()` within `mbed_start_application()` from `mbed_application.c`:

```
++ MbedOS Error Info ++
Error Status: 0x8001012F Code: 303 Module: 1
Error Message: Error - writing to a file in an ISR or critical section

Location: 0x8006BE5
Error Value: 0x1
Current Thread: main  Id: 0x200012B0 Entry: 0x8007AFF StackSize: 0x1000 StackMem: 0x20001C18 SP: 0x200028F8 
For more info, visit: https://mbed.com/s/error?error=0x8001012F&tgt=TRACK_PILOT2
-- MbedOS Error Info --
```

AGlass0fMilk, 2019-08-07:
Sounds like it’s trying to write debug/crash dump info to the `stdout` file handle during a critical section.

DrynnBavis, 2019-08-07:
Hey so I removed the print statements and as you suspected it solved the crash I was seeing on the debug profile. Finally, I was able to backtrace the crash and I found this:

```
(gdb) bt
#0  0x08006ce4 in write (fildes=2, buf=0x2004fec8, length=0) at libs/mbed-os/platform/mbed_retarget.cpp:677
#1  0x0800602c in mbed_error_puts (str=0x2004fec8 "\n++ MbedOS Fault Handler ++\n\nFaultType: ")
    at libs/mbed-os/platform/mbed_board.c:97
#2  0x08006010 in mbed_error_vprintf (format=0x8018de4 "\n++ MbedOS Fault Handler ++\n\nFaultType: ", arg=...)
    at libs/mbed-os/platform/mbed_board.c:71
#3  0x08005fca in mbed_error_printf (format=0x8018de4 "\n++ MbedOS Fault Handler ++\n\nFaultType: ")
    at libs/mbed-os/platform/mbed_board.c:55
#4  0x08005aee in mbed_fault_handler (fault_type=16, mbed_fault_context_in=0x8018de0 <mbed_fault_context>)
    at libs/mbed-os/platform/TARGET_CORTEX_M/mbed_fault_handler.c:48
#5  0x08005ad6 in Fault_Handler () at except.S:187
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

Not at all sure if there's anything helpful here, but I feel that this line:
`0x08005aee in mbed_fault_handler (fault_type=16, mbed_fault_context_in=0x8018de0 <mbed_fault_context>)` might help give us some context in what's going on here. I just don't know how to interpret it.

AGlass0fMilk, 2019-08-07:
Can you check the debug serial output and post that here? That stack trace is just showing there was a call to the Mbed fault handler that prints out debug information.

loverdeg-ep, 2019-08-07:
@DrynnBavis Are you printing to uart, SWO, or semihosting?

loverdeg-ep, 2019-08-07:
@DrynnBavis Also, if you can step debug down at the assembly I would see if their is a particular instruction that your fault is occurring on.

DrynnBavis, 2019-08-07:
@AGlass0fMilk The fault error never actually prints to serial. @loverdeg-ep I'm using UART.

DrynnBavis, 2019-08-07:
Just stepping through `mbed_start_application()` now, found these values of the core registers after assigning the new values to `sp` and `pc` from the address:
```
r0             0x8020000           134348800
r1             0xe000e100          3758153984
r2             0xe000e000          3758153728
r3             0x20050000          537198592
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0xffffffff          4294967295
sp             0x20002bd0          0x20002bd0 <_main_stack+4024>
lr             0x8005dbd           134241725
pc             0x8005dd2           0x8005dd2 <mbed_start_application+50>
xpsr           0x610e0000          1628307456
msp            0x2004ffb0          537198512
psp            0x20002bd0          536882128
primask        0x1                 1
basepri        0x0                 0
faultmask      0x0                 0
control        0x6                 6
fpscr          0x0                 0
```

loverdeg-ep, 2019-08-07:
Check value of VTOR as well.

DrynnBavis, 2019-08-07:
@loverdeg-ep I read VTOR while I was stepping through `powerdown_scb()` and found VTOR at that time to have a value of 0x08020000 (same as the value of `POST_APPLICATION_ADDR`)

DrynnBavis, 2019-08-07:
@AGlass0fMilk @loverdeg-ep 
Found the point of breaking to be right after this line:
```
Breakpoint 2, start_new_application (sp=0x20050000, pc=0x802635d) at libs/mbed-os/platform/mbed_application.c:175
175         __asm volatile(
(gdb) info registers
r0             0x20050000          537198592
r1             0x802635d           134374237
r2             0xe000e000          3758153728
r3             0x802635d           134374237
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0xffffffff          4294967295
sp             0x20002bc8          0x20002bc8 <_main_stack+4016>
lr             0x8005f03           134242051
pc             0x8005ff2           0x8005ff2 <start_new_application+6>
xpsr           0x10e0000           17694720
msp            0x2004ffb0          537198512
psp            0x20002bc8          536882120
primask        0x1                 1
basepri        0x0                 0
faultmask      0x0                 0
control        0x6                 6
fpscr          0x0                 0
(gdb) info locals
No locals.
(gdb) print sp
$1 = (void *) 0x20050000
(gdb) print pc
$2 = (void *) 0x802635d
```
after stepping one more time I get the crash:
```
(gdb) step

Program received signal SIGTRAP, Trace/breakpoint trap.
0x08006ca4 in write (fildes=2, buf=0x2004fec8, length=0) at libs/mbed-os/platform/mbed_retarget.cpp:678
678             errno = EBADF;
(gdb) 
```

loverdeg-ep, 2019-08-07:
Are you using command line GDB?

DrynnBavis, 2019-08-07:
> Are you using command line GDB?

yes

loverdeg-ep, 2019-08-07:
I don't know off hand but you'll need the disassembly.

Might need to make it an output of your build.

loverdeg-ep, 2019-08-07:
*disassembly listing

loverdeg-ep, 2019-08-07:
https://sourceware.org/gdb/onlinedocs/gdb/Machine-Code.html

^ may or may not help.

DrynnBavis, 2019-08-07:
> https://sourceware.org/gdb/onlinedocs/gdb/Machine-Code.html
> 
> ^ may or may not help.

```
(gdb) disassemble
Dump of assembler code for function start_new_application:
   0x08005fec <+0>:     sub     sp, #8
   0x08005fee <+2>:     str     r0, [sp, #4]
   0x08005ff0 <+4>:     str     r1, [sp, #0]
=> 0x08005ff2 <+6>:     ldr     r3, [sp, #4]
   0x08005ff4 <+8>:     ldr     r1, [sp, #0]
   0x08005ff6 <+10>:    movs    r2, #0
   0x08005ff8 <+12>:    msr     CONTROL, r2
   0x08005ffc <+16>:    mov     sp, r3
   0x08005ffe <+18>:    msr     PRIMASK, r2
   0x08006002 <+22>:    bx      r1
   0x08006004 <+24>:    nop
   0x08006006 <+26>:    add     sp, #8
   0x08006008 <+28>:    bx      lr
End of assembler dump.
```

DrynnBavis, 2019-08-07:
Found the actual precise step of machine code that fails here, check this out:
```
(gdb) disassemble
Dump of assembler code for function start_new_application:
   0x08005fec <+0>:     sub     sp, #8
   0x08005fee <+2>:     str     r0, [sp, #4]
   0x08005ff0 <+4>:     str     r1, [sp, #0]
   0x08005ff2 <+6>:     ldr     r3, [sp, #4]
   0x08005ff4 <+8>:     ldr     r1, [sp, #0]
   0x08005ff6 <+10>:    movs    r2, #0
   0x08005ff8 <+12>:    msr     CONTROL, r2
=> 0x08005ffc <+16>:    mov     sp, r3
   0x08005ffe <+18>:    msr     PRIMASK, r2
   0x08006002 <+22>:    bx      r1
   0x08006004 <+24>:    nop
   0x08006006 <+26>:    add     sp, #8
   0x08006008 <+28>:    bx      lr
End of assembler dump.
(gdb) stepi
0x08005ffe in start_new_application (sp=<error reading variable: Cannot access memory at address 0x20050004>, 
    pc=<error reading variable: Cannot access memory at address 0x20050000>)
    at libs/mbed-os/platform/mbed_application.c:175
175         __asm volatile(
(gdb) 
```

40Grit, 2019-08-07:
What value is in r3?

DrynnBavis, 2019-08-07:
> What value is in r3?

@40Grit at that same step:
```
(gdb) info registers
r0             0x20030000          537067520
r1             0x802635d           134374237
r2             0x0                 0
r3             0x20030000          537067520
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0xffffffff          4294967295
sp             0x2004ffb0          0x2004ffb0
lr             0x8005f01           134242049
pc             0x8005ffc           0x8005ffc <start_new_application+16>
xpsr           0x410e0000          1091436544
msp            0x2004ffb0          537198512
psp            0x20002bc8          536882120
primask        0x1                 1
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
fpscr          0x0                 0
```

DrynnBavis, 2019-08-07:
> What value is in r3?

From the same step point:
```
(gdb) info registers
r0             0x20050000          537198592
r1             0x802635d           134374237
r2             0x0                 0
r3             0x20050000          537198592
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0xffffffff          4294967295
sp             0x2004ffb0          0x2004ffb0
lr             0x8005f03           134242051
pc             0x8005ffc           0x8005ffc <start_new_application+16>
xpsr           0x410e0000          1091436544
msp            0x2004ffb0          537198512
psp            0x20002bc8          536882120
primask        0x1                 1
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
fpscr          0x0                 0
```

DrynnBavis, 2019-08-07:
Just realised something: why would the stack pointer ever be pointing to 0x20050000? My chip only has 328K SRAM, so that the end of the accessible SRAM address...

from the datasheet:
[image]

40Grit, 2019-08-07:
Show the memory map from the data sheet for good measure.

DrynnBavis, 2019-08-07:
> Show the memory map from the data sheet for good measure.

for sure:
[image]

40Grit, 2019-08-07:
I meant for the particular you are using

DrynnBavis, 2019-08-07:
> I meant for the particular you are using

update comment to show ROM usage of F413/F423 chips

DrynnBavis, 2019-08-07:
Is there a chance that the linker script for this chip is bugged? Would it be worth getting ST involved?

40Grit, 2019-08-08:
SP increases downward. SRAM ends at the address before what SP is set to.

Read up on what happens if stack pointer is set outside the range of writable addresses.

DrynnBavis, 2019-08-08:
> SP increases downward. SRAM ends at the address before what SP is set to.
> 
> Read up on what happens if stack pointer is set outside the range of writable addresses.

OK so something I missed is actually the end of the error warning (github hides it, you have to scroll horizontally a bit) but it's this line here:
```
sp=<error reading variable: Cannot access memory at address 0x20050004>, 
    pc=<error reading variable: Cannot access memory at address 0x20050000>
```

It's interesting because the value moved into SP was **not** `0x20050004`, but rather `0x20050000`. The region of `0x20050000` -> `0x3FFFFFFF` is reserved on my chip, AKA non-writeable. So this error indeed makes sense. Should I try hard coding the SP to be `0x2004FFFB` perhaps? Why is it trying to access `0x20050004` if it does indeed increase downward? @40Grit

DrynnBavis, 2019-08-08:
@40Grit So something interesting,

I have a DISCO_F413ZH in the office and I was able to flash my bootloader + main application package on it and it works just fine. The device name defined in this disco board is the STM32F413**ZH** where the processor used on my custom board is a STM32F413**RH**. I believe there should be virtually no difference between the two processors other than their physical packages...

What's so different about my board?

loverdeg-ep, 2019-08-08:
@DrynnBavis 
Can you use the .svd file for these parts, and dump all the registers and peripherals?

Then do a diff on the output
Do this just before the failure and maybe some other select places.

DrynnBavis, 2019-08-08:
> @DrynnBavis
> Can you use the .svd file for these parts, and dump all the registers and peripherals?
> 
> Then do a diff on the output
> Do this just before the failure and maybe some other select places.

@loverdeg-ep not entirely sure what an SVD does, but it seems to be some sort of an XML-like file that shows info about registers and such? Anyway I've return to the point before crashing and I've tried writing `svd` in my GDB debugger but it's not recognised. How do I print an svd file?

DrynnBavis, 2019-08-08:
@loverdeg-ep OK, got this SVD loaded into gdb: https://github.com/posborne/cmsis-svd/blob/master/data/STMicro/STM32F413.svd
after typing `svd` I get this, is this what you wanted?
```
Available Peripherals:
        AES:            Advanced encryption standard hardware accelerator
        TIM1:           Advanced-timers
        TIM8:           Advanced-timers
        ADC1:           Analog-to-digital converter
        TIM7:           Basic timers
        TIM6:           Basic timers
        CAN1:           Controller area network
        CAN2:           Controller area network
        CAN3:           Controller area network
        CRC:            Cryptographic processor
        DBG:            Debug support
        DFSDM2:         Digital filter for sigma delta modulators
        DFSDM1:         Digital filter for sigma delta modulators
        DAC:            Digital-to-analog converter
        DMA1:           DMA controller
        DMA2:           DMA controller
        EXTI:           External interrupt/event controller
        FMPI2C:         fast-mode Inter-integrated circuit
        FLASH:          FLASH
        TIM12:          General purpose timers
        TIM13:          General purpose timers
        TIM14:          General purpose timers
        TIM9:           General purpose timers
        TIM3:           General purpose timers
        TIM4:           General purpose timers
        TIM2:           General purpose timers
        GPIOF:          General-purpose I/Os
        GPIOH:          General-purpose I/Os
        GPIOE:          General-purpose I/Os
        GPIOC:          General-purpose I/Os
        GPIOG:          General-purpose I/Os
        GPIOD:          General-purpose I/Os
        GPIOB:          General-purpose I/Os
        GPIOA:          General-purpose I/Os
        TIM10:          General-purpose-timers
        TIM11:          General-purpose-timers
        TIM5:           General-purpose-timers
        IWDG:           Independent watchdog
        I2C2:           Inter-integrated circuit
        I2C1:           Inter-integrated circuit
        I2C3:           Inter-integrated circuit
        LPTIM:          Low power timer
        NVIC:           Nested Vectored Interrupt Controller
        PWR:            Power control
        QUADSPI:        QuadSPI interface
        RNG:            Random number generator
        RTC:            Real-time clock
        RCC:            Reset and clock control
        SDIO:           Secure digital input/output interface
        SAI:            Serial audio interface
        SPI5:           Serial peripheral interface
        SPI4:           Serial peripheral interface
        SPI1:           Serial peripheral interface
        SPI2:           Serial peripheral interface
        SPI6:           Serial peripheral interface
        SPI3:           Serial peripheral interface
        SYSCFG:         System configuration controller
        USART3:         Universal synchronous asynchronous receiver transmitter
        UART4:          Universal synchronous asynchronous receiver transmitter
        USART6:         Universal synchronous asynchronous receiver transmitter
        USART2:         Universal synchronous asynchronous receiver transmitter
        UART5:          Universal synchronous asynchronous receiver transmitter
        UART7:          Universal synchronous asynchronous receiver transmitter
        UART8:          Universal synchronous asynchronous receiver transmitter
        UART10:         Universal synchronous asynchronous receiver transmitter
        USART1:         Universal synchronous asynchronous receiver transmitter
        UART9:          Universal synchronous asynchronous receiver transmitter
        OTG_FS_DEVICE:  USB on the go full speed
        OTG_FS_HOST:    USB on the go full speed
        OTG_FS_PWRCLK:  USB on the go full speed
        OTG_FS_GLOBAL:  USB on the go full speed
        WWDG:           Window watchdog
        FSMC:           Flexible static memory controller
        ADC_Common:     ADC common registers
        FPU:            Floting point unit
        MPU:            Memory protection unit
        STK:            SysTick timer
        SCB:            System control block
        NVIC_STIR:      Nested vectored interrupt controller
        FPU_CPACR:      Floating point unit CPACR
        SCB_ACTRL:      System control block ACTLR
```

40Grit, 2019-08-08:
might as well dump them all then just post the diff between them for both parts.

DrynnBavis, 2019-08-09:
Took a while but here it is:
Nothing changed except for these two peripherals.

Before crashing ->
```
Registers in SCB:
        CPUID:                 1091551809  CPUID base register
        ICSR:                     4194304  Interrupt control and state register
        VTOR:                   134348800  Vector table offset register
        AIRCR:                 -100334848  Application interrupt and reset control register
        SCR:                            0  System control register
        CCR:                          512  Configuration and control register
        SHPR1:                          0  System handler priority registers
        SHPR2:                          0  System handler priority registers
        SHPR3:                          0  System handler priority registers
        SHCRS:                          0  System handler control and state register
        CFSR_UFSR_BFSR_MMFSR:           0  Configurable fault status register
        HFSR:                           0  Hard fault status registerAES
        MMFAR:                 -536809992  Memory management fault address register
        BFAR:                  -536809992  Bus fault address register
        AFSR:                           0  Auxiliary fault status register
```
After crashing ->
```
Registers in SCB:
        CPUID:                 1091551809  CPUID base register
        ICSR:                           3  Interrupt control and state register
        VTOR:                   536870912  Vector table offset register
        AIRCR:                 -100334848  Application interrupt and reset control register
        SCR:                            0  System control register
        CCR:                          512  Configuration and control register
        SHPR1:                          0  System handler priority registers
        SHPR2:                          0  System handler priority registers
        SHPR3:                          0  System handler priority registers
        SHCRS:                        128  System handler control and state register
        CFSR_UFSR_BFSR_MMFSR:         130  Configurable fault status register
        HFSR:                  1073741824  Hard fault status register
        MMFAR:                         72  Memory management fault address register
        BFAR:                          72  Bus fault address register
        AFSR:                           0  Auxiliary fault status register
```

And NVIC also changed.
Before crashing ->
```
Registers in NVIC:
        ISER0:       0  Interrupt Set-Enable Register
        ISER1:       0  Interrupt Set-Enable Register
        ISER2:       0  Interrupt Set-Enable Register
        ICER0:       0  Interrupt Clear-Enable Register
        ICER1:       0  Interrupt Clear-Enable Register
        ICER2:       0  Interrupt Clear-Enable Register
        ISPR0:       0  Interrupt Set-Pending Register
        ISPR1:  262144  Interrupt Set-Pending Register
        ISPR2:       0  Interrupt Set-Pending Register
        ICPR0:       0  Interrupt Clear-Pending Register
        ICPR1:  262144  Interrupt Clear-Pending Register
        ICPR2:       0  Interrupt Clear-Pending Register
        IABR0:       0  Interrupt Active Bit Register
        IABR1:       0  Interrupt Active Bit Register
        IABR2:       0  Interrupt Active Bit Register
        IPR0:        0  Interrupt Priority Register
        IPR1:        0  Interrupt Priority Register
        IPR2:        0  Interrupt Priority Register
        IPR3:        0  Interrupt Priority Register
        IPR4:        0  Interrupt Priority Register
        IPR5:        0  Interrupt Priority Register
        IPR6:        0  Interrupt Priority Register
        IPR7:        0  Interrupt Priority Register
        IPR8:        0  Interrupt Priority Register
        IPR9:        0  Interrupt Priority Register
        IPR10:       0  Interrupt Priority Register
        IPR11:       0  Interrupt Priority Register
        IPR12:       0  Interrupt Priority Register
        IPR13:       0  Interrupt Priority Register
        IPR14:       0  Interrupt Priority Register
        IPR15:       0  Interrupt Priority Register
        IPR16:       0  Interrupt Priority Register
```
after crashing ->
```
Registers in NVIC:
        ISER0:       0  Interrupt Set-Enable Register
        ISER1:  262144  Interrupt Set-Enable Register
        ISER2:       0  Interrupt Set-Enable Register
        ICER0:       0  Interrupt Clear-Enable Register
        ICER1:  262144  Interrupt Clear-Enable Register
        ICER2:       0  Interrupt Clear-Enable Register
        ISPR0:       0  Interrupt Set-Pending Register
        ISPR1:       0  Interrupt Set-Pending Register
        ISPR2:       0  Interrupt Set-Pending Register
        ICPR0:       0  Interrupt Clear-Pending Register
        ICPR1:       0  Interrupt Clear-Pending Register
        ICPR2:       0  Interrupt Clear-Pending Register
        IABR0:       0  Interrupt Active Bit Register
        IABR1:       0  Interrupt Active Bit Register
        IABR2:       0  Interrupt Active Bit Register
        IPR0:        0  Interrupt Priority Register
        IPR1:        0  Interrupt Priority Register
        IPR2:        0  Interrupt Priority Register
        IPR3:        0  Interrupt Priority Register
        IPR4:        0  Interrupt Priority Register
        IPR5:        0  Interrupt Priority Register
        IPR6:        0  Interrupt Priority Register
        IPR7:        0  Interrupt Priority Register
        IPR8:        0  Interrupt Priority Register
        IPR9:        0  Interrupt Priority Register
        IPR10:       0  Interrupt Priority Register
        IPR11:       0  Interrupt Priority Register
        IPR12:       0  Interrupt Priority Register
        IPR13:       0  Interrupt Priority Register
        IPR14:       0  Interrupt Priority Register
        IPR15:       0  Interrupt Priority Register
        IPR16:       0  Interrupt Priority Register
```

40Grit, 2019-08-09:
Can you do the same for the working part.

I guess just dump the peripherals that seem relevant.

it would be good to see a side by side diff of the dumps using a program like win merge or kdiff etc.

DrynnBavis, 2019-08-09:
> Can you do the same for the working part.
> 
> I guess just dump the peripherals that seem relevant.
> 
> it would be good to see a side by side diff of the dumps using a program like win merge or kdiff etc.

Assuming you're referring to the disco board, unfortunately I don't have a JTAG header soldered onto this board so I can't do that right now. Put an order in for one though.

40Grit, 2019-08-09:
disco has onboard st-link i thought.

you should be able to use pyocd, or some st tool to do the dump.

0xc0170, 2019-08-09:
Just in case cc @ARMmbed/team-st-mcd

DrynnBavis, 2019-08-10:
@40Grit Got GDB going on the disco board finally. So this is the same spot we were crashing previously with the register info of the disco board (not crashing)

```
(gdb) info registers
r0             0x20050000          537198592
r1             0x802667d           134375037
r2             0x0                 0
r3             0x20050000          537198592
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
r7             0x0                 0
r8             0x0                 0
r9             0x0                 0
r10            0x0                 0
r11            0x0                 0
r12            0x20002c08          536882184
sp             0x2004ffb0          0x2004ffb0
lr             0x800182b           134223915
pc             0x8001924           0x8001924 <start_new_application+16>
xPSR           0x410f0000          1091502080
fpscr          0x0                 0
msp            0x2004ffb0          0x2004ffb0
psp            0x20002808          0x20002808 <_main_stack+4040>
primask        0x1                 1
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
```

40Grit, 2019-08-10:
can you show it as a side by side diff?
I'm on my phone after hours usually hard to diff in my own.

40Grit, 2019-08-10:
screenshot of,

kdiff, git diff, winmerge. MS paint with highlighted differences.

any of the above

DrynnBavis, 2019-08-10:
[image]
bit of word wrap on the sp and pc lines but made it condense for you

40Grit, 2019-08-10:
you said these are the same parts except for package?

DrynnBavis, 2019-08-10:
> you said these are the same parts except for package?

Jut realised I was on line 16 rather than 6 in the disco debugging when I printed the registers, see comment above with new pic

40Grit, 2019-08-10:
same exact binary flashed to both parts?!?

[16 more replies]

closed 2019-08-13 by DrynnBavis