ARMmbed/mbed-os #11143: TT_M3HQ fails to compile with mbed-os master

kimlep01, 2019-08-01:
### Description
Target board TT_M3HQ fails to compile any Greentea tests currently when using mbed-os master. Happens with GCC_ARM, ARM and IAR.

Steps to reproduce:
checkout latest mbed-os master
mbed compile -m TT_M3HQ -t ARM

Compiler error:
```
Compile [ 95.5%]: i2c_api.c
[Error] i2c_api.c@69,22: assigning to 'struct i2c_s' from incompatible type 'TSB_I2C_TypeDef *'
[Error] i2c_api.c@74,22: assigning to 'struct i2c_s' from incompatible type 'TSB_I2C_TypeDef *'
[Error] i2c_api.c@79,22: assigning to 'struct i2c_s' from incompatible type 'TSB_I2C_TypeDef *'
[Error] i2c_api.c@83,22: assigning to 'struct i2c_s' from incompatible type 'TSB_I2C_TypeDef *'
[Error] i2c_api.c@100,13: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
[Error] i2c_api.c@100,15: no member named 'CR2' in 'struct i2c_s'
[Error] i2c_api.c@102,13: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
[Error] i2c_api.c@102,15: no member named 'OP' in 'struct i2c_s'
[Error] i2c_api.c@103,13: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
[Error] i2c_api.c@103,15: no member named 'IE' in 'struct i2c_s'
[Error] i2c_api.c@137,13: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
[Error] i2c_api.c@137,15: no member named 'CR1' in 'struct i2c_s'
[Error] i2c_api.c@138,13: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
[Error] i2c_api.c@138,15: no member named 'PRS' in 'struct i2c_s'
[Error] i2c_api.c@151,13: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
[Error] i2c_api.c@151,15: no member named 'CR2' in 'struct i2c_s'
[Error] i2c_api.c@152,21: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
[Error] i2c_api.c@152,23: no member named 'SR' in 'struct i2c_s'
[Error] i2c_api.c@162,13: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
[ERROR] ./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:69:22: error: assigning to 'struct i2c_s' from incompatible type 'TSB_I2C_TypeDef *'
            obj->i2c = TSB_I2C0;
                     ^ ~~~~~~~~
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:74:22: error: assigning to 'struct i2c_s' from incompatible type 'TSB_I2C_TypeDef *'
            obj->i2c = TSB_I2C1;
                     ^ ~~~~~~~~
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:79:22: error: assigning to 'struct i2c_s' from incompatible type 'TSB_I2C_TypeDef *'
            obj->i2c = TSB_I2C2;
                     ^ ~~~~~~~~
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:83:22: error: assigning to 'struct i2c_s' from incompatible type 'TSB_I2C_TypeDef *'
            obj->i2c = TSB_I2C3;
                     ^ ~~~~~~~~
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:100:13: error: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
    obj->i2c->CR2 = (I2CxCR2_I2CM_ENABLE | I2CxCR2_TRX | I2CxCR2_PIN_CLEAR |
    ~~~~~~~~^~
            .
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:100:15: error: no member named 'CR2' in 'struct i2c_s'
    obj->i2c->CR2 = (I2CxCR2_I2CM_ENABLE | I2CxCR2_TRX | I2CxCR2_PIN_CLEAR |
    ~~~~~~~~  ^
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:102:13: error: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
    obj->i2c->OP  = I2CxOP_INIT;
    ~~~~~~~~^~
            .
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:102:15: error: no member named 'OP' in 'struct i2c_s'
    obj->i2c->OP  = I2CxOP_INIT;
    ~~~~~~~~  ^
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:103:13: error: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
    obj->i2c->IE  = I2CxIE_CLEAR;
    ~~~~~~~~^~
            .
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:103:15: error: no member named 'IE' in 'struct i2c_s'
    obj->i2c->IE  = I2CxIE_CLEAR;
    ~~~~~~~~  ^
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:137:13: error: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
    obj->i2c->CR1 = (I2CxCR1_ACK | clk.sck);
    ~~~~~~~~^~
            .
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:137:15: error: no member named 'CR1' in 'struct i2c_s'
    obj->i2c->CR1 = (I2CxCR1_ACK | clk.sck);
    ~~~~~~~~  ^
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:138:13: error: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
    obj->i2c->PRS = (I2CxPRS_PRCK & clk.prsck);
    ~~~~~~~~^~
            .
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:138:15: error: no member named 'PRS' in 'struct i2c_s'
    obj->i2c->PRS = (I2CxPRS_PRCK & clk.prsck);
    ~~~~~~~~  ^
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:151:13: error: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
    obj->i2c->CR2 = I2CxCR2_STOP_CONDITION;
    ~~~~~~~~^~
            .
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:151:15: error: no member named 'CR2' in 'struct i2c_s'
    obj->i2c->CR2 = I2CxCR2_STOP_CONDITION;
    ~~~~~~~~  ^
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:152:21: error: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
    while ((obj->i2c->SR & I2CxSR_BB) == I2CxSR_BB) {
            ~~~~~~~~^~
                    .
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:152:23: error: no member named 'SR' in 'struct i2c_s'
    while ((obj->i2c->SR & I2CxSR_BB) == I2CxSR_BB) {
            ~~~~~~~~  ^
./targets/TARGET_TT/TARGET_TT_M3HQ/i2c_api.c:162:13: error: member reference type 'struct i2c_s' is not a pointer; did you mean to use '.'?
    obj->i2c->CR2 = I2CxCR2_SWRES_10;
    ~~~~~~~~^~
            .
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

linlingao, 2019-08-01:
mbed-os-example-blinky also fails to compile.

toyowata, 2019-08-06:
cc @Ronny-Liu @ThunderSoft123

matsujirushi, 2019-08-15:
FYI
mbed-os-5.13.0 is no error.
```
mbed update mbed-os-5.13.0
```

0xc0170, 2019-08-21:
@Tharazi97 How did you test your patch #11237 ? The latest nightly still contain a problem:

```
  * TT_M3HQ::ARMC6::TESTS-NETWORK-L3IP
        Building project l3ip (TT_M3HQ, ARMC6)
        Scan: ARM
        Scan: l3ip
        Link: l3ip
        [Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
        [Error] @0,0: L6218E: Undefined symbol i2c_abort_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/I2C.o).
        [Error] @0,0: L6218E: Undefined symbol i2c_active (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/I2C.o).
        [Error] @0,0: L6218E: Undefined symbol i2c_irq_handler_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/I2C.o).
        [Error] @0,0: L6218E: Undefined symbol i2c_transfer_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/I2C.o).
        [Error] @0,0: L6218E: Undefined symbol spi_abort_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPI.o).
        [Error] @0,0: L6218E: Undefined symbol spi_active (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPI.o).
        [Error] @0,0: L6218E: Undefined symbol spi_irq_handler_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPI.o).
        [Error] @0,0: L6218E: Undefined symbol spi_master_transfer (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPI.o).
        [Error] @0,0: L6218E: Undefined symbol spi_slave_read (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPISlave.o).
        [Error] @0,0: L6218E: Undefined symbol spi_slave_receive (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPISlave.o).
        [Error] @0,0: L6218E: Undefined symbol spi_slave_write (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPISlave.o).
        [Error] @0,0: L6218E: Undefined symbol rtc_init (referred from BUILD/tests/TT_M3HQ/ARM/platform/source/mbed_rtc_time.o).
        [Error] @0,0: L6218E: Undefined symbol rtc_isenabled (referred from BUILD/tests/TT_M3HQ/ARM/platform/source/mbed_rtc_time.o).
        [Error] @0,0: L6218E: Undefined symbol rtc_read (referred from BUILD/tests/TT_M3HQ/ARM/platform/source/mbed_rtc_time.o).
        [Error] @0,0: L6218E: Undefined symbol rtc_write (referred from BUILD/tests/TT_M3HQ/ARM/platform/source/mbed_rtc_time.o).
        Warning: L3912W: Option 'legacyalign' is deprecated.
        Error: L6218E: Undefined symbol i2c_abort_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/I2C.o).
        Error: L6218E: Undefined symbol i2c_active (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/I2C.o).
        Error: L6218E: Undefined symbol i2c_irq_handler_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/I2C.o).
        Error: L6218E: Undefined symbol i2c_transfer_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/I2C.o).
        Error: L6218E: Undefined symbol spi_abort_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPI.o).
        Error: L6218E: Undefined symbol spi_active (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPI.o).
        Error: L6218E: Undefined symbol spi_irq_handler_asynch (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPI.o).
        Error: L6218E: Undefined symbol spi_master_transfer (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPI.o).
        Error: L6218E: Undefined symbol spi_slave_read (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPISlave.o).
        Error: L6218E: Undefined symbol spi_slave_receive (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPISlave.o).
        Error: L6218E: Undefined symbol spi_slave_write (referred from BUILD/tests/TT_M3HQ/ARM/drivers/source/SPISlave.o).
        Error: L6218E: Undefined symbol rtc_init (referred from BUILD/tests/TT_M3HQ/ARM/platform/source/mbed_rtc_time.o).
        Error: L6218E: Undefined symbol rtc_isenabled (referred from BUILD/tests/TT_M3HQ/ARM/platform/source/mbed_rtc_time.o).
        Error: L6218E: Undefined symbol rtc_read (referred from BUILD/tests/TT_M3HQ/ARM/platform/source/mbed_rtc_time.o).
        Error: L6218E: Undefined symbol rtc_write (referred from BUILD/tests/TT_M3HQ/ARM/platform/source/mbed_rtc_time.o).
        Finished: 0 information, 1 warning and 15 error messages.
        
```

I can reproduce this locally as well. I'll reopen this issue

0xc0170, 2019-08-21:
Fix should be simple, I'll send PR shortly

PR #11237 (2019-08-20): Fix TT_M3HQ build problem
### Description

This fixes the problem with compiling i2c api, serial api and spi api on TT_M3HQ.

fixes: #11143
### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@jamesbeyond @maciejbocianski 
### Release Notes
commit 2cfea4fa (2019-08-28): target TT_: fix compilation errors

They define slave for spi, rtc but do not contain implementation - removing them.
Fixes #11143
closed 2019-08-28 by 0xc0170