ARMmbed/mbed-os #11144: BSD / POSIX Sockets lwip

JRubis, 2019-08-01:
### Description

I am working on porting a project that was originally written for POSIX compliant systems. 
Can I bypass Mbed's standard socket layer and use lwip directly? If so, what are the potential drawbacks of this approach?
### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

syhoangmai, 2019-08-07:
Hi, 
In my opinion, Mbed socket API relies on different "under layer" depending on the real physical network layer (please correct me if I am wrong). For example, if STM target use Ethernet then the Mbed socket will relies on lwip, but if a target use a cellular modem with AT interface then Lwip is kind of irrelevant. For portability and flexibility reason I think it is best to use MBed API for it (and MBed team takes care of it), unless there is a reason to not doing so.

JRubis, 2019-08-08:
I thought it was optional to either use the LwIP stack or the embedded stack in the communications modules. Seems I have seen that noted somewhere in the Cellular documentation. There are advantages and disadvantages to both. 

Since originally posting this, I have managed to successfully port the project over to use Mbed's standard socket interface. It just required more work that would have been necessary on a POSIX/BSD compliant system.

0xc0170, 2019-08-08:
cc @ARMmbed/mbed-os-ipcore

kjbracey-arm, 2019-08-12:
I sympathise with the issue - I'm not a fan of respelling standard APIs for the sake of it myself.

You could reconfigure lwIP to use its LWIP_COMPAT_SOCKETS, but you lose a chunk of portability while doing so - you're locking yourself in to lwIP-based network interfaces.

Mbed OS does provide POSIX filesystem wrapper functions, so you have the choice of using the C++ `FileHandle` etc classes or just POSIX.

It would make sense to have the plumbing to also provide sockets via POSIX APIs, and it's been filed under "nice to do some time". A prerequisite for doing it well would be bringing together the `Socket` and `FileHandle` abstractions, which is the tricky bit (considering backwards compatibility). That trickiness is why it keeps getting pushed back.

closed 2019-08-23 by JRubis