ARMmbed/mbed-os #11153: Wrapper around LittleFS makes static allocation unavailable

LaurentLouf, 2019-08-02:
### Description

- Target : not relevant
- Toolchain : not relevant
- Tools : not relevant
- Mbed OS : true for all versions implementing LittleFS

LittleFS provides internally the possibility to use statically allocated buffers for its operations, but the current wrapper around the library does not expose this functionality. There is basically two things : 

- in the `mbed_lib.json` file, the macro `LFS_NO_MALLOC` is not present
- in `LittleFileSystem::mount` and `LittleFileSystem::format`, there is no argument for static buffers, that would need to be set in the `struct lfs_config` structure : `read_buffer`, `prog_buffer`, `lookahead_buffer` and `file_buffer`

I realize that this means opening the pandora box of troubles, but the default behavior could stay the same with default values for the `mount` and `format` additional parameters needed.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

linlingao, 2019-08-02:
@geky can you comment?

geky, 2019-08-05:
@LaurentLouf thanks for creating an issue.

This would be nice to have. LFS_NO_MALLOC was added after the initial port to Mbed OS which is why it was missed.

Though note this would not be able to complete remove dynamic allocations due to quirks in the FileSystem API:
https://github.com/ARMmbed/mbed-os/blob/master/features/storage/filesystem/littlefs/LittleFileSystem.cpp#L411

It's possible to rework the FileSystem API to avoid these small allocations, but it would require quite a bit more work.

I don't really have any other comments.

closed 2019-10-31 by ciarmcom