ARMmbed/mbed-os #11158: LittleFS prog_size greater than read_size causes Hard Fault

LaurentLouf, 2019-08-02:
### Description

- Target : NUCLEO F103RB
- Toolchain : GNU Tools for Arm Embedded Processors 8-2018-q4-major
- Tools : mbed-cli 1.10.0
- Mbed OS : mbed-os-5.11.1

In my application, we are using LittleFS to store data in an external SPI Flash (W25Q64BV for reference). We have found a very strange behavior when trying to configure LittleFS : a prog_size greater than read_size causes a crash, systematically. Our crashing configuration is as follow : 

```
      "littlefs.block_size": 4096,
      "littlefs.prog_size": 1024,
      "littlefs.read_size": 512,
```

Note that this does not seem to be related to a lack of memory as when we increase `read_size` to 2048 (for example), all goes well. The code is basically the following : 

Main : 
```cpp
    for (int i = 0; i < 10; i++) {
        /* code */
        mbed_error_printf("----- RUN %d START -----\n", i);
        mbed_error_printf("DataStorage new()\n");
        DataStorage* file_system;
        file_system = new DataStorage(SPI_FLASH_STM32_USER);

        bool initial_setup_done = false;
        if (file_system->get_initial_setup_done(initial_setup_done) !=
            FS_RESULT_CONFIG_READ_FS_NOT_INITIALIZED) {
            if (!initial_setup_done) {
                mbed_error_printf("First initial setup\n");
                file_system->set_initial_setup_done();
                MOONA_LOG_VERBOSE("Initial setup done");
            }
        }
        delete file_system;
        mbed_error_printf("DataStorage deleted()\n");
        mbed_error_printf("----- RUN %d END -----\n", i);
    }
```

DataStorage : 
```cpp
DataStorage::DataStorage(spi_flash_user_t i_spi_flash_user)
    : initialized(false),
      spi_flash_user(i_spi_flash_user) {
    memset(full_night_directory_path, 0, PATH_NAME_MAX_SIZE);

    // If it's the STM32 that's gonna use the flash, initialize the block device and file system
    if (i_spi_flash_user == SPI_FLASH_STM32_USER) {
        block_device = new SPIFBlockDevice(PIN_FLASH_MOSI, PIN_FLASH_MISO, PIN_FLASH_SCLK,
                                           PIN_FLASH_CSEL, SPI_FLASH_CLOCK_FREQUENCY);
        file_system = new LittleFileSystem("fs");
        initialized = initialize();
    }
}

bool DataStorage::initialize(void) {
    if (initialized) {
        return true;
    } else {
        int error = file_system->mount(block_device);
        if (error) {
            // Reformat if we can't mount the filesystem, this should only happen on the first boot
            LittleFileSystem::format(block_device);
            error = file_system->mount(block_device);

            if (!error) {
                return true;
            } else {
                return false;
            }

        } else {
            return true;
        }
    }
}

fs_result_config_read_t DataStorage::get_initial_setup_done(bool &result) {
    if (initialized) {
        // Build the file path
        char file_path[PATH_NAME_MAX_SIZE] = {0};
        strncat(file_path, FILE_SYSTEM_ROOT_DIRECTORY_STM, PATH_NAME_MAX_SIZE - strlen(file_path));
        strncat(file_path, CONFIG_ROOT_DIRECTORY, PATH_NAME_MAX_SIZE - strlen(file_path));
        strncat(file_path, INITIAL_SETUP_DONE_FILE, PATH_NAME_MAX_SIZE - strlen(file_path));
        mbed_error_printf("file_path: %s\n", file_path);
        // Try to open the file : fail -> no initial setup done
        FILE *file = std::fopen(file_path, "r");
        if (!file) {
            result = false;
            return FS_RESULT_CONFIG_READ_FROM_FILE;
        } else {
            std::fclose(file);
            result = true;
            return FS_RESULT_CONFIG_READ_FROM_FILE;
        }
    } else {
        return FS_RESULT_CONFIG_READ_FS_NOT_INITIALIZED;
    }
}

fs_result_write_t DataStorage::set_initial_setup_done(void) {
    if (initialized) {
        // Build the file path
        char file_path[PATH_NAME_MAX_SIZE] = {0};
        strncat(file_path, FILE_SYSTEM_ROOT_DIRECTORY_STM, PATH_NAME_MAX_SIZE - strlen(file_path));
        strncat(file_path, CONFIG_ROOT_DIRECTORY, PATH_NAME_MAX_SIZE - strlen(file_path));
        strncat(file_path, INITIAL_SETUP_DONE_FILE, PATH_NAME_MAX_SIZE - strlen(file_path));
        mbed_error_printf("file_path: %s\n", file_path);
        FILE *file = std::fopen(file_path, "w");
        if (file) {
            std::fclose(file);
            return FS_RESULT_WRITE_SUCCESS;
        } else {
            return FS_RESULT_WRITE_ERROR;
        }
    } else {
        return FS_RESULT_WRITE_FS_NOT_INITIALIZED;
    }
}
```
With the SPI Block device implementation taken from https://github.com/ARMmbed/spif-driver/#5c49300390e360716389d0030d9e5017740b7f82 . 

Digging in `lfs.h` which is said to contain the latest the documentation, we even find : 
```cpp
    // Minimum size of a block read. This determines the size of read buffers.
    // This may be larger than the physical read size to improve performance
    // by caching more of the block device.
    lfs_size_t read_size;

    // Minimum size of a block program. This determines the size of program
    // buffers. This may be larger than the physical program size to improve
    // performance by caching more of the block device.
    // Must be a multiple of the read size.
    lfs_size_t prog_size;
```

So following the documentation seems to lead to an error. Is this just a problem in the documentation ? Any input @geky  ?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

LaurentLouf, 2019-08-02:
So far, the only indication of the size I have found in the code is in `lfs.c` in the function `lfs_file_opencfg` :

```cpp
    // allocate buffer if needed
    file->cache.block = 0xffffffff;
    if (file->cfg && file->cfg->buffer) {
        file->cache.buffer = file->cfg->buffer;
    } else if (lfs->cfg->file_buffer) {
        if (lfs->files) {
            // already in use
            return LFS_ERR_NOMEM;
        }
        file->cache.buffer = lfs->cfg->file_buffer;
    } else if ((file->flags & 3) == LFS_O_RDONLY) {
        file->cache.buffer = lfs_malloc(lfs->cfg->read_size);
        if (!file->cache.buffer) {
            return LFS_ERR_NOMEM;
        }
    } else {
        file->cache.buffer = lfs_malloc(lfs->cfg->prog_size);
        if (!file->cache.buffer) {
            return LFS_ERR_NOMEM;
        }
    }
```

I'd say that it sorts of implies that indeed `prog_size` should be greater than `read_size`, though it's not really crystal clear.

LaurentLouf, 2019-08-02:
There's a few asserts, but well, unless the precedence of operators is non standard in mbed-os, those are likely to always be true, because [you know](https://wiki.sei.cmu.edu/confluence/display/c/EXP00-C.+Use+parentheses+for+precedence+of+operation) (@geky there's a few of them in your code, even in the most recent version of it)
https://github.com/ARMmbed/mbed-os/blob/c966348d3f9ca80843be7cdc9b748f06ea73ced0/features/storage/filesystem/littlefs/littlefs/lfs.c#L2074

geky, 2019-08-05:
Hmm, yes that shouldn't happen. Do you know where it is crashing or have the log output?

There shouldn't be a restriction between prog_size and read_size, though this situation is not well tested as most devices have read_size < prog_size.

Also note that the mbed_lib.json geometry only acts as a suggestion and it will be overridden if it is too small for the block device:
https://github.com/ARMmbed/mbed-os/blob/master/features/storage/filesystem/littlefs/LittleFileSystem.cpp#L187-L190

LaurentLouf, 2019-08-05:
I have the following log : 

```txt
malloc call: 0x80178B9
ptr: 0x20004658 (+2528)
----- RUN 0 START -----
DataStorage new()

malloc call: 0x800F01D
ptr: 0x20005040 (+2412)

malloc call: 0x800F01D
ptr: 0x200059B0 (+248)

malloc call: 0x800F01D
ptr: 0x20005AB0 (+200)
lfs_init start
lfs_init malloc read buf

malloc call: 0x800C087
ptr: 0x20005B80 (+64)
lfs_init malloc prog buf

malloc call: 0x800C0B3
ptr: 0x20005BC8 (+256)
lfs_init lfs_cache_zero buffers...
lfs_init lfs_cache_zero read
lfs_init lfs_cache_zero prog
lfs_init malloc lookahead buf

malloc call: 0x800C0CD
ptr: 0x20005CD0 (+256)
lfs_init end 
file_path: /fs/config/initial_setup

malloc call: 0x8018CF1
ptr: 0x20005DD8 (+428)
mbed_retarget.cpp:open() start
mbed_retarget fs->open...
FileSystem::open() start

malloc call: 0x800F01D
ptr: 0x20005F88 (+12)
FileSystem::open() new Managed<File>
File::open
LittleFileSystem::file_open start

malloc call: 0x800F01D
ptr: 0x20005F98 (+56)
LittleFileSystem::file_open new lfs_file_t
file_open(0x0, "config/initial_setup", 0x0)

malloc call: 0x800DAA9
ptr: 0x20005FD8 (+64)
file_open -> 0File::open err = 0
FileSystem::open() err: 0
mbed_retarget fs->open done.
mbed_retarget.cpp:open() end

free call: 0x800D5F5
free ptr: 0x20005FD8

free call: 0x800B92D
free ptr: 0x20005F98

free call: 0x800B3FF
free ptr: 0x20005F88

free call: 0x800BEE9
free ptr: 0x20005B80

free call: 0x800BEF5
free ptr: 0x20005BC8

free call: 0x800CA6F
free ptr: 0x20005CD0

free call: 0x8016BBF
free ptr: 0x200059B0

free call: 0x800BC1B
free ptr: 0x20005AB0

free call: 0x80098E3
free ptr: 0x20005040
DataStorage deleted()
----- RUN 0 END -----
----- RUN 1 START -----
DataStorage new()

malloc call: 0x800F01D
ptr: 0x20005040 (+2412)

malloc call: 0x800F01D
ptr: 0x20005CD0 (+248)

malloc call: 0x800F01D
ptr: 0x200059B0 (+200)
lfs_init start
lfs_init malloc read buf

malloc call: 0x800C087
ptr: 0x20005A80 (+64)
lfs_init malloc prog buf

malloc call: 0x800C0B3

free call: 0x801A435
free ptr: 0x20005F88
ptr: 0x20007008 (+256)
lfs_init lfs_cache_zero buffers...
lfs_init lfs_cache_zero read
lfs_init lfs_cache_zero prog
lfs_init malloc lookahead buf

malloc call: 0x800C0CD
ptr: 0x20007110 (+256)
lfs_init end 
file_path: /fs/config/initial_setup
mbed_retarget.cpp:open() start
mbed_retarget fs->open...
FileSystem::open() start

malloc call: 0x800F01D
ptr: 0x20005F88 (+12)
FileSystem::open() new Managed<File>
File::open
LittleFileSystem::file_open start

malloc call: 0x800F01D
ptr: 0x20005F98 (+56)
LittleFileSystem::file_open new lfs_file_t
file_open(0x0, "config/initial_setup", 0x0)

malloc call: 0x800DAA9

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0: 20000320
R1: 1196920
R2: 11968D8
R3: FFFFFFFF
R4: 80
R5: 48
R6: 20000320
R7: 20000748
R8: 49
R9: 20000828
R10: 0
R11: 0
R12: 20000750
SP   : 20003C28
LR   : 20000840
PC   : 801A06A
xPSR : 21000000
PSP  : 20003C08
MSP  : 2000BFC0
CPUID: 411FC231
HFSR : 40000000
MMFSR: 0
BFSR : 0
UFSR : 100
DFSR : 0
AFSR : 0
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --
```

It crashes on a malloc.

geky, 2019-08-05:
Ah! I know why this is. There is a bad assumption in the file's sync logic that uses a read buffer as a temporary prog buffer. This doesn't work if read buffer > prog buffer:
https://github.com/ARMmbed/mbed-os/blob/master/features/storage/filesystem/littlefs/littlefs/lfs.c#L1473

This has been fixed in [v2](https://github.com/ARMmbed/littlefs/releases/tag/v2.0.0), which included a revamp of the caching code to separate the cache size from the physical read/prog sizes.

v2 has not made its way into Mbed OS yet, but can be added as a separate repo here:
https://github.com/armmbed/mbed-littlefs

I think the best answer is to say v1 does not support prog_size < read_size.

LaurentLouf, 2019-08-06:
> I think the best answer is to say v1 does not support prog_size < read_size.

The thing is, my crashing configuration is prog_size (1024) > read_size (512). But if a read buffer of size 512 is used as a temporary prog buffer of size 1024, it seems logic that it would fail, but in the opposite configuration of what you've described. Thus making the documentation misleading. Can you clarify, I feel like you said the opposite of what you wanted.

pilotak, 2019-08-06:
@geky could you please update [mbed-littlefs](https://github.com/ARMmbed/mbed-littlefs) to it's up to date with [littlefs](https://github.com/ARMmbed/littlefs)

geky, 2019-08-06:
> The thing is, my crashing configuration is prog_size (1024) > read_size (512).

Oh! My bad. I misread your original issue. prog_size > read_size should be fine in v1.

Ironically this is caused by a different bug that was fixed in v1.6.1, Mbed OS is at v1.6.0:
https://github.com/ARMmbed/littlefs/commit/3419284689ead61c6905d61359a2b62a8c09c75d

I was working on bringing littlefs in Mbed OS up to date, but had to put it on the back-burner because of life things.

---

@pilotak I'm assuming you mean v2.0 -> v2.1?

I'm currently working on a script that will update both of these automatically (and littlefs-fuse and any other dependencies if someone puts forward a script). So I can remove the human component (myself) which has been the limiting issue.

pilotak, 2019-08-07:
@geky yes. The script would be perfect i though i would do it manually but as you say human component is a limiting issue on my side too.

LaurentLouf, 2019-08-07:
Thanks @geky for the answer. Too bad you have a life outside of work :) ! It would definitely be nice to update to a version not producing a hard fault, but the workaround is pretty straightforward so for the time being, it's perfectly ok like that.

closed 2019-08-07 by LaurentLouf