ARMmbed/mbed-os #11170: Ublox cellular: broken support for custom board

pilotak, 2019-08-06:
### Description
PR #10877 really breaks support for custom targets with ublox modem, everything is **target specific** and not **modem specific**. I'm using SARA-G350 (part of C027) with STM32F412ZG/STM32F303RE.

Following command is not supported on this modem and fails to connect, it should be moved to a `line 62` where this variable is used

https://github.com/ARMmbed/mbed-os/blob/431c4c19ae2056785c85e2a119ed51dd35f79f3c/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT_CellularContext.cpp#L52

Following part is limiting, this is where it should be specified `#ifdef TARGET_UBLOX_C027 || TARGET_HAS_G350` or similar.

https://github.com/ARMmbed/mbed-os/blob/431c4c19ae2056785c85e2a119ed51dd35f79f3c/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT.cpp#L98-L110

 I know that i can override `init()` but than there would a problem with `config_authentication_parameters()` because it's **private** and of course overriding it takes aditional 192 bytes extra space

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jarvte, 2019-08-08:
@pilotak First one is an easy fix but the second one is a trickier. We don't have target for SARA-G350 to differentiate variants of C027 like we have for C030 (for example UBLOX_C030_R41XM). Should there be different variants of C027?
@mudassar-ublox do you have suggestions how to fix this?

jarvte, 2019-08-08:
@pilotak are you using SARA-G350 as a shield? Then you should set "provide-default" to true in /features/cellular/framework/targets/UBLOX/AT/mbed_lib.json

pilotak, 2019-08-09:
No i have a custom board (as description says). I thought that it would be nice to have all modems done like components for example storage. You just specify you have a ublox and which type, you override `get_target_default_instance` with your pins and you done. At the moment it's tied up to a specific board and if you decide to make you own board with different processor you are in trouble.

I noticed that Quectel has modems split into each one and not tied up to MCU which makes a goal for custom targets. I realize that managing one code for all the ublox modems is easier but you need to use same MCU because of library only.

At the moment i have my own init without `config_authentication_parameters();` as this is for PDP and therefore **not relevant when using AT commands** and stack on modem and i don't have to make the function public.
Another change a did as it works better/faster is to exclude modem shutdown in init. The reason behind it is that if you call `disconnect()` it will disconnect you from internet not from a cell. Next call to `connect()` will shutdown and wake up modem again, do network registration to catch a cell and than connect to internet = a lot of time wasted. If there isn't that command it takes ~2 sec to reconnect internet because it is already attached a cell. Maybe this is a mandatory for R412 i don't know but not for G350 as it works like a charm
https://github.com/ARMmbed/mbed-os/blob/431c4c19ae2056785c85e2a119ed51dd35f79f3c/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT.cpp#L99

@jarvte Could i kindly ask you to move the RAT thing where it belongs or do you want me to create a PR?

jarvte, 2019-08-09:
Yes, I will do a PR to move the RAT thing.

pilotak, 2019-08-09:
@jarvte What are your thoughts on above? (modems as component and init delays)

closed 2019-10-01 by pilotak