ARMmbed/mbed-os #11174: Thread Flags for Termination

checked01, 2019-08-06:
### Description

I am currently working with the STM32F407VG with Mbed OS 5 on PlatformIO. I have several threads currently running with while loops that are terminated by setting a boolean value (eg. thread_alive) to false so they can be joined.

I am trying to move to using thread flags to join the threads, but ran into issues. My threads are held by a class, which we can call ``MyClass``.

 I start a thread like so:
```
void MyClass::startThread() {
  thread.start(
      Callback<void()>(this, &MyClass::threadFunction));
}
```
And I would like to terminate it with:
```
void MyClass::joinThread() {
  thread.flags_set(STOP_FLAG);
  thread.join();
}
```
The function executing in the thread itself looks like:
```
void MyClass::threadFunction() {
  while (!ThisThread::flags_wait_any_for(STOP_FLAG, 1000)) {
    // DO STUFF
  }
}
```
When I try to stop the thread with ``joinThread()`` the program hangs on ``thread.join()``. When messing around I stuck ``thread.flags_set(STOP_FLAG)`` in my startThread() function to test, and it stopped the thread as expected.
```
void MyClass::startThread() {
  thread.start(
      Callback<void()>(this, &MyClass::threadFunction));
  thread.flags_set(STOP_FLAG);
}
```

The ``flags_set(STOP_FLAG)`` does not seem to be reaching the thread itself when executed in ``joinThread()``. Printing out the `ThisThread::flags_get()`` result in ``MyClass::threadFunction()`` showed flags remaining set to 0.

How can I use ``Thread::flags_set(flag)`` to help join my thread?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

kjbracey-arm, 2019-08-13:
> ```
> void MyClass::threadFunction() {
>   while (!ThisThread::flags_wait_any_for(STOP_FLAG, 1000)) {
>     // DO STUFF
>   }
> }
> ```

So this runs periodically, pausing for a second each time per iteration, with that second terminating early if the stop flag is set?

The actual stop flag part of that should work as you want. Are you definitely getting back to the loop top to do the check, not getting stuck somewhere in DO STUFF waiting for something else?

One thought - is `STOP_FLAG` possibly `1u<<31`? C++ docs fail to mention this, but bit 31 is reserved by the underlying CMSIS-RTOS for error returns. I think that would cause an assert error though.

> I am trying to move to using thread flags to join the threads, but ran into issues.

I don't see a compelling reason to make that change, necessarily. It depends on how/whether you need to actually wake the threads. Blocking to wait for the flag, and waking via that flag, is the functionality that thread flags adds.

If you really are running periodically like the above, you could just as well check a bool once per second. But for correctness, a bool accessed via multiple threads should be mutex-protected, or be atomic (a C `bool` accessed via `core_util_atomic_` functions, or a C++ `mstd::atomic_bool` on current master, coming in 5.14).

For maximally-flexible waking, as long as the waking is coming from thread not IRQ, I generally prefer to use condition variables. Then you can use whatever mutex-protected shared data structures you like, such as `bool stop_request` or `xxx *work_queue`, and use `ConditionVariable::wait` and `ConditionVariable::notify` to wait for and communicate changes in those shared structures between the threads.

checked01, 2019-08-13:
The code does run periodically, if the stop flag is set it will complete doing the 'STUFF' and then terminate on that iteration.

Fair point regarding the functionality, as the code in the thread simply needs to run continuously or stop running, there likely isn't a need for the added functionality provided by thread flags. I got flags working by changing some of the code within `//DO STUFF`, you were right, the code was never reaching back to the while condition.

I was not aware of condition variables. After reading up on mbed's rtos handbook, they could be very useful in other parts of my code where I've been relying solely on mutex's and bools to block execution.

I'm still fairly new to working with an RTOS, thanks for the help!

0xc0170, 2019-08-13:
@checked01 If your question is answered, we can close the issue?

checked01, 2019-08-13:
Yep!

closed 2019-08-13 by checked01