ARMmbed/mbed-os #11175: Watchdog issue with timeout

Tharazi97, 2019-08-07:
### Description

In docs regarding watchdog there is written that "The timeout must be accurate to the nearest millisecond.". What does that exactly mean? The timeout must be accurate to the nearest millisecond that user tried to initialize watchdog? According to my calculations when we use 8 bit prescaller (for big timeouts) we are accurate to 6,4 ms (for 40 kHz LSI). Or does that mean that it is accurate to the nearest possible millisecond? I think that needs a clarification.
docs: https://os.mbed.com/docs/mbed-os/v5.9/reference/watchdog-port.html

Also currently calculated timout is rounded down, but in this test:  
https://github.com/ARMmbed/mbed-os/blob/431c4c19ae2056785c85e2a119ed51dd35f79f3c/TESTS/mbed_drivers/watchdog/main.cpp#L223-L230
It is checked if calculated is bigger or equal to the value passed via init function. This test passes for most of the targets, because of their LSI frequency value which after division throgh it gives good integer value so that reload value can be calculated without any rounding. But not in the case of STM32-L0xx boards where the LSI frequency value is set to 37 kHz.

So should we change the test so we accept rounding down, or should we round the values up (if it doesn't exceed the upper limit of timeout)?

@maciejbocianski @mprse @fkjagodzinski @jamesbeyond 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

fkjagodzinski, 2019-08-14:
> So should we change the test so we accept rounding down, or should we round the values up (if it doesn't exceed the upper limit of timeout)?

The test verifies our timing requirements for watchdog timeout. Has to stay as it is. It is the implementation, that has to be updated, if it doesn't comply with requirements.

0xc0170, 2019-08-14:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2019-08-19:
Issue confirmed, you can st st device label

fkjagodzinski, 2019-08-22:
#11178 has been merged. @Tharazi97 could you confirm it fixed this issue?

Tharazi97, 2019-08-22:
Tested, it fixes the issue. Closing.

closed 2019-08-22 by Tharazi97