ARMmbed/mbed-os #11177: Are there any plans to update mbedlib?

valeros, 2019-08-07:
### Description

Hello! `mbedlib` hasn't been updated for [5 months](https://os.mbed.com/users/mbed_official/code/mbed-dev/shortlog/). Is there any ETA when it might be updated? Is there any way how to use the latest mbed OS with boards that only support mbed 2 (e.g. RedBearLab BLE Nano)?

Thanks!

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-08-12:
> Hello! mbedlib hasn't been updated for 5 months. Is there any ETA when it might be updated?

We will provide an update to mbed lib and mbed-dev in the next weeks.

> Is there any way how to use the latest mbed OS with boards that only support mbed 2 (e.g. RedBearLab BLE Nano)?

@linlingao would you know who could help to answer this question? I am not aware of this.

bulislaw, 2019-08-12:
@valeros would you be willing to try using your board with the bare metal profile and Mbed OS 5? https://os.mbed.com/docs/mbed-os/v5.13/reference/mbed-os-bare-metal.html

closed 2019-10-25 by ciarmcom