ARMmbed/mbed-os #11182: Get Total Bytes Sent and Received

JRubis, 2019-08-07:
### Description

Is there a way to get the total number of bytes sent and received over a specific network interface? I did a search of the API(s) and didn't see one for this.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-08-12:
cc @ARMmbed/mbed-os-ipcore

lauri-piikivi, 2019-08-14:
Hi,

there is support for statistics of sockets. Assuming that all sockets are for same NetworkInterface, the sockets can be read through, in file https://github.com/ARMmbed/mbed-os/blob/b82cfcc38be1b93fdb7a7192100569f060011569/features/netsocket/SocketStats.h#L45

JRubis, 2019-08-15:
@lauri-piikivi thank you. It seems these stats would keep track of bytes sent/received at the socket level but not the total sent at the networking interface level. For instance, bytes used by the IP header are not counted. Not sure if it would count the overhead added when TLS is enabled or not. I suspect it might since TLS is layered over sockets. Do you know?

lauri-piikivi, 2019-08-19:
Hi, the stats do not consider the protocol overhead, there did not seem to be enough justification to make the status in  the various integrated and offloaded stacks. Socket level was decided as the common layer.
If the interest is how the Pelion cloud client uses DTLS then client uses a TCP or UDP socket underneath, and stats would be calculated for that underlying socket. Similarly the TLSSocket uses a normal socket underneath, so the stats will include the DTLS overhead.

JRubis, 2019-08-19:
OK. Thanks for the confirmation. I do know that some of the network interface modules themselves e.g. BG96 have an AT command to get total bytes sent and received, but there isn't a standard API for this.

closed 2019-09-04 by JRubis