ARMmbed/mbed-os #11187: startAdvertising

jag102969, 2019-08-08:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

jag102969, 2019-08-08:
I am using mbed 5.13.1 with nRF52840 on both Client and Server and I get...
[image]

jag102969, 2019-08-08:
Does anyone have any suggestions?

0xc0170, 2019-08-09:
cc @ARMmbed/mbed-os-pan Any pointers?

jag102969, 2019-08-09:
If I call stopAdvertising() with a handle the error goes away....

1. I got this idea because startAdvertising() in the BLE_LED examples uses it this way.
/* Start advertising */
error = _ble.gap().stopAdvertising(ble::LEGACY_ADVERTISING_HANDLE);

2. The mbed documentation doesn't mention the handle.
https://os.mbed.com/docs/mbed-os/v5.13/mbed-os-api-doxy/class_b_l_e.html#ac504a922cbbc78b8b1dbf01f145e381e

jag102969, 2019-08-09:
When I fixed this problem I have the same problem with _ble.gap().disconnect()

When I call.....
        ble_error_t error = _ble.gap().disconnect( ble::LEGACY_ADVERTISING_HANDLE, 
                                                   Gap::REMOTE_USER_TERMINATED_CONNECTION );
I get the same **"Use of deprecated scan API with up to date API"**

paul-szczepanek-arm, 2019-08-12:
You cannot mix old API and new API in the same program. You need to use the new disconnect which takes a local_disconnection_reason_t as the second parameter.
```
    ble_error_t disconnect(
        connection_handle_t connectionHandle,
        local_disconnection_reason_t reason
    );
```
Also you should use a real connection handle you have received from a connection event.
So your code would be:
```
ble_error_t error = _ble.gap().disconnect(connection_handle, ble::local_disconnection_reason_t::USER_TERMINATION);
```

jag102969, 2019-08-13:
Great! This fixed my problem... To wrap this solution up I wanted to throw in a little more clarification.

**Declare:**
ble::connection_handle_t connectionHandle;

**Use to get handle:**
void onConnectionComplete(const ble::ConnectionCompleteEvent &event)
{
        connectionHandle = event.getConnectionHandle();
}

**Use to disconnect:**
ble_error_t error = _ble.gap().disconnect(   connectionHandle, ble::local_disconnection_reason_t::USER_TERMINATION );

closed 2019-08-13 by jag102969