ARMmbed/mbed-os #11192: Quectel EC2X: Connection fails most of the time

Arjun765, 2019-08-09:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

I'm using the following code to try to establish a connection. I have a EC20 eval board connected to my custom board with a serial cable. I press the reset button on the EC20, wait for both the blue leds to light up and then release the reset on my board. Most of the time connection fails.

        UARTSerial uart(EXT_SERIAL_TX, EXT_SERIAL_RX, 115200);

        QUECTEL_EC2X modem(&uart, PinName::NC, true);

        CellularContext *ctx = modem.create_context();

        ctx->set_credentials("internet");

        int pass = 0, fail = 0;

        while (true) {

                while (!ctx->is_connected()) {
                        printf("Connecting...\r\n");
                        auto result = ctx->connect();
                        if (result != NSAPI_ERROR_OK) {
                                printf("ERROR: Connect failed: %d\r\n", result);
                                fail++;
                        }
                }
                pass++;
                green = true;
                printf(COLOR_GREEN "Connected succesfully %d %d\r\n" COLOR_RESET, pass, fail);

                char buffer[] = "GET / HTTP/1.1\r\n"
                                "Host: 1.1.1.1\r\n"
                                "\r\n";

                TCPSocket socket;
                socket.open(ctx->get_stack());
                socket.connect("1.1.1.1", 80);
                socket.send(buffer, sizeof(buffer));
                socket.recv(buffer, sizeof(buffer));

                printf("%s\r\n", buffer);
                wait(10);
                printf("Disconneting...\r\n");
                green = false;
                ctx->disconnect();
                printf("Disconnected\r\n");
        }

I'm getting the following output

    [INFO][CELL]: Found PDP context 4
    [INFO][CELL]: CellularContext PPP connect
    [INFO][CELL]: AT TX (16): AT+CGDATA="PPP",
    [INFO][CELL]: AT TX (1): 4
    [INFO][CELL]: AT TX (1): <cr>
    [INFO][CELL]: AT RX (6): <cr><ln>CONN
    [INFO][CELL]: AT RX (2): EC
    [INFO][CELL]: AT RX (1): T

I added a print in the lwip's `pppos_input` function. The modem appears to be sending a "NO CARRIER" message. Tried this both on a EC20 and a EC25 as well, facing the same issue.

0xc0170, 2019-08-09:
cc @ARMmbed/mbed-os-wan

AriParkkila, 2019-08-12:
@Arjun765 That could be e.g. IP version conflict. You could debug it more by changing properties in `mbed_app.json` such as `lwip.ipv6-enabled": false`, and modifying properties at `QUECTEL_EC2X.cpp`, such as:
```
    0,  // AT_CGDATA
    1,  // PROPERTY_IPV4_STACK
    0,  // PROPERTY_IPV6_STACK
    0,  // PROPERTY_IPV4V6_STACK
```
Can you get more logs, especially about PDP context create/activate.

Arjun765, 2019-08-12:
Looks like some kind of timing issue. I use ITM to print log output. If i add a 100us delay in my "putc" function, it starts working very well.

>That could be e.g. IP version conflict
This did not help

Arjun765, 2019-08-12:
>Can you get more logs, especially about PDP context create/activate.

>>>[INFO][CELL]: New CellularContext  (0x2000b148)
Connecting...
[INFO][CELL]: CellularContext connect
[INFO][CELL]: AT TX ( 6): AT+CSQ
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 7): AT+CSQ<cr>
[INFO][CELL]: AT RX ( 2): <cr><ln>
[INFO][CELL]: AT RX ( 1): +
[INFO][CELL]: AT RX ( 1): C
[INFO][CELL]: AT RX ( 2): SQ
[INFO][CELL]: AT RX ( 1): :
[INFO][CELL]: AT RX ( 1):  
[INFO][CELL]: AT RX ( 1): 1
[INFO][CELL]: AT RX ( 1): 9
[INFO][CELL]: AT RX ( 1): ,
[INFO][CELL]: AT RX ( 1): 9
[INFO][CELL]: AT RX ( 1): 9
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <ln>
[INFO][CELL]: AT RX ( 2): <cr><ln>
[INFO][CELL]: AT RX ( 1): O
[INFO][CELL]: AT RX ( 1): K
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <ln>
[DBG ][CELL]: callback: 4104, err: 0, data: -1
[INFO][CELL]: RSSI -75 dBm
[DBG ][CELL]: callback: 4106, err: 0, data: 4096
[INFO][CELL]: Start connecting (timeout 1000 ms)
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 7): AT<cr><cr><ln>OK
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <ln>
[INFO][CELL]: AT TX ( 2): AT
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 9): AT<cr><cr><ln>OK<cr><ln>
[DBG ][CELL]: Init => Device ready
[INFO][CELL]: AT TX ( 6): AT+CSQ
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 7): AT+CSQ<cr>
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): <ln>+
[INFO][CELL]: AT RX ( 1): C
[INFO][CELL]: AT RX ( 1): S
[INFO][CELL]: AT RX ( 2): Q:
[INFO][CELL]: AT RX ( 1):  
[INFO][CELL]: AT RX ( 1): 2
[INFO][CELL]: AT RX ( 1): 5
[INFO][CELL]: AT RX ( 1): ,
[INFO][CELL]: AT RX ( 1): 0
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <ln>
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): <ln>O
[INFO][CELL]: AT RX ( 1): K
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <ln>
[DBG ][CELL]: callback: 4104, err: 0, data: -1
[INFO][CELL]: RSSI -63 dBm
[DBG ][CELL]: callback: 4106, err: 0, data: 4096
[DBG ][CELL]: AT flush
[INFO][CELL]: AT TX ( 4): ATE0
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 5): ATE0<cr>
[INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[INFO][CELL]: AT TX ( 9): AT+CMEE=1
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 3): <cr><ln>O
[... 898 lines cut ...]
[INFO][CELL]: AT RX ( 1): 0
[INFO][CELL]: AT RX ( 1): .
[INFO][CELL]: AT RX ( 1): 0
[INFO][CELL]: AT RX ( 1): "
[INFO][CELL]: AT RX ( 1): ,
[INFO][CELL]: AT RX ( 1): 0
[INFO][CELL]: AT RX ( 1): ,
[INFO][CELL]: AT RX ( 1): 0
[INFO][CELL]: AT RX ( 1): ,
[INFO][CELL]: AT RX ( 1): 0
[INFO][CELL]: AT RX ( 1): ,
[INFO][CELL]: AT RX ( 1): 0
[INFO][CELL]: AT RX ( 1): <cr>
[INFO][CELL]: AT RX ( 1): <ln>
[INFO][CELL]: AT RX ( 2): <cr><ln>
[INFO][CELL]: AT RX ( 1): O
[INFO][CELL]: AT RX ( 1): K
[INFO][CELL]: AT RX ( 2): <cr><ln>
[INFO][CELL]: Found PDP context 1
[INFO][CELL]: CellularContext PPP connect
[INFO][CELL]: AT TX ( 9): ATD*99***
[INFO][CELL]: AT TX ( 1): 1
[INFO][CELL]: AT TX ( 1): #
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX (10): <cr><ln>CONNECT

AriParkkila, 2019-08-13:
@Arjun765 The modem has many PDP contexts which may cause some (PPP) connect problems, so those could be cleaned by `AT+CGDCONT=#`. After CONNECT it starts PPP negotiation so you probably need to enable PPP logs too.
.

AriParkkila, 2019-08-13:
@Arjun765 AT command should be `AT+CGDCONT=1`, `AT+CGDCONT=2`, etc., see the log for actual IDs. I have seen PPP connect problems when the modem already had an existing/active contexts, but those were later at NCP phase and not timing related. It's difficult to say why increasing CPU load that little would help to connect LCP... What is the PPP failure reason and how long time it takes to fail, maybe could try to change some PPP timeouts?

Arjun765, 2019-08-19:
Ive made two changes and so far it seems to be working. Disabled `AT_CGDATA` and dial `*99#` instead of `*99***<CID>#`

0xc0170, 2019-08-19:
@Arjun765 Shall this issue be closed?

Arjun765, 2019-08-20:
Not sure, i had to make two changes to Mbed to get it to work. If these are necessary and correct the changes should be added to Mbed.

AriParkkila, 2019-08-21:
@Arjun765 Can you check that CID is correct in `*99***<CID>#`, there are previously no auth/credentials set and no contexts activated. Also could check if EC has the latest firmware..?

AriParkkila, 2019-09-20:
@Arjun765 Mbed OS master has now a configuration `cellular.clear-on-connect: true` to clear any extra contexts, can you add that in your `mbed_app.json`.

AriParkkila, 2019-10-01:
@Arjun765 Any update on this?

AriParkkila, 2019-12-19:
This issue can be closed due to inactivity?

closed 2020-05-08 by 0xc0170