ARMmbed/mbed-os #11197: What is the ultimate end effect of setting bare-metal in the configuration

pauluap, 2019-08-11:
### Description

Mbed-OS v5.12

I'm using a custom CMake build process because I have requirements that mbed-cli is unable to accommodate.  

I want to set up a rtos-less build.  I've seen issues #7800 and #7794.  I have no problems with eliminating the files those two issues cover, but I've also seen a reference to bare-metal in https://os.mbed.com/docs/mbed-os/v5.12/tutorials/migrating-to-mbed-os-5.html

I would like to know what effect that configuration has - what defines are added or whatnot?  I tried grepping for bare-metal in the mbed-cli and tools directory but came up empty.  Is there a central source for the mapping of the directives in the json files to defines/files/etc that are added to the build process?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

40Grit, 2019-08-11:
The tools team is still in a bit of a transition.

There is a particular pull request that enabled baremetal config.

Do a search for it in the mbed-os GitHub repository.

pauluap, 2019-08-11:
Are you referring to #9800? It seems to say that bare metal just needs for drivers and platform to be pulled in.

But there's also this - https://github.com/ARMmbed/mbed-os/blob/7d7416558a28916b576feee2845e31bc550af080/rtos/Mutex.h#L60 which seems to say that there is some kind of define change - and later in the file, there is one, `MBED_CONF_RTOS_PRESENT` but I don't see anything in the tools/cli stuff that actually turns that define on/off.  Are there any other defines that should be set?

40Grit, 2019-08-11:
@theotherjimmy, just in case you have a quick answer

0xc0170, 2019-08-12:
Internally, we used to use `MBED_CONF_RTOS_PRESENT ` to check if RTOS is included or not.

Tools team might correct this: `requires` just selects folders (and then it pulls its config as well) - does not do any additional magic (specific macros or configs additions, all is already in the tree). In case of rtos - the present macro would be defined if rtos folder included (thus can be used to check if rtos is present or not).

cc @ARMmbed/mbed-os-tools

pauluap, 2019-08-12:
Ah, I think that I get it now - the defines and values in mbed_conf.h file comes from the processing of the .json file, so in this file, https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.12/rtos/mbed_lib.json, the `"present": 1` gets processed to `#define MBED_CONF_RTOS_PRESENT 1` and `"timer-thread-stack-size": {...}` gets processed to `#define MBED_CONF_RTOS_TIMER_THREAD_STACK_SIZE <value>` and so forth without a separate central mapping for json content to defines in the tools source.

0xc0170, 2019-08-12:
That is my understanding as well. 
Your own tool should do something similar (implement config) or reuse what tools provide (if possible, to access Config object and use its functionality).

pauluap, 2019-08-12:
Thank you.  This answers my question.

closed 2019-08-12 by pauluap