ARMmbed/mbed-os #11202: Tracking issue for output synchronisation between mbed_trace, printf...

ithinuel, 2019-08-12:
### Description

This is a tracking issue to discuss to possible solution(s) to the output synchronisation problem.

Mbed-os currently provides several mutually unaware ways to output traces and log messages to either a specific output and/or a retargetable destination.

This creates friction during development :
- corrupted greentea kv token generates false negative during CI testing.
- decreased readability of log renders difficult the debugging multithreaded tests.
- lowers the quality of the developer experience.

Prior discussion on #9954.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

ithinuel, 2019-08-12:
For the background history, I faced this issue when trying to build a Network Interface driver for a wifi module.
The driver dispatches its events in its own thread and prints logs using `mbed_trace`. However, greentea uses direct `RawSerial`, unity `putc` and some (most) of the netsocket tests are using `printf`.
Trying to reconstruct the corrupted log of intermingled messages coming from various parts of the software was a hassle, hence the hack proposed in #9954.

loverdeg-ep, 2019-08-12:
What is the IDEAL design and feature-set for the various serial debug, test, logging, command line interfaces?

How do we hack what exists to get on the path to that ideal?

40Grit, 2019-11-15:
@adbridge Can this be considered a defect so that the discussion #11861 can be addressed?

Or should it be considered a new feature and PR created to discuss and implement?

adbridge, 2019-11-15:
@40Grit we can make an exception in this case I think.

40Grit, 2019-11-18:
@0xc0170 @kjbracey-arm @SeppoTakalo @kivaisan 

@trowbridgec 

What is the best long term solution.
We have no fear of getting dirty to make it clean.

0xc0170, 2019-11-22:
> What is the best long term solution.

@ARMmbed/mbed-os-core Please review

kivaisan, 2019-12-03:
cc: @AriParkkila 
Could we get swo support for Daplink? And use that for mbed-os internal traces? That would probably solve the problem with Greentea which is really pain in the ass if you need to debug some test failures.

loverdeg-ep, 2019-12-03:
Sure would be nice but not all hic's support swo though? Possibly not all targets hook it up either?

AGlass0fMilk, 2019-12-27:
> Sure would be nice but not all hic's support swo though? Possibly not all targets hook it up either?

(Including Cortex-M0)

adbridge, 2020-10-15:
Is this still an issue in Mbed 6 ?

loverdeg, 2020-11-30:
@AGlass0fMilk @trowbridgec is this still an issue in os6?

0xc0170, 2020-11-30:
> @ithinuel it has been 5 days since the last reminder. Could you please update the issue header as previously requested?

Sorry for the too many pokes here, we will fix the bot.

loverdeg, 2020-11-30:
Bot has been leading to me bumping the stale issues a little more vigorously. So maybe just extend it to 10 days?

ithinuel, 2020-12-10:
@adbridge As far as I know this hasn't been fixed yet.

@0xc0170 Np, GH is too noisy so it's filtered out anyway ;)

ithinuel, 2021-02-20:
I (finally) updated the description to please the Bot Overlord :) 

As far as I know this is definitely still an issue in Mbed 6.

ghseb, 2021-07-22:
Hello, Im not sure if this is the right direction, but regarding resource conflicts between greentea and mbed_trace there might be the possibility to alter the configuration of mbed-trace in `_GREENTEA_SETUP_COMMON` in a way that a print function is set (via `mbed_trace_print_function_set`) that is capable of tunneling trace printouts via greentea's key-value protocol.

Related section:
https://github.com/ARMmbed/mbed-os/blob/master/features/frameworks/greentea-client/source/greentea_test_env.cpp#L85 

For a short term fix (to avoid conflicts and false negative tests) it might make sense to temporary disable trace printouts right there.