ARMmbed/mbed-os #11207: STM32F7 Asynch UART bug with read/write

embeddedteam103, 2019-08-13:
Hello 
I've noticed a bug when using asynchronous UARTSerial on STMF7.

I have used read to get some bytes from UART interface,
so far all ok, 
when the reading was completed i called write (also asynch) then , i've received an intereupt that called my callback with RX_EVENT , (when im using TX !) .

Has anyone noticed this bug? 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-08-13:
cc @ARMmbed/team-st-mcd Please review

jeromecoutant, 2019-08-19:
Hi
Is it possible to share a short application code which reproduces the issue ?
Regards,

0xc0170, 2019-10-24:
@embeddedteam103 Please respond with a code snippet - how can we reproduce. + reopen

I'll close this meanwhile.

closed 2019-10-24 by 0xc0170