ARMmbed/mbed-os #11210: Support for STM32F427

WasabiFan, 2019-08-13:
### Description

The STM32 devices listed in `targets/TARGET_STM/TARGET_STM32F4` don't include the STM32F427. Is this chip supported, and if so, what device type does it fall under? The ARM website lists it together with the STM32F437 -- is that appropriate?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-08-13:
@ARMmbed/team-st-mcd Please review

LMESTM, 2019-08-13:
Hello @WasabiFan 
indeed this MCU is not yet part of the supported ones. The mbed port currently requires a board to be used as a TARGET for a given MCU. What board including STM32F427 are you looking for ? 

The steps on how to add a new platform is described here: 
https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform
MCU STM32F427 would be very close to STM32F427. We definitely encourage you to add this new MCU support by sending a Pull Request, we will then review the proposal and approve it on @ARMmbed/team-st-mcd behalf.

WasabiFan, 2019-08-13:
Thanks for the clarification. If I decide to pursue mbed for my application I'll certainly open a PR.

Were you referring to STM32F437 when you said STM32F427 the last time, or a different chip?

> What board including STM32F427 are you looking for ?

[RoboMaster Development Board Type A](https://store.dji.com/product/rm-development-board-type-a).

LMESTM, 2019-08-13:
> Thanks for the clarification. If I decide to pursue mbed for my application I'll certainly open a PR.

Great  !

> Were you referring to STM32F437 when you said STM32F427 the last time, or a different chip?
> 

Yes indeed - I fixed my comment

0xc0170, 2019-08-13:
> The steps on how to add a new platform is described here:
https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform
MCU STM32F437 would be very close to STM32F427. We definitely encourage you to add this new MCU support by sending a Pull Request, we will then review the proposal and approve it on @ARMmbed/team-st-mcd behalf.

🎉 

Thanks. I consider this is answered, will close as "not supported by default, can be added anytime via new PR".

closed 2019-08-13 by 0xc0170

jhgorse, 2023-10-20:
targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F427xI
targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F427xG
appear to exist now.  FWIW.

https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32F4/