ARMmbed/mbed-os #11214: Preprocessing of linker script does not include macro definitions

jacob-larsen-samsung, 2019-08-13:
### Description

Any linker scripts defined by a target are put through the C preprocessor before linking. Unfortunately, the macro definitions from config and COMPONENT/FEATURE/TARGET are not included in this process, which limits the usefulness of this. 

I patched around a bit, and if I change to the following in gcc.py in the link() function, it works as expected:

```
        # Preprocess
        if mem_map:
            preproc_output = join(dirname(output), ".link_script.ld")
            cmd = (
                self.preproc + [mem_map] + self.ld[1:] + ["-o", preproc_output] +
                self.get_compile_options(self.get_symbols(), [])
            )
            self.notify.cc_verbose("Preproc: %s" % ' '.join(cmd))
            self.default_cmd(cmd)
            mem_map = preproc_output
```

Would this be something that can be patched into mbed?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-08-13:
Is this a limitation for specific use case? We implemented preprocessing linker scripts for bootloader - how to get ADDR macros in. Therefore macros should get in but probably not all from the config?

Can you provide code snippet of application to reproduce the issue?

cc @ARMmbed/mbed-os-tools

jacob-larsen-samsung, 2019-08-13:
My use case is that I want a section to link differently depending on if a config option is enabled or not. It enables a debug mode where log strings are not present in the image that is loaded on target, but instead the strings get referenced from the host where a debug tool loads them from the elf file. This is an optional feature that should be enabled through a configuration option.

Usage example:
```
#if MBED_CONF_TARGET_OFFLOAD_LOG
	.none 0x1000000 (DSECT) :
	{
		*(.rodata.notloaded.*)
		*(.rodata.__FUNCTION__.*)
	}
#else
	.none :
	{
		*(.rodata.notloaded.*)
		*(.rodata.__FUNCTION__.*)
	} > flash
#endif
```

andrewc-arm, 2019-08-16:
Hi, @jh228
As we spoke, above internal JIRA case is what our a bot auto created. It's just for Arm internal discussion space where we discuss internally and later publish our finding to public space like this ticket. 😁 

----

Hi, @jacob-larsen-samsung 
I reviewed your code change around https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/gcc.py#L246 and I think your code change improves our gcc.py script. Adding preprocessor symbols to preprocessor and linker script improves versatility of user linker scripts. Now I understand better from your code.
Since you are the main source contributor, could you kindly make this as a proper Pull Request?
https://github.com/ARMmbed/mbed-os/pulls. Please CC me to the case and github will auto bring in other expert people who submitted the code near your change.

Thank you!

(CC: @MarceloSalazar , @ohadhawk )

jh228, 2019-08-16:
Hi @andrewc-arm 

Thanks for support. :-)
Jacob will update his patch and comment at pull-request site.

closed 2019-09-10 by 0xc0170